ros2/ros2 #111: Client requests are not sent immediately

esteve, 2015-09-03:
Services tests sleep for a short while to ensure that the client actually sends the request:

https://github.com/ros2/system_tests/pull/42#discussion-diff-38600456R62

jacquelinekay, 2015-11-11:
@esteve Do you mind if a mark this as a duplicate of duplicate of ros2/rmw_connext#112?

esteve, 2015-11-11:
@jacquelinekay sure, no problem!

wjwwood, 2016-01-26:
I'm just trying to clean up some old tickets. @esteve can we close this if it is a duplicate of ros2/rmw_connext#112? Or perhaps vice versa?

esteve, 2016-01-26:
I think it's safe to close this ticket and focus on @jacquelinekay 's ros2/rmw_connext#112

wjwwood, 2016-01-26:
Ok, thanks.

dirk-thomas, 2016-01-26:
I think `sending immediately` and `deterministic` are very different. Imo sending the request immediately would be good to support (if possible). Either this should be explicitly mentioned in the other ticket or this should stay open.

wjwwood, 2016-01-26:
Well I'll reopen it then. If someone consolidates the issues in the other ticket then they can re-close this one.

dirk-thomas, 2019-09-05:
By now there is an API for the client to check / wait for the service to be available which is the right thing to do in an asynchronous system. Closing.

closed 2019-09-05 by dirk-thomas