ros2/ros2 #112: Python2.7 PyYaml mixing with Python3.4

IanTheEngineer, 2015-09-04:
I'm not sure if I should be posting this on the `vcstool` repo or here. In following the Linux Development Setup guide wiki page, I ran into an issue of python 2.7 messing with python 3.4, even after installing the proper PyYaml package for Python 3.4:
```
$ vcs import ~/ros2_ws/src < ros2.repos
Traceback (most recent call last):
  File "/usr/local/bin/vcs", line 9, in <module>
    load_entry_point('vcstool==0.1.14', 'console_scripts', 'vcs')()
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 558, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 2682, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 2355, in load
    return self.resolve()
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 2361, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.4/dist-packages/vcstool/commands/__init__.py", line 5, in <module>
    from .import_ import ImportCommand
  File "/usr/local/lib/python3.4/dist-packages/vcstool/commands/import_.py", line 6, in <module>
    import yaml
  File "/usr/local/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    from error import *
ImportError: No module named 'error'
```
Forcing python 3.4 in the front of my `PYTHONPATH` fixes the issue, but I'm sure there must be a more elegant way to help those installing Python 3.x for the first time on their workstation:
```
$ echo $PYTHONPATH
/usr/local/lib/python2.7/dist-packages:
$ PYTHONPATH=/usr/local/lib/python3.4/dist-packages:$PYTHONPATH
$ echo $PYTHONPATH
/usr/local/lib/python3.4/dist-packages:/usr/local/lib/python2.7/dist-packages:
$ vcs import ~/ros2_ws/src < ros2.repos
........................
=== /home/bb8/ros2_ws/src/ament/ament_cmake (git) ===
....
```
Now vcstool is happy :)

dirk-thomas, 2015-09-04:
I assume you have installed the Python 3 version of `vcstool` but have the Python 2 folder on the `PYTHONPATH`? Then your workaround is probably the only way around the problem.

If you want to avoid this you might either remove the Python 2 path from the environment or install the Python 2 package of `vcstool`.

dirk-thomas, 2015-09-15:
I will close this ticket since it seems to be a configuration problem of the environment.

closed 2015-09-15 by dirk-thomas

IanTheEngineer, 2015-09-16:
Ah, it was always an environment configuration issue, but I think it should
be documented in the tutorial. If you think so as well, I'll submit a pull
request.

dirk-thomas, 2015-09-16:
Sure, please go ahead and purpose a patch.