ros2/ros2 #113: Problem compiling ros1_bridge

EnricoMingo, 2015-09-04:
Hi, I have just tried to install ROS2 but I have this error when I compile the ros1_bridge:

```ImportError: dynamic module does not define init function (PyInit__roslz4)```

I am using the script:

src/ament/ament_tools/scripts/ament.py

dirk-thomas, 2015-09-04:
Please pay the exact steps you performed as well as the full error message.

wjwwood, 2015-09-04:
@EnricoMingo it may be that you missed this note:

From: https://github.com/ros2/ros1_bridge/blob/master/README.md#build-the-bridge-from-source
> Note: due to a bug in ROS 1 rosbag you currently must patch an installed file or build ROS 1 ros_comm from source. Edit the file /opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py with root permissions and comment out the line containing import roslz4. Make sure to revert the change after building the bridge. The next ROS 1 patch release will resolve this problem (https://github.com/ros/ros_comm/pull/642).

Until we get a new release of `ros_comm` out, you'll have to patch the `rospy` file as the note suggests. You can undo the patch after building.

EnricoMingo, 2015-09-04:
> will resolve this problem (ros/ros_comm#642

Ok, thank you! I'll try!

dirk-thomas, 2015-09-15:
I will close this ticket for now. If you still run into the issue please comment here and the ticket can be reopened.

closed 2015-09-15 by dirk-thomas