ros2/ros2 #123: generated files

ainchauspe, 2015-09-21:
Hi,

This is Anita from Erle Robotics. I come to you because I'm getting an error when building this package https://github.com/erlerobot/ros2_joy and I'm asking for some advice.

This is not the first time I get this error: some of the includes do not exist. I think it might be something I do wrong in CMakeLists.txt, and consequently, some of the files are not generated when the package is built. I do not know what I am missing, any idea?

Thanks,

Anita

wjwwood, 2015-09-21:
@ainchauspe could you provide the output of the error?

ainchauspe, 2015-09-21:
sure @wjwwood 
```
In file included from /home/anita/ros2_ws/install/include/sensor_msgs/msg/joy.h:12:0,
                 from /home/anita/ros2_ws/src/ros2/joy/src/joy_node.cpp:41:
/home/anita/ros2_ws/install/include/sensor_msgs/msg/joy__struct.h:13:31: fatal error: std_msgs/Header-c.h: No existe el archivo o el directorio
 #include <std_msgs/Header-c.h>
```

1- `joy__struct.h` is a file generated when the package is built.
2- the file `std_msgs/Header-c.h` does not exist.

wjwwood, 2015-09-21:
The files which end in `.h` are for our, as yet incomplete, C library. You'll need to use `.hpp` every where. I'd recommend you do `#include <sensor_msgs/msg/joy.hpp>` instead.

(please close the issue if this works for you)

closed 2015-09-21 by ainchauspe