ros2/ros2 #124: use FastRTPS

dirk-thomas, 2015-09-24:
This tickets collects all necessary steps in order to build ROS 2 with FastRTPS:

* [x] add the eProsima repos: https://github.com/ros2/ros2/pull/116

**Linux:**
* [x] build CI job
  * [x] Docker needs to install missing boost dependencies: https://github.com/ros2/ros2/pull/115
* [ ] pass all tests
  * http://ci.ros2.org/job/ros2_batch_ci_linux/389/
  * [ ] `rmw_fastrtps_cpp` uses all common linters but has different code style and fails those tests: https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/pull/4
  * http://ci.ros2.org/job/ros2_batch_ci_linux/391/

**OS X:**
* [ ] build CI job
  * http://ci.ros2.org/job/ros2_batch_ci_osx/289/
  * fixed handling of unicode characters in `vcstool`
  * http://ci.ros2.org/job/ros2_batch_ci_osx/290/
  * [ ] set locale in order for Python not to use `ascii` encoding
* [ ] pass all tests on

**Windows:**
* [ ] build CI job
  * http://ci.ros2.org/job/ros2_batch_ci_windows/436/
  * [ ] slave needs missing boost dependency
* [ ] pass all tests

dirk-thomas, 2015-09-25:
All tests pass on Linux now.

dirk-thomas, 2015-10-06:
The latest builds passed on all three platforms.

But on OS X and Windows there are some compiler warnings as well as failing tests.

dirk-thomas, 2015-11-04:
* [x] eProsima/ROS-RMW-Fast-RTPS-cpp#6
* [ ] eProsima/Fast-RTPS#8

dirk-thomas, 2015-11-10:
Latest set of builds:
* http://ci.ros2.org/job/ros2_batch_ci_linux/578/
* http://ci.ros2.org/job/ros2_batch_ci_osx/462/ (2 times double free warning)
* http://ci.ros2.org/job/ros2_batch_ci_windows/621/ (we can't do cross-vendors tests on Windows anymore, so the tests are only covering rmw_fastrtps_cpp now) I don't know why most executables fail with the rc 3221225781. @richiprosima Any idea?

richiprosima, 2015-11-11:
Hi Dirk,

I've checked on my Windows. My dll libraries are installed in ${CMAKE_INSTALL_PREFIX}/lib. Although install/local_setup.bat was executed, my libraries weren't found by the system.
I've updated our repositories and now dll libraries are installed in ${CMAKE_INSTALL_PREFIX}/bin.

dirk-thomas, 2015-11-11:
I triggered another Windows build:
* http://ci.ros2.org/job/ros2_batch_ci_windows/626/

richiprosima, 2015-11-11:
It keeps on failing in your system.  Can you copy your build options please?

dirk-thomas, 2015-11-11:
We don't pass any specific options. In this build neither Connext nor OpenSplice is being sourced. It uses the repos defined in https://raw.githubusercontent.com/ros2/ros2/eprosima/ros2.repos. You can see the exact hashed in the console output (after this command `vcs log -l1 "src"`). The build is being started with the following command:

    env.bat "C:\tools\python\python.exe" -u ".\src\ament\ament_tools\scripts\ament.py" build --build-tests --build-space "build" --install-space "install" "src" --isolated

The content of the `env.bat` is shown above the command. It only sets up the Visual Studio environment.

richiprosima, 2015-11-11:
I use those repos. But I use

 "C:\tools\python\python.exe" -u ".\src\ament\ament_tools\scripts\ament.py" build --build-tests

I will try with yours. 

Thanks

dirk-thomas, 2015-11-11:
The `--build-space` and `--install-space` arguments are only repeating the default values. The same for the `src` argument. But `--isolated` does make a difference.

richiprosima, 2015-11-12:
I used `--isolated` and my test results were:

```
88% tests passed, 2 tests failed out of 17

Label Time Summary:
copyright     =   0.64 sec
cppcheck      =   1.06 sec
cpplint       =   1.53 sec
lint_cmake    =   0.64 sec
linter        =   5.09 sec
nose          =  41.28 sec
uncrustify    =   1.22 sec

Total Test time (real) =  50.05 sec

The following tests FAILED:
          1 - test_publisher_subscriber_cpp__rmw_fastrtps_cpp (Not Run)
          2 - test_subscription_valid_data_cpp__rmw_fastrtps_cpp (Not Run)
Errors while running CTest

Stopped after package 'test_communication'

c:\Users\eprosima\ros2>python src\ament\ament_tools\scripts\ament.py test_results
Summary: 43 tests, 0 errors, 0 failures
```
Are boost libraries accessible via PATH environment variable?

dirk-thomas, 2015-11-12:
Boost was not on the PATH until now. The slave only had the following env variables set:

    BOOST_LIBRARYDIR=C:\local\boost_1_59_0\lib64-msvc-14.0
    BOOST_ROOT=C:\local\boost_1_59_0

I added the lib folder to the PATH and triggered another Windows build: http://ci.ros2.org/job/ros2_batch_ci_windows/641/

dirk-thomas, 2015-11-12:
Boost was not on the PATH until now. The slave only had the following env variables set:

    BOOST_LIBRARYDIR=C:\local\boost_1_59_0\lib64-msvc-14.0
    BOOST_ROOT=C:\local\boost_1_59_0

I added the lib folder to the PATH and triggered another Windows build: http://ci.ros2.org/job/ros2_batch_ci_windows/646/ (this build does not show compiler warnings since the Jenkins plugin failed to extract them from the FastRTPS build)

dirk-thomas, 2015-12-03:
@richiprosima The last update has been three weeks ago. The current state is:
* http://ci.ros2.org/job/ci_linux/658/ several tests fail
* http://ci.ros2.org/job/ci_osx/541/ still many compiler warnings and more failing tests then on Linux
* http://ci.ros2.org/job/ci_windows_fastrtps/3/ this job runs without warning parsing since the Jenkins plugin seem to not work well when building FastRTPS, 

What is your ETA to have it build on all platforms without warnings and pass all tests? We would like to announce FastRTPS support in an upcoming alpha release but the date of next one (Alpha 3, Dec 18th.) is approaching fast. several tests fail

Also how is the support for fragmentation coming along? Will it be ready by the end of the month?

richiprosima, 2015-12-11:
We've updated repository [Fast-RTPS](http://github.com/eProsima/Fast-RTPS) and [ROS-RMW-Fast-RTPS-cpp](eProsima/ROS-RMW-Fast-RTPS-cpp). Also we've created a pull request ros2/system_tests#85.

Now on my linux machine only fails cpplint and uncrustify tests. After running tests ament displays:
```
The following tests FAILED:
     17 - cpplint (Failed)
     19 - uncrustify (Failed)
Errors while running CTest
make: *** [test] Error 8
```
We didn't have time to check it on Windows. We will check next week.

dirk-thomas, 2015-12-14:
Great, the latest round of jobs (without the PR to system_tests) looks much better:
* http://ci.ros2.org/job/ci_linux/720/ one failing test related to FastRTPS, expected to be fixed with changed sleep times
* http://ci.ros2.org/job/ci_osx/597/ all failing tests are unrelated to FastRTPS
* http://ci.ros2.org/job/ci_windows_fastrtps/4/ multiple failing tests related to FastRTPS, expected to be fixed with changed sleep times

It looks like the listed failed tests on you local machine (`cpplint`, `uncrustify`) are left over result file from earlier runs. I think we removed those linters a while ago: eProsima/ROS-RMW-Fast-RTPS-cpp#4 Have you tried to delete the result file and do those failing tests reappear after running the tests again?

Any updates on fragmentation support?

gerkey, 2015-12-15:
@wjwwood Please make PR to FastRTPS for upcoming rmw API changes.

gerkey, 2015-12-15:
@dirk-thomas To add FastRTPS to our CI jobs as a supported rmw implementation.

dirk-thomas, 2015-12-16:
@richiprosima I have created a pull request to fix the compiler warnings in Fast-CDR: eProsima/Fast-CDR#3

For the remaining warning in Fast-RTPS I am not sure how you want to address them. Many .cpp files define a constant `CLASS_NAME` which isn't used anywhere (e.g. https://github.com/eProsima/Fast-RTPS/blob/master/src/cpp/rtps/builtin/data/ParticipantProxyData.cpp#L39 https://github.com/eProsima/Fast-RTPS/blob/master/src/cpp/rtps/builtin/data/ReaderProxyData.cpp#L24). Can these variables be removed or should the warning be disabled with a scoped pragma?

And one more warning regarding this line: https://github.com/eProsima/Fast-RTPS/blob/eb6ecddf018ddd0552c25a3ed298546e2c4fe92f/src/cpp/rtps/resources/ResourceSendImpl.h#L86 The private variable is set to null in the constructor. Should it be removed or the warning suppressed otherwise?

richiprosima, 2015-12-17:
@dirk-thomas I have merged eProsima/Fast-CDR#3

Also I have removed warnings you suggested and I have updated the eProsima/Fast-RTPS@c58b2b4 repository.

dirk-thomas, 2015-12-17:
The newest OS X build looks good - zero warnings: http://ci.ros2.org/job/ci_osx/639/

dirk-thomas, 2015-12-17:
I have made changes to our CI stuff which allows to cleanly disable any rmw impl.:
* https://github.com/ros2/ros2/pull/171
* https://github.com/ros2/ros2/pull/172
* https://github.com/ros2/ros2/pull/116

dirk-thomas, 2015-12-17:
@richiprosima Can you please comment on the progress for fragmentation support.

dirk-thomas, 2016-01-09:
I will close this ticket since we have fully integrated FastRTPS into our CI infrastructure.

Regarding fragmentation support the new ETA from eProsima for the feature is end of May.

closed 2016-01-09 by dirk-thomas

murikill, 2019-03-27:
Hi @dirk-thomas,

may I use ros2-FastRTPS with standalone (non ros2) FastRTPS? Do you have any tips about settings?

Many thanks!

dirk-thomas, 2019-03-27:
@murikill this is a three year old closed ticket and not the right place to ask random questions. You might want to ask on answers.ros.org (and post a link to your question here for future readers to find it).