ros2/ros2 #127: flaky pub / sub and request / response tests

dirk-thomas, 2015-10-13:
I ran a set of jobs as a base line:

* http://ci.ros2.org/job/ros2_batch_ci_osx/335/ **0** failing tests
* http://ci.ros2.org/job/ros2_batch_ci_windows/475/ **57** failing tests

Then I ran a set of jobs with `--ctest-args --repeat-until-fail 20`:
* http://ci.ros2.org/job/ros2_batch_ci_osx/336/ **4** failing tests
* http://ci.ros2.org/job/ros2_batch_ci_windows/476/ **59** failing tests

The arguments have no effect on Linux since the CMake version used on Ubuntu does not support this argument.

jacquelinekay, 2015-10-15:
duplicates:

ros2/rmw_connext#29 (has some logs and records of other failures)
ros2/system_tests#37 (link to RTI community post about known shared memory issue, probably not the cause since we see it in Connext and Opensplice)

wjwwood, 2016-01-26:
Should we close this since all referenced tickets are closed and we are tracking this with our nightly jobs?

dirk-thomas, 2016-01-26:
I would suggest to keep this open and aim to get it resolved in this cycle.

We happen to switch between creating tickets for failing jobs like this one and then later closing them since the problem is "covered" by the failing test. We should decide on a policy and stick to it.

dirk-thomas, 2016-10-14:
In the mean time we have updated to Xenial which has a newer CMake version and also got most of our tests stable (and for the ones which aren't retry them a few times). Closing...

closed 2016-10-14 by dirk-thomas