ros2/ros2 #129: "Ament Tutorial" build error

flobotics, 2015-10-18:
hi,
when i try to follow this https://github.com/ros2/ros2/wiki/Ament-Tutorial , it returns after running "src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install" command, this error:

+++ Building 'ament_index_cpp'
==> '. /home/ros/ros2_ws/build/ament_index_cpp/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/ros/ros2_ws/build/ament_index_cpp'
==> '. /home/ros/ros2_ws/build/ament_index_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/ros/ros2_ws/build/ament_index_cpp'
[ 16%] [ 50%] [ 83%] Built target gtest
Built target gtest_main
Built target ament_index_cpp
Linking CXX executable ament_index_cpp_utest
gtest/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<testing::TestPartResultReporterInterface*>::~ThreadLocal()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED2Ev[_ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED5Ev]+0x15): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED2Ev[_ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED5Ev]+0x2a): undefined reference to `pthread_key_delete'
gtest/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<std::vector<testing::internal::TraceInfo, std::allocator<testing::internal::TraceInfo> > >::~ThreadLocal()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED2Ev[_ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED5Ev]+0x15): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED2Ev[_ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED5Ev]+0x2a): undefined reference to `pthread_key_delete'
gtest/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<std::vector<testing::internal::TraceInfo, std::allocator<testing::internal::TraceInfo> > >::GetOrCreateValue() const':
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv]+0x18): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv]+0x7c): undefined reference to `pthread_setspecific'
gtest/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<testing::TestPartResultReporterInterface*>::CreateKey()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE9CreateKeyEv[_ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE9CreateKeyEv]+0x16): undefined reference to `pthread_key_create'
gtest/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<std::vector<testing::internal::TraceInfo, std::allocator<testing::internal::TraceInfo> > >::CreateKey()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE9CreateKeyEv[_ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE9CreateKeyEv]+0x16): undefined reference to `pthread_key_create'
gtest/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<testing::TestPartResultReporterInterface*>::GetOrCreateValue() const':
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv]+0x16): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv]+0x7a): undefined reference to `pthread_setspecific'
collect2: error: ld returned 1 exit status
make[2]: *** [ament_index_cpp_utest] Error 1
make[1]: *** [CMakeFiles/ament_index_cpp_utest.dir/all] Error 2
make: *** [all] Error 2

<== Command '. /home/ros/ros2_ws/build/ament_index_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' failed with exit code '2'

P.S.:  If someone installs completly new, you could perhaps mention that you need to install python3-vcstools and python3-vcstool

tested on ubuntu 14.04

thanks for help
flo

dirk-thomas, 2015-10-18:
Have you built the workspace before? If yes, it might use a different cached location of gtest. Please try to remove the file `build/ament_index_cpp/CMakeCache.txt` and then call the build command again.

The tutorial does not mention `vcstools` at all and doesn't need it.

The tutorial does not cover the install instructions. They are described on the page https://github.com/ros2/ros2/wiki/Linux-Development-Setup which also mentions Python 3 explicitly.

flobotics, 2015-10-18:
hi,
i only followed the tutorial at https://github.com/ros2/ros2/wiki/Ament-Tutorial, where i only can run the first 6 commands, then the error occured.

As i create the ros2_ws  directory in this tutorial, i dont think that i have built the workspace before, or?  But after deleting the build/ament_index_cpp/CMakeCache.txt file, the same error occured ?

The tutorial mentions a line "vcs import ~/ros2_ws/src < ros2.repos" , which i could only run, after installing python3-vcstools python3-vcstool, so i think its needed, or?

dirk-thomas, 2015-10-18:
Does you workspace contain a checkout of the https://github.com/ament/gtest_vendor repo? If not, do you have the Debian package `gtest` installed?

You only need `vcstool` to run that line - `vcstools` is a completely separate tool and is not used here.

flobotics, 2015-10-19:
in my ros2_ws/src directory are only a "ros2" and a "ament" directory, no gtest_vendor ? 

I installed libgtest-dev, deleted CMackeCache.txt file and build again, and now another error occured:

-- Build files have been written to: /home/ros/ros2_ws/build/rosidl_generator_cpp
==> '. /home/ros/ros2_ws/build/rosidl_generator_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/ros/ros2_ws/build/rosidl_generator_cpp'
Scanning dependencies of target rosidl_generator_cpp__c
Scanning dependencies of target rosidl_generator_cpp__cpp
[ 33%] [ 66%] Generating C code for ROS interfaces
Generating C++ code for ROS interfaces
Traceback (most recent call last):
  File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/bin/rosidl_generator_cpp", line 8, in <module>
    from rosidl_generator_cpp import generate_cpp
ImportError: No module named 'rosidl_generator_cpp'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
Traceback (most recent call last):
  File "/home/ros/ros2_ws/install/lib/rosidl_generator_c/rosidl_generator_c", line 6, in <module>
  File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/bin/rosidl_generator_cpp", line 20, in <module>
    rosidl_generator_cpp = loader.load_module()
  File "<frozen importlib._bootstrap>", line 539, in _check_name_wrapper
    from rosidl_generator_c import generate_c
  File "/home/ros/ros2_ws/install/lib/python3.4/site-packages/rosidl_generator_c/__init__.py", line 17, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/home/ros/ros2_ws/install/lib/python3.4/site-packages/rosidl_cmake/__init__.py", line 15, in <module>
    import em
  File "<frozen importlib._bootstrap>", line 1591, in load_module
ImportError: No module named 'em'
  File "<frozen importlib._bootstrap>", line 596, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 1220, in load
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1448, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/rosidl_generator_cpp/__init__.py", line 17, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/home/ros/ros2_ws/install/lib/python3.4/site-packages/rosidl_cmake/__init__.py", line 15, in <module>
    import em
ImportError: No module named 'em'
make[2]: *** [rosidl_generator_c/rosidl_generator_cpp/msg/bounded_array_bounded.h] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_cpp__c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [rosidl_generator_cpp/rosidl_generator_cpp/msg/bounded_array_bounded.hpp] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_cpp__cpp.dir/all] Error 2
make: *** [all] Error 2

<== Command '. /home/ros/ros2_ws/build/rosidl_generator_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' failed with exit code '2'

dirk-thomas, 2015-10-25:
I can't follow how you get the above error. Please provide a complete sequence of commands you ran (including how you installed the dependencies and cloned the repos). Maybe that helps to find out why it doesn't work for you.

flobotics, 2015-10-26:
it was a complete new installation of ubuntu 14.04 and here are all commands run so far:

    1  cd Dokumente/
    2  tar xf linux-4.1.7.tar.gz 
    3  ls
    4  cd linux-4.1.7/
    5  cp /boot/config-3.13.0-65-generic .config
    6  make olddefconfig
    7  sudo apt-get install libncurses5 libncurses5-dev
    8  make menuconfig
    9  zcat ../patch-4.1.7-rt8.patch.gz |patch -p1
   10  make menuconfig
   11  sudo apt-get install build-essentials devscripts
   12  sudo apt-get install build-essential devscripts
   13  make -j4
   14  sudo make -j4 modules_install
   15  cp arch/x86_64/boot/bzImage /boot/vmlinuz-4.1.7-rt8
   16  sudo cp arch/x86_64/boot/bzImage /boot/vmlinuz-4.1.7-rt8
   17  update-grub
   18  sudo update-grub
   19  sync
   20  sudo reboot
   21  sudo apt-get install git
   22  cd Dokumente/
   23  ls
   24  git clone git://git.kernel.org/pub/scm/linux/kernel/git/clrkwllms/rt-tests.git
   25  ls
   26  cd rt-tests/
   27  ls
   28  make all
   29  sudo apt-cache search numa
   30  sudo apt-get install libnuma1 libnuma-dev
   31  sudo apt-get update
   32  sudo apt-get install libnuma1 libnuma-dev
   33  make all
   34  ls
   35  ./cyclictest -t1 -p 80 -n -i 10000 -l 1000
   36  sudo apt-get install vim
   37  groups
   38  sudo vi /etc/security/limits.conf 
   39  sudo vi /etc/security/pam_env.conf 
   40  ulimit -a
   41  cd Dokumente/rt-tests/
   42  ls
   43  history 
   44  ./cyclictest -t1 -p 80 -n -i 10000 -l 1000
   45  cd ../
   46  ls
   47  mkdir ros2
   48  cd ros2/
   49  ls
   50  git clone https://github.com/ros2/rttest.git
   51  cd rttest/
   52  ls
   53  mkdir build
   54  cd build/
   55  sudo apt-get install cmake
   56  cmake ..
   57  cmake .. -DCMAKE_INSTALL_PREFIX=./
   58  make install
   59  cd ../examples/
   60  mkdir build
   61  cd build/
   62  cmake .. -DCMAKE_INSTALL_PREFIX=./
   63  make
   64  history 
   65  ls
   66  ls Dokumente/
   67  ls
   68  mkdir ~/ros2_ws
[... 80 lines cut ...]
  149  sudo vi ~/.bashrc 
  150  source ~/.bashrc 
  151  src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install 
  152  history |grep vcs
  153  cd ros2_ws/
  154  ls
  155  ls build/ament_index_cpp/CMakeCache.txt 
  156  rm build/ament_index_cpp/CMakeCache.txt 
  157  history 
  158  src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install 
  159  sudo apt-cache search vcstool
  160  cd ros2_ws/
  161  ls
  162  ls src/
  163  ls src/ament/
  164  sudo apt-get search gtest
  165  sudo apt-cache search gtest
  166  sudo apt-get install libgtest-dev
  167  rm build/ament_index_cpp/CMakeCache.txt 
  168  history 
  169  src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install 
  170  ls
  171  ls src/
  172  ls src/ros2/
  173  history

dirk-thomas, 2015-10-26:
I am sorry, but I can't help you based on the history of your shell.

Please try build the ROS 2 code again starting from a clean shell and empty folder. If the problem persists please share the commands and their output.

flobotics, 2015-10-26:
i tried now these commands:

181  sudo rm -r ros2_ws/
  182  mkdir ~/ros2_ws
  183  mkdir ~/ros2_ws/src
  184  cd ~/ros2_ws/
  185  wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
  186  ls
  187  vcs import ~/ros2_ws/src < ros2.repos
  188  ls
  189  src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install
  190  history 

and this is the result:

-- Build files have been written to: /home/ros/ros2_ws/build/rosidl_generator_cpp
==> '. /home/ros/ros2_ws/build/rosidl_generator_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/ros/ros2_ws/build/rosidl_generator_cpp'
Scanning dependencies of target rosidl_generator_cpp__cpp
[ 33%] Generating C++ code for ROS interfaces
Traceback (most recent call last):
  File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/bin/rosidl_generator_cpp", line 8, in <module>
    from rosidl_generator_cpp import generate_cpp
ImportError: No module named 'rosidl_generator_cpp'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/bin/rosidl_generator_cpp", line 20, in <module>
    rosidl_generator_cpp = loader.load_module()
  File "<frozen importlib._bootstrap>", line 539, in _check_name_wrapper
  File "<frozen importlib._bootstrap>", line 1591, in load_module
  File "<frozen importlib._bootstrap>", line 596, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 1220, in load
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1448, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/rosidl_generator_cpp/__init__.py", line 17, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/home/ros/ros2_ws/install/lib/python3.4/site-packages/rosidl_cmake/__init__.py", line 15, in <module>
    import em
ImportError: No module named 'em'
make[2]: *** [rosidl_generator_cpp/rosidl_generator_cpp/msg/bounded_array_bounded.hpp] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_cpp__cpp.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!

<== Command '. /home/ros/ros2_ws/build/rosidl_generator_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' failed with exit code '2'

dirk-thomas, 2015-10-26:
Looking at the code I don't see why that is happening for you and without being able to reproduce it I can't debug it myself.

Can you please add some debug output to the file in question (`/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/bin/rosidl_generator_cpp`). E.g. add `print('...')` statements between lines 7-21. Maybe that help narrowing down what happens.

flobotics, 2015-10-26:
line 7 and 8 looks:

try:
    from rosidl_generator_cpp import generate_cpp

and the error output says:
File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/bin/rosidl_generator_cpp", line 8, in
from rosidl_generator_cpp import generate_cpp
ImportError: No module named 'rosidl_generator_cpp'

couldnt it be, that this module is build, but not copied/placed where it should be ?

dirk-thomas, 2015-10-26:
That's why the code falls back to a relative import in the lines below.

flobotics, 2015-10-26:
can i look in my ros2_ws  if there is a rosidl_generator_cpp module exists ?

flobotics, 2015-10-26:
is this correct ?

 tree  src/ros2/rosidl/rosidl_generator_cpp/
src/ros2/rosidl/rosidl_generator_cpp/
├── bin
│   └── rosidl_generator_cpp
├── cmake
│   ├── register_cpp.cmake
│   └── rosidl_generator_cpp_generate_interfaces.cmake
├── CMakeLists.txt
├── include
│   └── rosidl_generator_cpp
│       ├── message_type_support_decl.hpp
│       └── service_type_support_decl.hpp
├── msg
│   ├── BoundedArrayBounded.msg
│   ├── BoundedArrayStatic.msg
│   ├── BoundedArrayUnbounded.msg
│   ├── Empty.msg
│   ├── PrimitivesBounded.msg
│   ├── PrimitivesStatic.msg
│   ├── PrimitiveStaticArrays.msg
│   ├── PrimitivesUnbounded.msg
│   ├── StaticArrayBounded.msg
│   ├── StaticArrayStatic.msg
│   ├── StaticArrayUnbounded.msg
│   ├── UnboundedArrayBounded.msg
│   ├── UnboundedArrayStatic.msg
│   └── UnboundedArrayUnbounded.msg
├── package.xml
├── resource
│   ├── msg.hpp.template
│   ├── msg__struct.hpp.template
│   ├── msg__traits.hpp.template
│   ├── srv.hpp.template
│   ├── srv__struct.hpp.template
│   └── srv__traits.hpp.template
├── rosidl_generator_cpp
│   ├── __init__.py
│   └── __pycache__
│       └── __init__.cpython-34.pyc
├── rosidl_generator_cpp-extras.cmake.in
└── test
    └── test_interfaces.cpp

9 directories, 31 files

dirk-thomas, 2015-10-26:
This is the layout of this package: https://github.com/ros2/rosidl/tree/master/rosidl_generator_cpp

flobotics, 2015-10-26:
i put in these print lines:

try:
    from rosidl_generator_cpp import generate_cpp
except ImportError:
    # modifying sys.path and importing the Python package with the same
    # name as this script does not work on Windows
    rosidl_generator_cpp_root = os.path.dirname(os.path.dirname(__file__))
    print( 'test0:' + rosidl_generator_cpp_root)
    rosidl_generator_cpp_module = os.path.join(
        rosidl_generator_cpp_root, 'rosidl_generator_cpp', '__init__.py')
    print('test1:' + rosidl_generator_cpp_module)
    if not os.path.exists(rosidl_generator_cpp_module):
        raise
    from importlib.machinery import SourceFileLoader
    print('test2:')
    loader = SourceFileLoader('rosidl_generator_cpp', rosidl_generator_cpp_module)
    print('test3:')
    rosidl_generator_cpp = loader.load_module()
    print('test4:')
    generate_cpp = rosidl_generator_cpp.generate_cpp
    print('test5:')

and this is the output:
+++ Building 'rosidl_generator_cpp'
==> '. /home/ros/ros2_ws/build/rosidl_generator_cpp/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/ros/ros2_ws/build/rosidl_generator_cpp'
==> '. /home/ros/ros2_ws/build/rosidl_generator_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/ros/ros2_ws/build/rosidl_generator_cpp'
[ 33%] [ 66%] Generating C++ code for ROS interfaces
Generating C code for ROS interfaces
test0:/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp
test1:/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/rosidl_generator_cpp/__init__.py
test2:
test3:
Traceback (most recent call last):
  File "/home/ros/ros2_ws/install/lib/rosidl_generator_c/rosidl_generator_c", line 6, in <module>
    from rosidl_generator_c import generate_c
  File "/home/ros/ros2_ws/install/lib/python3.4/site-packages/rosidl_generator_c/__init__.py", line 17, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/home/ros/ros2_ws/install/lib/python3.4/site-packages/rosidl_cmake/__init__.py", line 15, in <module>
    import em
ImportError: No module named 'em'
Traceback (most recent call last):
  File "/home/ros/ros2_ws/src/ros2/rosidl/rosidl_generator_cpp/bin/rosidl_generator_cpp", line 8, in <module>
    from rosidl_generator_cpp import generate_cpp
ImportError: No module named 'rosidl_generator_cpp'

flobotics, 2015-10-26:
dont know why, but the test1:  output is at the end   __init__.py   and not  init.py

dirk-thomas, 2015-10-26:
The double underscores are interpreted as markdown syntax. Please edit your comment and make the output a code snippet to prevent that.

flobotics, 2015-10-26:
what markdown do i need for code ?

wjwwood, 2015-10-26:
@flobotics You need to wrap them in triple ` ` `.

dirk-thomas, 2015-10-26:
Have you installed `em` as described in the documentation?

    ImportError: No module named 'em'

flobotics, 2015-10-26:
no, never read something about em?

dirk-thomas, 2015-10-26:
https://github.com/ros2/ros2/wiki/Linux-Development-Setup

flobotics, 2015-10-26:
ah, i saw this page, but i thought i dont need a dev setup, but .... shame on me..  sorry

perhaps if it interrests you, the test output:

100% tests passed, 0 tests failed out of 15

Label Time Summary:
copyright     =   0.28 sec
cppcheck      =   0.34 sec
gtest         =  13.99 sec
lint_cmake    =   0.27 sec
linter        =   1.19 sec
nose          =   3.15 sec
uncrustify    =   0.30 sec

Total Test time (real) =  18.34 sec

wjwwood, 2015-10-26:
@flobotics that's all right, but you do need the dev setup to build from source. You can avoid that step if you use the binaries. I'll go ahead and close this issue then. Please feel free to ask us to reopen it if you think your problem hasn't been solved.

closed 2015-10-26 by wjwwood

sailingmaster92, 2017-12-20:
I am new to ROS, ROS2 and Linux, but I ran into a similar problem. 
I was following the Ament-Tutorial: https://github.com/ros2/ros2/wiki/Ament-Tutorial

At this section I got stuck for about 3 hours. 
`vcs import ~/ros2_ws/src < ros2.repos`

`vcs` was not found and it was very difficult for me to find out, what I have to install. I finally installed python3-vcstools, but I wasn't aware, that there is another package called python3-vcstool. From my point of you, there should be a hint in this section of the tutorial. Got kind of frustrated.

clalancette, 2017-12-20:
@sailingmaster92 Sorry you had trouble with it.  The https://github.com/ros2/ros2/wiki/Ament-Tutorial does have a section at the top that says you should setup your development environment using the Building-from-source instructions, which installs `python3-vcstool` as part of it.  If you think that is still unclear, please suggest an update to the wiki instructions.

tfoote, 2017-12-20:
I moved the development environment setup into the prerequisites section instead of being a trailing sentence in the overview. Hopefully that will be clearer.