ros2/ros2 #139: "Windows Build from source" fails in trying to use Cmake

physincubus, 2015-11-03:
I believe that this is because the current build guide, for whatever reason, doesn't add cmake to the path. this should be fixed, or there should at least be an addition to the tutorial that it should be added to the path manually

dirk-thomas, 2015-11-03:
Can you please point to the tutorial you are referring to and what exactly you did?

https://github.com/ros2/ros2/wiki/Windows-Development-Setup#installing-build-tools mentions how to install CMake (`choco install -y cmake`) and that you might need to open a new shell in order to be able to run the new command.

physincubus, 2015-11-03:
The tutorial you referenced is the one. I followed the process and choco definitely installed cmake, but even using a new command prompt, upon building i get the error:

+++ Building 'ament_cmake_core'
Running cmake because arguments have changed.
ament build: Could not find 'cmake' executable

dirk-thomas, 2015-11-03:
Did the CMake installation show and UI where you had to select options?

physincubus, 2015-11-03:
no, just was output in the command prompt

dirk-thomas, 2015-11-03:
Thank. I added a note: https://github.com/ros2/ros2/wiki/Windows-Development-Setup/_compare/7b53810fa7373aeca6b5f6418f7c2208b32237ed...dcc1146b75986648a7e666c12f1822e70c30c30e

closed 2015-11-03 by dirk-thomas