ros2/ros2 #141: change packaging to not use venv

dirk-thomas, 2015-11-03:
In order to avoid embedding a venv specific Python path into the shebang lines.

wjwwood, 2015-12-18:
What did we do in the mean time? We need to do the same in alpha3. @gerkey @dirk-thomas I get this when testing the OS X binary:

```
% source ./setup.zsh
/Users/william/testing_alpha3/ros2-osx/local_setup.sh:22: no such file or directory: /usr/bin/env python3
```

Which is because of this snippet from the `local_setup.sh`:
```
# use the Python executable known at configure time
_ament_python_executable="/Users/osrf/jenkins/workspace/packaging_osx/ws/venv/bin/python"
```

wjwwood, 2015-12-18:
It looks like the issue (on my machine) is that the `/usr/bin/env python3` fallback option is getting evaluated as a single executable, hence the `no such file or directory: /usr/bin/env python3`.

I'll open a pr soon, but I'm still not sure why we haven't seen that.

closed 2016-10-25 by dirk-thomas