ros2/ros2 #144: Alpha 3 release (2015-12-18)

dirk-thomas, 2015-11-04:
* [ ] Cover all demos/examples with unit tests: ros2/ros2#143
* [ ] Refactor rclcpp into a library
  * [ ] Refactor into rcl
    * [ ] Python client library
* [ ] Include C message structures
* [ ] Component life-cycle
  * [ ] Introspection and orchestration APIs
  * [ ] Launch system
    * [ ] Use life-cycle and orchestration
  * [ ] Use class_loader / pluginlib
* [ ] Real-time-safe intra-process messaging
* [ ] Add TF support
* [ ] Support for FastRTPS
* [ ] FreeRTPS improvements: RMW support

dirk-thomas, 2015-11-17:
We have deferred "Component life-cycle" until Alpha 4.

dirk-thomas, 2015-12-16:
I have striked through the item "reduce memory requirement by splitting generated factories into separate source files". It looks like the current code the generated cpp file compiles and links in a reasonable time frame and without a significant memory usage. Therefore I don't think any further work is necessary right now.

Since I updated the content in the description this message is going to self destruct in 3, 2, 1 :bomb:

tfoote, 2016-01-05:
Leftover items moved to #180.

closed 2016-01-05 by tfoote