ros2/ros2 #147: Select one of connext or connext dynamic on Windows Jenkins job

jacquelinekay, 2015-11-04:
Needed for rclcpp_library

wjwwood, 2015-11-05:
This job should be only Connext Static: http://ci.ros2.org/job/ros2_batch_ci_linux/557/

wjwwood, 2015-11-06:
Ok if these CI work, then I think it will be good to go:

- http://ci.ros2.org/job/ros2_batch_ci_linux/562/
- http://ci.ros2.org/job/ros2_batch_ci_osx/430/
- http://ci.ros2.org/job/ros2_batch_ci_windows/578/ (opensplice)
- http://ci.ros2.org/job/ros2_batch_ci_windows/579/ (connext static)
- http://ci.ros2.org/job/ros2_batch_ci_windows/580/ (connext dynamic)

wjwwood, 2015-11-06:
Well those all failed ... Here are more Linux and OS X:

- http://ci.ros2.org/job/ros2_batch_ci_linux/563/
- http://ci.ros2.org/job/ros2_batch_ci_osx/431/

I'm working on Windows again...

wjwwood, 2015-11-06:
Ok, this should be the final list of CI:

- http://ci.ros2.org/job/ros2_batch_ci_linux/563/
- http://ci.ros2.org/job/ros2_batch_ci_osx/437/
- http://ci.ros2.org/job/ros2_batch_ci_windows/581/ (opensplice)
- http://ci.ros2.org/job/ros2_batch_ci_windows/582/ (connext static)
- http://ci.ros2.org/job/ros2_batch_ci_windows/583/ (connext dynamic)

wjwwood, 2015-11-06:
I'll restart the CI's on https://github.com/ros2/rclcpp/pull/140 when @dirk-thomas has figured out how to keep the history.

closed 2015-11-06 by wjwwood