ros2/ros2 #152: by the way:  rpi2B and ros2

flobotics, 2015-11-06:
hi,
i know, till now only x86 is supported, but perhaps it interrests you. I build ros2 on rpi2B + ubuntu-14.04 + rt-preempt. When i came to the end it prints:

94% tests passed, 1 tests failed out of 16

Label Time Summary:
copyright     =   2.37 sec
cppcheck      =   3.23 sec
gtest         =  26.87 sec
lint_cmake    =   2.37 sec
linter        =  10.63 sec
nose          =   8.85 sec
uncrustify    =   2.66 sec

Total Test time (real) =  46.44 sec

The following tests FAILED:
	  1 - gtest_allocator__rmw_opensplice_cpp (Failed)
Errors while running CTest
make: *** [test] Error 8

After that i run talker and listener, which outputs a warning after ending the talker with ctrl+c, which looks :

ubuntu@ubuntu:~/ros2_ws$ listener&
[1] 8054
ubuntu@ubuntu:~/ros2_ws$ talker
Publishing: 'Hello World: 1'
I heard: [Hello World: 1]
Publishing: 'Hello World: 2'
I heard: [Hello World: 2]
Publishing: 'Hello World: 3'
I heard: [Hello World: 3]
Publishing: 'Hello World: 4'
I heard: [Hello World: 4]
Publishing: 'Hello World: 5'
I heard: [Hello World: 5]
Publishing: 'Hello World: 6'
I heard: [Hello World: 6]
Publishing: 'Hello World: 7'
I heard: [Hello World: 7]
Publishing: 'Hello World: 8'
I heard: [Hello World: 8]
Publishing: 'Hello World: 9'
I heard: [Hello World: 9]
^Csignal_handler(2)
ubuntu@ubuntu:~/ros2_ws$ ps a========================================================================================
Report      : WARNING
Date        : Fri Nov 06 21:54:25 UTC 2015
Description : Missed heartbeat for node - heartbeat id : 1603243355. (DCPSPublication state 1)
Node        : ubuntu
Process     : listener <8054>
Thread      : OSPL Garbage Collector 7597f440
Internals   : V6.4.140407OSS/3b83e82/3b83e82/v_spliced/v_spliced.c/2439/0/1446846865.091159539

i build everything from source and used  the prismtech rpi dds package

good work, thanks

tfoote, 2015-11-06:
Thanks for letting us know. Did you have to patch anything to get it to work?

flobotics, 2015-11-06:
for the ros2 part i did not patch anything, i just used the image from https://wiki.ubuntu.com/ARM/RaspberryPi   and then followed https://github.com/ros2/ros2/wiki/Linux-Development-Setup

for the rt-preempt kernel i patched a raspbian 4.1.25 kernel, which gave me max: 90 microseconds with rt-preempts cyclictest

flobotics, 2015-11-07:
running listener on x86_64 host and running talker on rpi2B over network, works out of the box

gerkey, 2015-11-10:
That's great to know, thanks!

vmayoral, 2015-11-10:
>running listener on x86_64 host and running talker on rpi2B over network, works out of the box

+1. Double checked it last weekend and I can confirm ROS 2 works totally fine on ARM platforms (RPi2 and CHIP, "the 9$ computer" tested so far).

@flobotics It's great seeing people interested in ROS 2 and ARM-based platforms. We should get in touch.

We've released the ROS 2 (alpha 2) in our last Debian release http://forum.erlerobotics.com/t/erle-brain-2-debian-image-azul/449/4 so that people can start giving it a try.

gerkey, 2015-11-10:
As you start testing it, please file new issues as you run into problems.

closed 2015-11-10 by gerkey

flobotics, 2015-11-11:
hi,at the moment i am a little bit testing , dont do much with ARM specially. saw your erle bots, nice thing, you saw http://spectrum.ieee.org/automaton/robotics/robotics-software/disney-software-design-and-print-custom-walking-robots    perhaps interresting for

jeremy-a, 2016-01-26:
@flobotics Did you build the ROS 1 bridge? It's a heavyweight. I'd be surprised if the ROS 1 bridge built on the pi.

Did you cross compile for the pi or did you build ROS 2 right on your pi? If cross compiling, what was your setup.