ros2/ros2 #174: missing run dependency between message package with typesupport and rmw

tfoote, 2015-12-18:
Running my unit tests, there's a new instance of missing dylibs  The typesupport libraries in builtin_interfaces is failing to link against librmw via rpath. I've confirmed that rmw is not in the came__test.sh generated for the test. 

This happens for both connect and opensplice.

http://ci.ros2.org/job/ci_osx/669/testReport/junit/(root)/projectroot/test_simple/

http://ci.ros2.org/job/ci_osx/670/testReport/junit/(root)/projectroot/test_simple/

```
-- run_test.py: invoking following command in '/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2':
 - /Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_simple --gtest_output=xml:/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_results/tf2/test_simple.gtest.xml
dyld: Library not loaded: @rpath/librmw.dylib
  Referenced from: /Users/osrf/jenkins/workspace/ci_osx/ws/install/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_opensplice_cpp.dylib
  Reason: image not found
-- run_test.py: return code -5
-- run_test.py: generate result file '/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_results/tf2/test_simple.gtest.xml' with failed test
-- run_test.py: verify result file '/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_results/tf2/test_simple.gtest.xml'
```

gerkey, 2015-12-18:
CI job testing potential fix:

http://ci.ros2.org/job/ci_osx/676/

gerkey, 2015-12-18:
The latest attempt moved the dependency problem elsewhere:

~~~
-- run_test.py: invoking following command in '/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2':
 - /Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_static_cache_unittest --gtest_output=xml:/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_results/tf2/test_static_cache_unittest.gtest.xml
dyld: Library not loaded: @rpath/librosidl_typesupport_connext_cpp.dylib
  Referenced from: /Users/osrf/jenkins/workspace/ci_osx/ws/install/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_connext_cpp.dylib
  Reason: image not found
-- run_test.py: return code -5
-- run_test.py: generate result file '/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_results/tf2/test_static_cache_unittest.gtest.xml' with failed test
-- run_test.py: verify result file '/Users/osrf/jenkins/workspace/ci_osx/ws/build/tf2/test_results/tf2/test_static_cache_unittest.gtest.xml'
~~~

gerkey, 2015-12-18:
Testing by building with the ros2.repos file from #160.

wjwwood, 2015-12-18:
I added more dependencies to `rosidl_default_runtime` as a workaround, but that's not the final solution: https://github.com/ros2/rosidl/pull/58

I triggered a new OS X job to test it out: http://ci.ros2.org/job/ci_osx/678/

wjwwood, 2015-12-18:
Here is a new OS X CI job: http://ci.ros2.org/job/ci_osx/679/ for this fix which also requires (uses) http://ci.ros2.org/job/ci_osx/679/

wjwwood, 2015-12-18:
Link problems look to be gone, but there are other (normal) test failures.

dirk-thomas, 2016-06-22:
ros2/rosidl#158 removes all type support exports from `rosidl_default_runtime` which was only meant to be a temporary workaround. Can this be closed?

tfoote, 2018-03-08:
Linking has been restructured since then. I believe this is resolved now.

closed 2018-03-08 by tfoote