ros2/ros2 #178: Create an equivalent to ros::Time

tfoote, 2015-12-22:
This should provide a standard way to communicate simulated time and present it to the developer through the client libraries in as native a manner as possible.

tfoote, 2015-12-31:
Design document: https://github.com/ros2/design/pull/72

mikaelarguedas, 2017-04-05:
will be picked up in the near future for partial implementation needed for several already in progress tasks (links to be added)

tfoote, 2018-04-26:
I've ticketed the remaining features in ros2/rclpy#186 and ros2/rclcpp#465 for more specific callouts of what's necessary to close out this meta ticket.

sloretz, 2020-09-03:
Closing in favor of individual tickets.

It looks like there's only one remaining feature, and it has it's own issue https://github.com/ros2/rclcpp/issues/465 . This meta ticket also overlaps another time meta ticket #530

closed 2020-09-03 by sloretz