ros2/ros2 #179: Add instructions for building/installing ros2 on debian

firesurfer, 2016-01-05:
Hi,
I would really like to add instructions to the wiki about how to build ros2 on a debian system (as i use debian myself for developing). At the moment these are just some simple changes that need to made during the installation of the required libs. 
I'm not sure how you are handling edits to wiki at the moment or even if it's desired to have instructions about building ros2 on a system different to ubuntu 14.04.

vmayoral, 2016-01-05:
@firesurfer with regard installing ROS from its sources in Debian, feel free to grab the steps from http://erlerobotics.com/docs/Robot_Operating_System/ROS_2/Installing_ROS_2.html (tested with alpha-2).

firesurfer, 2016-01-05:
@vmayoral I already managed to install ROS2 from sources on Debian. My question was if it's okey to put these instructions to the ROS2 wiki hosted on github.
The instructions on the link you've posted are also kind of unnecessary complex. 
It's easier to use the OpenSplice Debian Packages built by OSRF because you dont need to compile them by yourself.

gerkey, 2016-01-05:
@firesurfer We'd love to have your Debian instructions added to the wiki. While the wiki is a git repo (https://github.com/ros2/ros2.wiki.git), GitHub doesn't seem to support PRs there, so I guess the easiest thing would be to provide a diff here.

firesurfer, 2016-01-06:
Hi, i just added a paragraph called Installation on different Linux than Ubuntu 14.04.
The Instructions do work for me on a Debian stretch with gcc 5.3 and a Debian jessie with gcc 4.9.
Two compiler/linker errors I had occuring during the compilation are submitted as issues/PRs in the ros2/rcl and ros/geometry_experimental git repos.

I attached my git diff of Linux-Development-Setup.md. 

[changes_debian.txt](https://github.com/ros2/ros2/files/80028/changes_debian.txt)

gerkey, 2016-01-11:
Added, thanks!

closed 2016-01-11 by gerkey