ros2/ros2 #181: How to use rosjava to use ros2?

jabrena, 2016-01-11:
Hi,

I would like to use Java to develop nodes for ROS2. How to use Rosjava with ROS2 tools to build a node?

Many thanks in advance.

Juan Antonio

dirk-thomas, 2016-01-11:
ROS 2 does currently not support a Java client library. The focus for now is on C++, C and Python (see https://github.com/ros2/ros2/wiki/Alpha3-Overview).

If you want to use Java you should continue to use ROS 1.

closed 2016-01-11 by dirk-thomas

tfoote, 2016-01-11:
Note there's a thread with more information on this here: https://groups.google.com/forum/#!topic/ros-sig-ng-ros/W7eSS-uacOk

jabrena, 2016-01-20:
Many thanks @tfoote I will follow it. I think that if we create a group, we could develop the support.
Cheers

Hyrtsi, 2024-10-15:
Hi, future here.

https://github.com/ros2-java/ros2_java