ros2/ros2 #195: Figure out how to resolve console_bridge installation path

tfoote, 2016-02-18:
When installed on linux the lib goes into a subdirectory such as `x86_64-linux-gnu` which means that it's not on the path. 

To be able to use it we need to explicitly extend the library path. For example we worked around it in tf2_ros here: https://github.com/ros/geometry_experimental/pull/157 

This ticket is to track this issue and hopefully push for a solution upstream or else find a generic fix instead of needing all packages to do the same as tf2_ros

dirk-thomas, 2016-02-18:
I think the workaround is the only viable one. It needs to happen in every package which needs console bridge since we can't / don't want to modify console bridge. As of Xenial we can rely on the system package and remove the workaround. Building console bridge from source in the same workspace won't be an option but that is likely true for any package with non standard paths.

traversaro, 2016-02-18:
Consider that this is a CMake issue that was solved in CMake 3.0 : 
https://cmake.org/pipermail/cmake-developers/2014-January/009435.html
https://github.com/Kitware/CMake/commit/620939e4e6f5a61cd5c0fac2704de4bfda0eb7ef . 
Given that this is a known CMake issue, a possible solution is  to modify the `console_bridge` build system to workaround this issue when using a version of CMake < 3.0 .

tfoote, 2016-02-18:
@traversaro That's great to see that it's fixed in CMake upstream. I agree that we should ask that console bridge integrate this workaround to support linux installs outside of `/usr`.

mikaelarguedas, 2017-09-14:
closed via https://github.com/ros2/geometry2/pull/47

closed 2017-09-14 by mikaelarguedas