ros2/ros2 #197: ros_comm dev. support

vmayoral, 2016-02-23:
Hello,

We recently have been using ROS2 actively and look forward to its adoption. Our results are promising however we've reached a point where `rostopic`, `rosnode` and similar tools would be extremely valuable things to have. These, i believe will be implemented in an similar abstraction to what `ros_comm` was for ROS.

I am aware you're extremely busy so in case it makes sense, we'd like to contribute and speed this up. @ahcorde and I have some bandwidth in the following weeks. How can we help?

esteve, 2016-02-23:
That's awesome! The `rclpy` API is still a bit crude, but it'd be interesting to see it being used for tools so we can find bugs in it and improve it.

Thanks.

dirk-thomas, 2016-02-23:
Any such command line tool needs intimate knowledge about the state of the communication graph. It needs to be able to query the currently available nodes, topics, etc. While the top of this will likely be Python scripts the missing piece and first thing to develop is on the C / C++ level and likely requires extending the rmw interface.

I think the first step would be to identify for each desired tools what information it requires and then come up with an API draft which can provide these. Then the API needs to be implemented down from rmw layer up to the Python client library.

gerkey, 2016-02-24:
Fantastic!

We talked about this in our weekly ROS 2 meeting today (which you could consider joining at some point). As @dirk-thomas said, we feel like the best place to start is to pick a specific piece of functionality, like `rostopic list` or `rostopic echo`, then go top to bottom implementing the necessary APIs at each level. Assuming that you do it in Python (which we'd recommend), you'll need to modify rclpy, rcl, and rmw (which implies modifying the various rmw implementations).

One specific thing to look out for is performance. As you know, DDS is peer-to-peer, and it can take a non-trivial amount of time after you start listening to hear about all the topics (or all the nodes). To give the chance for a tool like `rostopic list` to be responsive, we're expecting to need to support at the lower levels at least 3 ways of getting the topic list: (i) the generic way: create a listener and wait to hear; (ii) an rmw-implementation-specific call get its internally cached list (Connext supports this); and (iii) run our own daemon that just sits there listening and will respond to service calls to give the accumulated information to others. Hopefully that's not too confusing. Just ask if you have any questions.

p.s. We took the liberty of writing you into our Alpha 5 release ticket #198. Pressure's on :)

jacquelinekay, 2016-02-24:
Should we schedule a quick call to discuss the technical details in the next few days? @vmayoral, can you email me with your availability?

vmayoral, 2016-02-24:
Thanks everyone, sounds good. 
@jacquelinekay let's do it. I'll PM you to schedule a call.

vmayoral, 2016-03-01:
Here's a quick update of what we've been up to:
- Reviewed DDS, rmw and rcl
- Reviewed previous ROS `ros_comm` package code
- Targetting `rosnode list` for starters. Here's the API we'd like to implement to provide this functionality:

##### rosnode package (user land?)
```
rosnode_listnodes(namespace=None, list_uri=False, list_all=False):
```
##### ROS 2 client library (rcl)
```
get_node_names()
```

##### ROS 2 middleware interface (rmw)
```
## Option 1: Built-in topics (validated with ros_dds prototype at https://github.com/erlerobot/ros_dds/commit/9a248ed7bebb44f521cff8ef6752dd2e4b62f182) 
get_builtin_subscriber();
builtinSubscriber->lookup_datareader("DCPSParticipant");

# Option 2
get_discovered_participants()
get_discovered_participant_data()
```

Feedback is welcome. Here's what we plan on doing next:
- [ ] Get a better understanding of `rcl` and `rclc`. Hopefully that'll get address in today's call.
- [ ] Implement the API starting at the `rmw` level targetting FastRTPS and OpenSplice implementations (we got in touch with RTI for a license but apparently they denied it so we'll leave a stub)
- Go up implementing the API

vmayoral, 2016-03-02:
Hi @dirk-thomas, here's an attempt the make our intentions more clear (we've removed the parameters in the rclpy layer for simplicity, to be added in further iterations):

#### `rosnode list` tentative API

##### ROS 2 client library (rclpy)
```
string[] rosnode_listnodes(void):
```

##### ROS 2 client library (rcl)
```
char** get_node_names(void)
```

##### ROS 2 middleware interface (rmw)
```

rmw_nodes_t* get_nodes(void)
    # which will be using Built-in types, mainly:
    Subscriber_ptr get_builtin_subscriber(void); # DDS primitive
    DataReader_ptr lookup_datareader ("DCPSParticipant"); # DDS primitive
```
As for the ownership of the returned data, it'll be the upper layer the one in charge of freeing those resources.

dirk-thomas, 2016-03-02:
The return type `char**` doesn't clearly state how many nice names have been returned. Same for `rmw_nodes_t` which afaik is currently not a type.

Regarding the ownership: I don't think it is a good idea to separate allocation and deallocation - especially across library boundaries (https://blogs.msdn.microsoft.com/oldnewthing/20060915-04/?p=29723/).

vmayoral, 2016-03-02:
Thanks, for the quick feedback. 

>The return type char** doesn't clearly state how many nice names have been returned.

We can work a type for that. Should we use something like:
```
typedef struct RMW_PUBLIC_TYPE rmw_nodes_name_t
{
  size_t names_count;
  char ** nodes_name;
} rmw_nodes_name_t;

```
or would you prefer something more generic such as `rmw_strings_t`?

>Same for rmw_nodes_t which afaik is currently not a type.

Yeap, that's one we defined following the coding style:
https://github.com/erlerobot/rmw/commit/d7a5440b294e850b023f7a097f05fcf0750ae0ef

For the middleware, how about:
##### ROS 2 middleware interface (rmw)
```
rmw_ret_t rmw_destroy_nodes(rmw_nodes_t * node)
rmw_nodes_t* get_nodes(void)
    # which will be using Built-in types, mainly:
    Subscriber_ptr get_builtin_subscriber(void); # DDS primitive
    DataReader_ptr lookup_datareader ("DCPSParticipant"); # DDS primitive
```

dirk-thomas, 2016-03-02:
Since we will likely have several of these function all returning a set of names `rmw_strings_t` might be a good idea.

The rmw interface shouldn't return `rmw_node_t`s since I don't think the returned value can be used with any of the other API having `rmw_node_t` as an argument. Wouldn't the rmw API look the same as the rcl one: `bool get_node_names(rmw_strings_t * node_names)` and `bool destroy_strings(rmw_strings_t * strings)`?

wjwwood, 2016-03-02:
Another thing to keep in mind is that we will be adding allocator functions to the `rmw` functions as many of the `rcl` functions already have, e.g.: https://github.com/ros2/rcl/blob/master/rcl/include/rcl/node.h#L58

So it might be good to go ahead and include them in the design of your rcl functions even if they remain unpassed/unused in the `rmw` API for now. Also the other `rcl` functions all the top level structure to be allocated by the user.
Adjusting for the style of other parts of `rcl` this is what I would imagine the `rcl` portion of the interface should look like:

```C
typedef struct rcl_strings_t
{
  size_t count;
  char ** data;
} rcl_strings_t;

rcl_strings_t
rcl_get_zero_initialized_strings()
{
  static rcl_strings_t null_strings = {0, 0};
  return null_strings;
}

rcl_ret_t
rcl_get_node_names(rcl_strings_t * node_names, rcl_allocator_t allocator);

rcl_ret_t
rcl_strings_fini(rcl_strings_t * strings);

// Example:
rcl_strings_t node_names;
rcl_ret_t ret = rcl_get_node_names(&node_names, rcl_get_default_allocator());
if (ret != RCL_RET_OK) {
  // Error handling...
}
// Use the node names...
ret = rcl_strings_fini(&node_names);
```

That's just off the cuff, so there may be some issues with it, but hopefully that gives you some more information to start with.

dirk-thomas, 2016-03-02:
Just for information I added a comment about the language decision here: https://github.com/ros2/ros2/issues/198#issuecomment-191351754

vmayoral, 2016-03-04:
Wrapping the recommendations provided above, here's the API status:

#### `rosnode list` tentative API version 2

##### ROS 2 client library (rcl)
```
rcl_strings_t
rcl_get_zero_initialized_strings()

rcl_ret_t
rcl_get_node_names(rcl_strings_t * node_names, rcl_allocator_t allocator);

rcl_ret_t
rcl_strings_fini(rcl_strings_t * strings);
```

##### ROS 2 middleware interface (rmw)
```
rmw_strings_t * 
rmw_get_node_names(void);

bool 
rmw_destroy_strings(rmw_strings_t * strings);
```

We've put up a minimal prototype to validate the discovery of other DDS participants using [ros2-erle.repos](https://gist.github.com/vmayoral/ca523ec7de46ece420b2) (not meant to be integrated, not all parts of the API have been implemented). This helped to start building the skeleton of the code and explore some of the design choices.

Our next objective is to implement a way for ROS nodes to request/send their node name to other participants of the DDS network. As far as we understand, the "node name" is a ROS abstraction coded at the rmw level (and up to the rcl) that needs to be sent over the network as a payload.
Given this assumption, we believe that the simplest way to do so is to make each rmw node to get a publisher and a subscriber to a topic dedicated to share some specific ROS information (for now we need the node name but more fields will show up as we implement more tools, e.g.: "ros_meta"?). This will likely require us to hack [rmw_create_node](https://github.com/erlerobot/rmw_opensplice/blob/master/rmw_opensplice_cpp/src/rmw_node.cpp#L31).

What's your opinion about this strategy? We looked at the DDS API (OpenSplice) looking for DDS primitives that could help us solve this matter in a different way but didn't find anything convincing. Any suggestions?

vmayoral, 2016-03-07:
We finally got a `rosnode list` prototype for ROS 2 using OpenSplice API. We evaluated 2 implementation strategies, both trying to follow the API described above but only the second one allowed us to complete a prototype: 
- Using Built-in types
- Creating a DDS `ros_meta` topic

#### `rosnode list` using Built-in types
We initially believed that we could fetch the ROS node name of a DDS participant using Built-in types however we couldn't find any primitives (with OpenSplice) to do so. The only thing we could fetch was a list of participants, each with its associated ID that had to be mapped to its ROS node name. This, as far as we understand makes sense due to the fact that the ROS node name is an non-DDS abstraction and led us to consider using a dedicated DDS topic.

Built-in types can probably help at identifying the topics that each participant is subscribed to but we'll still need to find a way to map participants with its ROS node names.

#### `rosnode list` using a DDS `ros_meta` topic
This implementation strategy makes each node created though the `rmw_create_node` function launch a thread that publishes (once per second) its ROS node name to a `ros_meta` topic (string messages for now). 
The `rmw_get_node_names` listens to this DDS topic and pushes the values  up to the rcl layer.

wjwwood, 2016-03-08:
That sounds reasonable to me, having each node publish a mapping from its implementation specific middleware ID (in this case the DDS participant ID) and the node's name. Then we can associate publishers and subscriptions with the corresponding nodes. A few notes though:

- Can you use durability (https://github.com/ros2/rmw/blob/master/rmw/include/rmw/types.h#L135-L140) to acheive the desired "latching" effect without having a thread that publishes every second?
- For examples of how to generically represent a globally unique ID (which is backed by a DDS specific GUID), see: 
  - https://github.com/ros2/rmw/blob/master/rmw/include/rmw/types.h#L34-L36
  - https://github.com/ros2/rmw/blob/master/rmw/include/rmw/types.h#L157-L161
  - https://github.com/ros2/rmw/blob/master/rmw/include/rmw/types.h#L166
  - https://github.com/ros2/rmw/blob/master/rmw/include/rmw/rmw.h#L248-L256
  - And their corresponding implementations in the various rmw implementations.

Thanks for working on this, I think it is headed in the right direction!

vmayoral, 2016-03-14:
Thanks @wjwwood for the pointers. We exchanged the previous threaded implementation to one that uses the durability property. It made things much simpler and we are happy with the result.

Here's our current result (listener and talker launched previously):
```
victor@ubuntu:~/ros2_ws$ rosnode_py list
talker
listener
victor@ubuntu:~/ros2_ws$ time rosnode_py list
listener
talker

real	0m1.857s
user	0m0.676s
sys	0m0.044s

```
The timing with rclpy seems to be between 1 to 2 seconds and clearly slower than using the primitives with rclcpp.

#### Implementation

From the 3 alternatives mentioned above by @gerkey: 
- (i) the generic way: create a listener and wait to hear; 
- (ii) an rmw-implementation-specific call get its internally cached list (Connext supports this); and 
- (iii) run our own daemon that just sits there listening and will respond to service calls to give the accumulated information to others.

Only the first option has been implemented using a 1 milisecond timer (we tested about 15-20 nodes and still got all of them every single time) https://github.com/erlerobot/rmw_opensplice/blob/master/rmw_opensplice_cpp/src/rmw_node.cpp#L511. Code is in good condition to try to do something about (iii).

We've assembled all the code at http://paste.ubuntu.com/15383463/ in case someone wants to reproduce it.

#### Support for other middlewares
We've based our development in OpenSplice. For now, this is the only middleware with support for this prototype. We also looked briefly at FastRTPS but got type support issues at https://github.com/erlerobot/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/functions.cpp#L287. It could be that there's an error in our CMakeList.txt (we've used one similar to the OpenSplice) or that there's a bug somewhere else.

#### Next steps
We've got a bit of additional time this week so we could either a) try to finish this tool by implementing more alternatives as suggested above or b) explore `rostopic list` and build a prototype for this tool. Do you have any preference?

jacquelinekay, 2016-03-16:
hey @vmayoral,

sorry for coming late to the party, but I have some comments:

My general feeling for the design/implementation on `rmw_get_node_names` is that it breaks encapsulation. I don't think that this function as it was written should be in rmw. It could be implemented using rclpy or rclcpp. I don't think that the busy loop in rmw_get_node_names is a good idea; instead you can re-use the code in rclcpp (or even rcl...) to accomplish the same thing.

For example, using rclcpp it would look something like this (no promises that this will compile):

```
auto node = std::make_shared<Node>("get_node_names");
// ... make a qos_profile with the right parameters
auto sub = node->create_subscription<introspection_msgs::ROSMeta>("ros_meta", qos_profile, callback);

spin(node);  //Or do a while loop with spin_once and a 1ms wait
```

then in the subscription callback you should have code that effectively does what the while loop in your current implementation does, and you should have some way of storing the ros_meta state.

similarly I'm pretty sure `create_publisher_ros_meta` in rmw could be implemented simply as a publisher of type `introspection_msgs::ROSMeta` in rclcpp.

Where do you think the bottleneck is in your current implementation? 1-2 seconds is slower than we would like for the final implementation.

Why did you choose pub/sub instead of service/client? Is it because services are not implemented for Python yet or do you think pub/sub is better for this case?

FastRTPS is not supported in Python yet (it's in the pipeline), that's our bad. Sorry that you went down a false path!

Let me know if you would like to schedule another call to ask questions.

dirk-thomas, 2016-03-16:
The disadvantage of moving the implementation into rcl* is that if a specific rmw impl. could provide the information more efficiently it has no way to provide a "better implementation". For `rosnode list` where the name of a node is a ROS specific thing that might be not a problem. But for e.g. `rostopic list` I think this will be an issue since e.g. Connext will provide an API to query that information directly (as far as I remember).

vmayoral, 2016-03-20:
Thanks @jacquelinekay and @dirk-thomas for the feedback.

With regard the `rcl` vs `rmw` dicussion, We evaluated both alternatives actually. After a few prototypes found that coding the rosnode logic in `rcl` was actually more comfortable (and possibly more intuitive) however it didn't gave us fine grain access to the middleware layer, which at the time, we thought it could be needed. Steered by the comments we decided to go down to `rmw`. I'm happy we did that because it helped us gaining a good understanding however, I personally don't have a strong opinion in which should be used for a final implementation. We also found relevant differences in the DURABILITY implementations among OpenSplice and FastRTPS which leads us to think that there's a tradeoff here between maintenance effort and performance.

>Where do you think the bottleneck is in your current implementation? 1-2 seconds is slower than we would like for the final implementation.

I did a bit of testing on our `rosnode list` implementations and the timings. Find below the output of three consecutive runs of `rosnode list` command with 3 nodes running (*time is presented in a cumulative way on each layer*).
```
victor@ubuntu:~/ros2_ws$ time rosnode list
(rclpy)-----> Option parsing: 0.00016900000000000248
(rcl) -----> check_args:  0.00000 
(rmw_opensplice_cpp) -----> env. variables stuff: 7e-06
(rmw_opensplice_cpp) -----> rmw_create_node: 0.078616
(rmw_opensplice_cpp) -----> rmw_create_subscription: 0.07974
(rmw_opensplice_cpp) -----> introspection_msgs__msg__ROSMeta__init: 0.079782
(rmw_opensplice_cpp) -----> finished while loop: 0.364329
(rmw_opensplice_cpp) -----> freeing memory: 0.365937
(rcl) -----> rmw_get_node_names: 1.50562 
(rcl) -----> allocation and init: 1.50563 
(rcl) -----> filled the structures: 1.50564 
(rcl) -----> destroy: 1.50565 
(rclpy)-----> rclpy.get_node_names: 0.366225
rostopic
listener
talker

real	0m1.659s
user	0m0.476s
sys	0m0.040s
victor@ubuntu:~/ros2_ws$ time rosnode list
(rclpy)-----> Option parsing: 0.00016900000000000248
(rcl) -----> check_args:  0.00000 
(rmw_opensplice_cpp) -----> env. variables stuff: 7e-06
(rmw_opensplice_cpp) -----> rmw_create_node: 0.081235
(rmw_opensplice_cpp) -----> rmw_create_subscription: 0.083286
(rmw_opensplice_cpp) -----> introspection_msgs__msg__ROSMeta__init: 0.083535
(rmw_opensplice_cpp) -----> finished while loop: 0.176523
(rmw_opensplice_cpp) -----> freeing memory: 0.178632
(rcl) -----> rmw_get_node_names: 1.31824 
(rcl) -----> allocation and init: 1.31826 
(rcl) -----> filled the structures: 1.31827 
(rcl) -----> destroy: 1.31828 
(rclpy)-----> rclpy.get_node_names: 0.17894600000000002
rostopic
listener
talker

real	0m1.480s
user	0m0.272s
sys	0m0.064s
victor@ubuntu:~/ros2_ws$ time rosnode list
(rclpy)-----> Option parsing: 0.00017099999999997673
(rcl) -----> check_args:  0.00000 
(rmw_opensplice_cpp) -----> env. variables stuff: 7e-06
(rmw_opensplice_cpp) -----> rmw_create_node: 0.083868
(rmw_opensplice_cpp) -----> rmw_create_subscription: 0.084985
(rmw_opensplice_cpp) -----> introspection_msgs__msg__ROSMeta__init: 0.085018
(rmw_opensplice_cpp) -----> finished while loop: 0.530038
(rmw_opensplice_cpp) -----> freeing memory: 0.532068
(rcl) -----> rmw_get_node_names: 1.66018 
(rcl) -----> allocation and init: 1.66019 
(rcl) -----> filled the structures: 1.66035 
(rcl) -----> destroy: 1.66037 
(rclpy)-----> rclpy.get_node_names: 0.532455
talker
rostopic
listener

real	0m1.816s
user	0m0.656s
sys	0m0.028s

```

Used the following branches for the tests:
- https://github.com/erlerobot/rmw_opensplice/tree/timetest
- https://github.com/erlerobot/rcl/tree/timetest
- https://github.com/erlerobot/rclpy/tree/timetest

@jacquelinekay my believe is that the bottleneck is somewhere between the `rmw` and `rcl`. Looking at the previous tests:
```
(rcl) -----> check_args:  0.00000 
(rmw_opensplice_cpp) -----> env. variables stuff: 7e-06
(rmw_opensplice_cpp) -----> rmw_create_node: 0.078616
(rmw_opensplice_cpp) -----> rmw_create_subscription: 0.07974
(rmw_opensplice_cpp) -----> introspection_msgs__msg__ROSMeta__init: 0.079782
(rmw_opensplice_cpp) -----> finished while loop: 0.364329
(rmw_opensplice_cpp) -----> freeing memory: 0.365937
(rcl) -----> rmw_get_node_names: 1.50562 
```
The overall `get_node_names()` in rmw https://github.com/erlerobot/rmw_opensplice/blob/timetest/rmw_opensplice_cpp/src/rmw_node.cpp#L393-L527 takes about **0.36 seconds** however when looking at this function call from above (rcl), it tells us that takes **1.50562 seconds** https://github.com/erlerobot/rcl/blob/timetest/rcl/src/rcl/node.c#L78. Not sure if measurements can be performed this way but if so, there's a mismatch. Each of the runs presented above, shows something similar.

jacquelinekay, 2016-03-21:
Random thought that occurred to me while looking at your timing data: what optimization level are you compiling the code with?

I find it strange that there's such a huge gap between the time taken inside of `rmw_get_node_names` and the time directly after returning from the function. It could be related to the way Opensplice does memory management. Opensplice has a garbage collector running in a separate thread, which I don't know much about, but I think it's plausible that the garbage collection events might not happen directly after calling destroying to Opensplice representation of the node, but are instead scheduled on the next context switch.

To confirm this theory, you would have to separate out `get_node_names` so that initialization and destruction of the node and subscriber happen in a separate context, and pointers to those entities are passed to the `rmw_get_node_names` function. And if my theory about the Opensplice garbage collector is wrong, I still think separating initialization, execution, and destruction is a pattern we want to follow throughout ROS 2. If `rmw_get_node_names` were to be called multiple times in one process, it would be more efficient to be able to reuse the same node and subscriber.

Also, I think there's a more direct way to see if the garbage collector is messing with you by increasing the verbosity level in your ospl.xml file. You can also turn down the frequency of garbage collection events apparently:
http://download.prismtech.com/docs/Vortex/html/ospl/DeploymentGuide/guide.html#garbagecollector

Regarding QoS settings, if you do need access to certain QoS policies to get the behavior you want, you can look into making a pull request to expose more QoS settings to the client library layer. If this is something you're interested in I could give you more advice on the path to go for that.

vmayoral, 2016-03-29:
@jacquelinekay the default optimization options were used for the tests above so that could indeed be a good guess. Thanks the pointers.  We're back on tight schedules and we'll have limited time so I've gone ahead and submitted our contributions so far:
- https://github.com/ros2/rmw_opensplice/pull/125
- https://github.com/ros2/rmw/pull/60
- https://github.com/ros2/rclpy/pull/9
- https://github.com/ros2/rcl/pull/34
- https://github.com/ros2/common_interfaces/pull/12
- https://github.com/ros2/examples/pull/97

Those should help getting the functionality into alpha 5 (https://github.com/ros2/ros2/issues/198).

wjwwood, 2016-03-30:
@vmayoral I skimmed over the pr's and I found a few places where we'll need to move code around to fit our architectural goals, but I think we can probably use most of the code contributed and we'll certainly try to do so. Since we'll need to move some code around, fix stuff to make the linters happy, write tests for the new API elements, and fill out more rmw implementations, I think we should set the expectation that not all of this will make it into alpha 5. I'm going to try and get a few of the low level interfaces proposed for alpha 5, but even that might not happen by the end of the week. But that's ok, we can continue to work on it and push it to later releases if needed.

Thanks for getting us started on this and thanks again for the contribution!

vmayoral, 2016-04-03:
Thanks for having a look at it @wjwwood.
We'll stay tuned to alpha 5 and try to accommodate the code based on everyone's comments.

vmayoral, 2016-06-30:
Hello again everyone, here's an update:
- We've successfully ported most of our code (including basic rostopic and rosnode functionality) to FastRTPS
- Results are much better (timing-wise) when compared to what we shared before. This is mainly caused because instead of creating our own DDS participants at the `rmw` level, we're now using primitives from the DDS implementation. With this, we get outputs from tools like `rosnode list` or `rostopic list` under 500 ms. And on top of that we still believe that there's room for improvement.

Due to our existing deadlines, we've been pushing the tools slower than we'd have liked  but we expect to have some bandwidth by the end of July to rebase the code and implement the remaining functionalities.

wjwwood, 2016-07-05:
@vmayoral Thanks for the update.

I'll have to re-review these changes in the context of the new stuff I added to support `wait_for_service`. For example, event driven changes to the graph are notified through the `GraphListener` interface in `rclcpp`, which is new. I don't know if those changes will affect what you're doing here without looking in more detail.

vmayoral, 2016-08-03:
We've reviewed the tools once again at this branch https://github.com/erlerobot/rclpy/commits/master-alpha7. 

As it is, here're the tools available:
```
victor@ubuntu:~/hros_ws$ rosnode -h
rosnode is a command-line tool for printing information about ROS Nodes.
Commands:
	rosnode ping	test connectivity to node (NOT IMPLEMENTED)
	rosnode list	list active nodes
	rosnode info	print information about node (NOT IMPLEMENTED)
	rosnode machine	list nodes running on a particular machine or list machines
	rosnode kill	kill a running node (NOT IMPLEMENTED)
	rosnode cleanup	purge registration information of unreachable nodes (NOT IMPLEMENTED)
Type rosnode <command> -h for more detailed usage, e.g. 'rosnode ping -h'

victor@ubuntu:~/hros_ws$ rostopic -h
rostopic is a command-line tool for printing information about ROS Topics.
Commands:
	rostopic bw	display bandwidth used by topic (NOT IMPLEMENTED)
	rostopic delay	display delay of topic from timestamp in header (NOT IMPLEMENTED)
	rostopic echo	print messages to screen (NOT IMPLEMENTED)
	rostopic find	find topics by type (NOT IMPLEMENTED)
	rostopic hz	display publishing rate of topic (NOT IMPLEMENTED)
	rostopic info	print information about active topic (NOT IMPLEMENTED)
	rostopic list	list active topics
	rostopic pub	publish data to topic (NOT IMPLEMENTED)
	rostopic type	print topic type
Type rostopic <command> -h for more detailed usage, e.g. 'rostopic echo -h'

```

We've got prototypes of `rostopic echo` and `rostopic pub` with OpenSplice but we won't be pushing this forward and will instead wait until there's FastRTPS support in rclpy. We'll need to adapt some abstractions in the rmw and rcl to the [graph API](https://github.com/ros2/rcl/blob/master/rcl/src/rcl/graph.c) that @wjwwood  created but it seems manageable.

Karsten1987, 2016-10-20:
I was recently playing around with Fast-RTPS and I didn't easily find a way to get a list of current participants out-of-the-box. Thus, I opened this PR: https://github.com/eProsima/Fast-RTPS/pull/68

This would then enable some `rosnode list` and `rostopic list` functionality: https://github.com/Karsten1987/fastrtps_introspection/blob/master/src/fastrtps_info.cpp

I'll just drop this here knowing that this is far away from being any API, more like sharing some ideas.

mkhansen-intel, 2017-02-02:
@vmayoral - What's the status of the rosnode and rostopic tools? I see there was a branch in your comment above, but doesn't look like it was ever merged, is it abandoned?

vmayoral, 2017-02-02:
@mkhansen-intel, latest trial of getting it integrated was described at https://discourse.ros.org/t/ros-graph-information-tools-implementation-discussion/674/2?u=vmayoral. 

There also currently some nice work being done by Karsten at https://github.com/ros2/rcl/pull/108

dirk-thomas, 2017-06-14:
The new repository https://github.com/ros2/ros2cli contains the latest Python command line tools. They provide almost all the commands / verbs known from ROS 1 (as far as they are available in `rclpy`, e.g. services and parameters are still missig features). The tools use a common infrastructure and utilze a daemon process (transparently forked in the background to improve the responsiveness). Therefore I will close this ticket. Please feel free to continue commenting here.

closed 2017-06-14 by dirk-thomas