ros2/ros2 #198: Alpha 5 release (2016-04-01)

tfoote, 2016-02-23:
* [ ] Fix all test failures
* [ ] Add support for C data structures in Fast and Connext Dynamic rmw implementations @wjwwood 
  * Consider ticketing eProsima to do the FastRTPS implementation.
* [ ] FreeRTPS improvements: RMW support @codebot and @jacquelinekay
  * [ ] rmw_take()
  * [ ] Tie back into rmw (1 weeks)
  * [ ] Testing/CI for linux case (1 weeks)
    * [ ] Add testing and test reporting to the existing CI job
* [ ] Add doc job at ci.ros2.org (call doxygen, sphinx, whatever)
* [ ] Add rostime sources
* [ ] Get first version of rostopic-like functionality, including some graph API, identifying rclpy improvements along the way.
* [ ] rosconsole-like replacement for printf (not necessarily wire transport version or other outputs)
* [ ] Support services in C (1.5 weeks)
  * [ ] Support parameters in C (2 weeks)
    * [ ] Minimal implementation of component life cycle in C.
* [ ] Refactor rclcpp to use rcl (2 weeks)
  * [ ] Refactor rmw to be more like rcl (3 weeks)
* [ ] Refactor type support to have less duplication and easier reuse

Stretch goals
* [ ] Component life-cycle in C++ as defined in the design article (with ROS-agnostic API) (2 weeks)
  * [ ] Introspection and orchestration APIs as ROS topics / services (2 weeks)
    * [ ] Use the orchestration API in launch to make tests deterministic (1 week)
  * [ ] Use C++ program to demonstrate life-cycle and orchestration in a tutorial/demo (1 week)
  * [ ] Use class_loader / pluginlib (1.5 weeks)

dirk-thomas, 2016-03-02:
While the choice to use Python for most of the command line tools I think we should reconsider this decision. Based on experience with ROS 1 I think certain commands needs to be C++ since the startup time of a Python script might already incur a too high latency when the result is e.g. being used for auto completion. E.g. rospack providing package names and files within packages is written in C++ for exactly that reason.

wjwwood, 2016-03-02:
I agree, but I think it would be useful to start in Python (capability should be there anyways) and actually take time to profile CLI tools afterwards. For slow tools there are lots of things you can do to speed up Python (avoiding unnecessary imports for example), and if that is not enough you can rewrite in C++. We should be wary of premature optimization.

dirk-thomas, 2016-03-02:
I was less referring to the actual performance of the implementation but the inherent startup time of Python which has been shown to be a significant problem for auto completion (even if the tool itself would run in no time). Therefore I don't think this classifies as premature optimization but considering for which cases we already foresee that a Python based implementation will not meet the requirements of the intended use cases.

wjwwood, 2016-03-02:
Depending on the use of the tool for tab completion the startup time of Python shouldn't be a real issue (on the order of 40 ms and on my machine rospack's is around 20ms). So, I think it's possible to use Python in tab completion if you're careful.

For example, listing directories from an environment variable (a likely task for tab completion) is only 37ms:

```
% time python3 -c "import os; print(['{0}: {1} items'.format(x, len(os.listdir(x))) for x in os.environ['PATH'].split(':') if x])"
['/Users/william/indigo/devel/bin: 52 items', '/usr/local/opt/ccache/libexec: 44 items', '/usr/local/sbin: 8 items', '/usr/local/bin: 638 items', '/Users/william/.bin: 1 items', '/usr/bin: 1060 items', '/bin: 37 items', '/usr/sbin: 241 items', '/sbin: 57 items', '/usr/local/bin: 638 items']
python3 -c   0.03s user 0.01s system 91% cpu 0.037 total
```

If I use `-S` to avoid unnecessary elements in the `PYTHONPATH` it goes down even further:

```
% time python3 -S -c "import os; print(['{0}: {1} items'.format(x, len(os.listdir(x))) for x in os.environ['PATH'].split(':') if x])"
['/Users/william/indigo/devel/bin: 52 items', '/usr/local/opt/ccache/libexec: 44 items', '/usr/local/sbin: 8 items', '/usr/local/bin: 638 items', '/Users/william/.bin: 1 items', '/usr/bin: 1060 items', '/bin: 37 items', '/usr/sbin: 241 items', '/sbin: 57 items', '/usr/local/bin: 638 items']
python3 -S -c   0.01s user 0.01s system 86% cpu 0.023 total
```

23ms is almost as fast as `rospack` (obviously it's doing less work, but hopefully moving away from RPP will address that).

BTW, Python2 is a good deal faster, maybe Python3's performance will improve:

```
% time python -S -c "import os; print(['{0}: {1} items'.format(x, len(os.listdir(x))) for x in os.environ['PATH'].split(':') if x])"
['/Users/william/indigo/devel/bin: 52 items', '/usr/local/opt/ccache/libexec: 44 items', '/usr/local/sbin: 8 items', '/usr/local/bin: 638 items', '/Users/william/.bin: 1 items', '/usr/bin: 1060 items', '/bin: 37 items', '/usr/sbin: 241 items', '/sbin: 57 items', '/usr/local/bin: 638 items']
python -S -c   0.01s user 0.01s system 83% cpu 0.016 total
```

So, it's not clear cut to me that starting in C++ for any tools make sense, especially when weighed against the difficulty of implementing a nice to use command line tool without `argparse`. I do agree both should be possible to implement based on the available functionality in Python and C++.

closed 2016-04-06 by gerkey