ros2/ros2 #199: A possible bug in ROS2-alpha4 release

RameezI, 2016-02-27:
The alpha4 release is not able to publish image messages. The publish method throws a run time exception with a message of vector overflow. The exception can be reproduced by running the  Image_Pipeline_all_in_one  demo.

wjwwood, 2016-02-27:
Hi @RameezI, thanks for trying the alpha out and reporting the issue. I just tried the binaries for OS X and Linux and I didn't get the error. Can you tell me what OS you're using and how you have OpenCV installed (from where and what version)?

Also, it's not 100% clear that you're using the binaries, and if you're not (you built from source) it would be useful to know which middleware you're targeting (OpenSplice, Connext, or FastRTPS).

madratman, 2016-02-27:
Hi @wjwwood I just checked this. Got the same. Ubuntu 14.04 + Opensplice, compiled from source
```
$ image_pipeline_all_in_one
VIDIOC_QUERYMENU: Invalid argument
VIDIOC_QUERYMENU: Invalid argument
VIDIOC_QUERYMENU: Invalid argument
VIDIOC_QUERYMENU: Invalid argument
VIDIOC_QUERYMENU: Invalid argument
VIDIOC_QUERYMENU: Invalid argument
vector overcomes the maximum length
terminate called after throwing an instance of 'std::runtime_error'
  what():  vector overcomes the maximum length
Aborted (core dumped)
```

RameezI, 2016-02-27:
I am on Ubuntu 14.04 and compiled alpha4 from the source. I am using  Opencv-2.4.11 installed from source using cmake the target middleware is 'OpenSplice'. 
Furthermore, I tested alpha3 release afterwards with the same configuration and it worked fine for me.

wjwwood, 2016-02-28:
I think the `VIDIOC_QUERYMENU: Invalid argument` messages are a red herring.

The `vector overcomes the maximum length` error comes from FastRTPS:

https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/0a50bfb0005074dc0e3361a31f41ed5cedd62633/rmw_fastrtps_cpp/src/TypeSupport.cpp#L181

I think this is because it does not support fragmentation yet, and so for very large messages (like images) it cannot send the data all at once without fragmentation. eProsima is working on this feature still. So image demos won't work with their library for now.

There might be an underlying linking issue with our build system which is causing your binaries to use FastRTPS rather than OpenSplice. Is there a `image_pipeline_all_in_one__rmw_opensplice_cpp` you guys could try?

I can try later tonight when I have a Linux machine handy.

madratman, 2016-02-28:
You mean a opensplice specific demo of image_pipeline_all_in_one? I didn't see any. Or something else?
If you can point me to the possible linking issue, I wouldn't mind trying to fix it.

RameezI, 2016-02-28:
@wjwwood 
 I just checked the `image_pipeline_all_in_one__rmw_opensplice_cpp` for the alpha4 version and it works perfect for me!

wjwwood, 2016-02-29:
@RameezI that's good to hear. I'm gonna close this since it's a known limitation and the linking issue (if that's in play here) has already been ticketed and fixed by @esteve: https://github.com/ros2/examples/issues/91

The only thing I think we could improve is the error message in this case, but it shouldn't be a permanent limitation.

closed 2016-02-29 by wjwwood