ros2/ros2 #203: Cross-Compile to ARM fails for rcl_interfaces and example_interfaces

bbaillio, 2016-03-10:
When trying to use a cross-compile toolchain, there are errors that prevent a successful compile.  I even tried to pare down which packages to compile to skirt some of the issues, but I can only get so far before critical packages won't compile.  The most common failure I see is this:
**RCL_INTERFACES** package
[ 31%] Generating C++ type support for PrismTech OpenSplice
/lib/ld-linux.so.3: No such file or directory
Traceback (most recent call last):
  File "/home/perception/alpha3_ros2_ws/install/lib/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp", line 60, in <module>
    sys.exit(main())
  File "/home/perception/alpha3_ros2_ws/install/lib/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp", line 55, in main
    args.idl_pp,
  File "/home/perception/alpha3_ros2_ws/install/lib/python3.4/site-packages/rosidl_typesupport_opensplice_cpp/__init__.py", line 72, in generate_dds_opensplice_cpp
    subprocess.check_call(cmd)
  File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/opt/opensplice/bin/idlpp', '-I', '/home/perception/alpha3_ros2_ws/build/rcl_interfaces/rosidl_generator_dds_idl', '-I', '/opt/opensplice/etc/idlpp', '-S', '-l', 'cpp', '-d', '/home/perception/alpha3_ros2_ws/build/rcl_interfaces/rosidl_typesupport_opensplice_cpp/rcl_interfaces/msg/dds_opensplice', '/home/perception/alpha3_ros2_ws/build/rcl_interfaces/rosidl_generator_dds_idl/rcl_interfaces/msg/dds_opensplice/IntraProcessMessage_.idl']' returned non-zero exit status 255
make[2]: *** [rosidl_typesupport_opensplice_cpp/rcl_interfaces/msg/dds_opensplice/intra_process_message__type_support.hpp] Error 1
make[1]: *** [CMakeFiles/rcl_interfaces__rosidl_typesupport_opensplice_cpp.dir/all] Error 2
make: *** [all] Error 2

wjwwood, 2016-03-10:
Based on that my guess is that the `idlpp` program which is provided by OpenSplice is not running because it cannot find the `ld-linux.so` shared object that it was linked against when built. If that file exists, you might just try adding it's folder to the `LD_LIBRARY_PATH`. I'm not 100% sure though. You can also try running the program `/opt/opensplice/bin/idlpp` directly and see what happens.

bbaillio, 2016-03-10:
Ok.  Isn't the idlpp file cross-compiled, or natively compiled for ARM?  Wouldn't that be part of the compile process of generating an ARM build of OpenSplice to link against?
So then, running idlpp on an x86 would never work, even if the libraries are there.

I'll try and get the missing libraries linked up, but I'm dubious it'll work.  Unless I'm flat-out wrong and it isn't something that gets cross-compiled.

wjwwood, 2016-03-10:
Yeah, this may be a case where you want to use the `idlpp` file from the host (and not from the target) build of OpenSplice. Our code doesn't differentiate between these two, so probably it would take some work to make that happen. Even if our code did differentiate it would require you to have both the cross-compiled OpenSplice (for linking against and installing to the target device) as well as the host version of OpenSplice from which you can run `idlpp`.

wjwwood, 2016-03-10:
It might be possible for you to pass in the host `idlpp`'s path to CMake, but it will probably require logic to allow it to be overriden either here:

https://github.com/ros2/rmw_opensplice/blob/master/opensplice_cmake_module/cmake/Modules/FindOpenSplice.cmake#L89

and/or here:

https://github.com/ros2/rmw_opensplice/blob/master/opensplice_cmake_module/cmake/Modules/FindOpenSplice.cmake#L108

bbaillio, 2016-03-11:
So the issue is the idlpp tool uses the same libraries that would be needed for cross-linking.
So you essentially need two copies of OpenSplice: 1 that is x86 compatible to run the tools and 1 that is ARM compatible for linking.  And the ament/cmake/make system needs to be smart about which one to use.

I think this same problem happens with Python. If I want my python code to execute on ARM, I need to link against ARM version of the python libs, but the ament build process seems to use x86 python for validation purposes.

gerkey, 2016-03-11:
It sounds like OpenSplice's build system should be modified so that when cross-compiling it will build idlpp and its required libs for host, because that's where they're needed. Their build system is a bit unusual, so I don't know whether that approach is feasible.

dirk-thomas, 2016-03-11:
@gerkey I don't think the build system of OpenSplice can make that distinction. It is a viable use case to cross compile OpenSplice and then use it as well as its message generator on the target platform only.

I think OpenSplice should be built twice, once for the host and once for the target. Then our infrastructure needs to choose `idlpp` from the host and the include paths and libraries from the target.

gerkey, 2016-03-11:
@dirk-thomas Good point. Because we're both cross-compiling to use OpenSplice on the target and wanting to use some of its tools (idlpp) on host, it's on us to make it work.

bbaillio, 2016-03-11:
As long as the usage of idlpp has the lib path of the x86, since it has dependencies.

tfoote, 2018-03-08:
Fixed in #412

closed 2018-03-08 by tfoote