ros2/ros2 #204: pip install path on windows 10 seems a bit different from the instructions

codebot, 2016-03-14:
Maybe I did something weird/wrong while following the [Windows installation instructions](https://github.com/ros2/ros2/wiki/Windows-Development-Setup) :goat: but unfortunately the vcs scripts installed by pip seem to be landing here:
```
C:\ProgramData\chocolatey\lib\python3\tools\Scripts
```
instead of `C:\tools\python\Scripts` which is where the instructions seem to expect they are landing. As a result, "vcs" doesn't work on the command line. Maybe this is due to a version change in pip? Choco is saying that I have 7.1.2 installed, perhaps automatically when python was installed, but it's suggesting that I upgrade to 8.1.0. :hand: Does anyone remember what version was used for the instructions?

THX :cow2: :chicken:

codebot, 2016-03-14:
Related: note that the ProgramData folder is hidden by default, even from the command line or interpreter or whatever they call it. [Here is a forum post](http://answers.microsoft.com/en-us/windows/forum/windows_7-files/can-not-find-the-program-data-folder/bb22413e-7977-e011-8dfc-68b599b31bf5?auth=1) describing how to show hidden folders.

codebot, 2016-03-14:
Related: minesweeper is no longer installed by default. You have to install it from the Windows Store.

wjwwood, 2016-03-14:
Looks like chocolatey changed where it installed Python. So you'll have to
change your path. And you should install the latest pip if you can.

mikaelarguedas, 2016-06-03:
I notices that the same warning appears on jobs on eatable. Should we upgrade pip to 8.1.2 to match windshield's configuration?

mikaelarguedas, 2017-02-01:
I think this can be closed since chocolatey changed (again) where it installs Python. Now it's in `C:\Python35\` and the Scripts folder is `C:\Python35\Scripts` which seems to be on the `PATH` by default. @clalancette (or anybody that set up a Windows machine recently) can you confirm that statement? If so I'll update the install page accordingly and close this.

clalancette, 2017-02-01:
On my latest Windows 10 VM, I actually have Python 3.6 installed from chocolatey at \Python36, and the scripts directory is indeed \Python36\Scripts.  So yeah, I would say you can update the install page and close this out.

mikaelarguedas, 2017-02-01:
Oh that's good to know that chocolatey ships python3.6 now. That means that the entire section about installing python debug libraries is outdated as well. Thanks

mikaelarguedas, 2017-02-01:
instructions updated

closed 2017-02-01 by mikaelarguedas

dirk-thomas, 2017-02-01:
@mikaelarguedas Can you please add a link to the diff for future records.

mikaelarguedas, 2017-02-01:
good point: https://github.com/ros2/ros2/wiki/Windows-Development-Setup/_compare/87546c8a00c54688957e90963ac573899cad19d3...29510ce12f4e8ad5d203a7355245224df6d7898d