ros2/ros2 #206: ARM CI builds hang during cmake

gerkey, 2016-03-19:
On aarch64, console output (http://ci.ros2.org/job/ci_linux-armhf/8/console):

~~~
+++ Building 'ament_cmake_target_dependencies'
01:56:15 Running cmake because arguments have changed.
01:56:15 ==> '. /home/rosbuild/ci_scripts/ws/build/ament_cmake_target_dependencies/cmake__build.sh && /usr/bin/cmake /home/rosbuild/ci_scripts/ws/src/ament/ament_cmake/ament_cmake_target_dependencies -DAMENT_ENABLE_TESTING=1 -DCMAKE_INSTALL_PREFIX=/home/rosbuild/ci_scripts/ws/install/ament_cmake_target_dependencies' in '/home/rosbuild/ci_scripts/ws/build/ament_cmake_target_dependencies'
01:56:15 -- Found ament_cmake_core: 0.0.0 (/home/rosbuild/ci_scripts/ws/install/ament_cmake_core/share/ament_cmake_core/cmake)
01:56:15 -- Found PythonInterp: /home/rosbuild/ci_scripts/venv/bin/python3 (found suitable version "3.4.3", minimum required is "3") 
01:56:15 -- Using PYTHON_EXECUTABLE: /home/rosbuild/ci_scripts/venv/bin/python3
~~~

Call tree:
~~~
  |-docker,31448 daemon
  |   |-sudo,3915 /usr/bin/sudo -H -u rosbuild -E -- /bin/sh -c python3 -u run_ros2_batch.py $CI_ARGS
  |   |   `-sh,3942 /bin/sh -c python3 -u run_ros2_batch.py $CI_ARGS
  |   |       `-python3,3943 /usr/bin/python3 -u run_ros2_batch.py --do-venv --force-ansi-color --opensplice --isolated ...
  |   |           `-sh,4030 /bin/sh -c...
  |   |               `-python,4032 /home/rosbuild/ci_scripts/venv/bin/python -u src/ament/ament_tools/scripts/ament.py build ...
  |   |                   `-sh,4419 /bin/sh -c...
  |   |                       `-cmake,4420 /usr/bin/cmake ...
  |   |                           `-(python3,4428)
~~~

That last python process seems to be a zombie:

~~~
jenkins   4428  0.0  0.0      0     0 ?        Z    01:54   0:01 [python3] <defunct>
~~~

gerkey, 2016-03-19:
Looks like we're hitting this unfortunate interaction between CMake's method for launching child processes and qemu's handling of signals: https://bugs.launchpad.net/qemu/+bug/955379.

This comment from that thread explains the race condition: https://bugs.launchpad.net/qemu/+bug/955379/comments/15

The child exits in a way that the parent (CMake) doesn't hear about it, so a `select()` call blocks forever, leaving behind the child as a zombie. That's what we're seeing, in our case most often with the child being `python3`.

tfoote, 2016-03-21:
We saw these types of problems when we were first doing qemu based builds on the old buildfarm (lucid era). But have not seen anything like that recently.

rohbotics, 2016-08-02:
Looks like I am seeing this issue on this build http://ci.ros2.org/job/ci_linux-armhf/31/console

mikaelarguedas, 2017-05-24:
@gerkey Should this be closed now that we have native builds ? Or should we keep this around for reference if we use qemu again in the future (armhf jobs for example)

gerkey, 2017-05-24:
Closing because we're no longer building with qemu, and I don't recommend that we go back to doing that.

closed 2017-05-24 by gerkey