ros2/ros2 #207: Install folder is not relocatable

bbaillio, 2016-03-22:
On an embedded arm build and on a Windows build downloaded from ros2 GitHub, I am not able to start building a custom project based on the ROS2 core binaries/libraries from the Install folder.
The build process Config tries to look for files at a path on the original build machine, not the current one.
This is despite having sourced the local_setup.bash or setup.bash files.

dirk-thomas, 2016-03-23:
Please provide more information: e.g. the steps you did and which exact paths were pointing to the build machine.

bbaillio, 2016-03-23:
I compiled ROS2 on an ARM target running Ubuntu 12.04 with an upgraded gcc compiler and all the dependencies needed for ROS2 (python, etc).

After the compilation succeeded, I created a tarball from the install/ directory.
I then relocated that tarball to a second ARM target and extracted the tarball.
I can source the setup.bash file just fine. (and subsequently the opensplice release.com file).
When I run ament build on my own, custom project, I get the following error:

File /path/I/compiled/ROS2/on/other/computer/install/lib/python3.4/site-packages/ament_package/template/package_level/local_setup.bash.in does not exist

bbaillio, 2016-03-23:
I download the Windows binaries from the ros2 GitHub releases, and extract them into a folder.
I source the opensplice release.com and the install/setup.bat files.
I run ament build on my own project and I get the following error:

> -- Found PrismTech OpenSplice: C:/Users/Brad.Baillio/_Dev/ros/opensplice/HDE/x86_64.win64
CMake Warning at C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_opensplice_cpp/cmake/ament_cmake_export_include_directories-extras.cmake:11 (message):
  Package 'rmw_opensplice_cpp' exports the include directory
  'C:/dev/opensplice/HDE/x86_64.win64/include' which doesn't exist
Call Stack (most recent call first):
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_opensplice_cpp/cmake/rmw_opensplice_cppConfig.cmake:30 (include)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:21 (find_package)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:30 (include)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rclcpp/cmake/rclcppConfig.cmake:30 (include)
  CMakeLists.txt:10 (find_package)

> CMake Warning at C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_opensplice_cpp/cmake/ament_cmake_export_include_directories-extras.cmake:11 (message):
  Package 'rmw_opensplice_cpp' exports the include directory
  'C:/dev/opensplice/HDE/x86_64.win64/include/sys' which doesn't exist
Call Stack (most recent call first):
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_opensplice_cpp/cmake/rmw_opensplice_cppConfig.cmake:30 (include)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:21 (find_package)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:30 (include)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rclcpp/cmake/rclcppConfig.cmake:30 (include)
  CMakeLists.txt:10 (find_package)

> CMake Warning at C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_opensplice_cpp/cmake/ament_cmake_export_include_directories-extras.cmake:11 (message):
  Package 'rmw_opensplice_cpp' exports the include directory
  'C:/dev/opensplice/HDE/x86_64.win64/include/dcps/C++/SACPP' which doesn't
  exist
Call Stack (most recent call first):
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_opensplice_cpp/cmake/rmw_opensplice_cppConfig.cmake:30 (include)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:21 (find_package)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:30 (include)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rclcpp/cmake/rclcppConfig.cmake:30 (include)
  CMakeLists.txt:10 (find_package)

> -- Found rmw_implementation_cmake: 0.0.0 (C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/rmw_implementation_cmake/cmake)
-- Found std_msgs: 0.0.0 (C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/std_msgs/cmake)
CMake Error: File C:/J/workspace/packaging_windows/ws/install/Lib/site-packages/ament_package/template/package_level/local_setup.bat.in does not exist.
CMake Error at C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/ament_cmake_core/cmake/environment_hooks/ament_generate_package_environment.cmake:68 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/ament_cmake_core/cmake/environment_hooks/ament_cmake_environment_hooks_package_hook.cmake:26 (ament_generate_package_environment)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/Users/Brad.Baillio/_Dev/ros/4-alpha_ros2/ros2-windows/share/ament_cmake_core/cmake/core/ament_package.cmake:57 (ament_execute_extensions)
  CMakeLists.txt:87 (ament_package)

dirk-thomas, 2016-03-23:
I think this is a duplicate of ros2/ros2#141.

bbaillio, 2016-03-23:
`/path/I/compiled/ROS2/on/other/computer/install/lib/python3.4/site-packages/ament_package/template/package_level/local_setup.bash.in does not exist`

But 
`/path/I/extracted/ROS2/on/this/computer/install/lib/python3.4/site-packages/ament_package/template/package_level/local_setup.bash.in` **_does_** exist.

So I don't understand why it's trying to access a file that's packaged in the install directory at a location it was built/installed at on a different computer.
This has nothing to do with the location of the python3 executable, I don't think.

I can get this same error if I take the install directory, tar it up, move it to a different location on the same computer, and extract it there and try to build on top of it.

dirk-thomas, 2016-03-23:
Regarding the wrong path for the templates the problem seems to be this template (https://github.com/ament/ament_cmake/blob/master/ament_cmake_core/ament_cmake_package_templates-extras.cmake.in) as well as the generating script (https://github.com/ament/ament_cmake/blob/master/ament_cmake_core/cmake/package_templates/templates_2_cmake.py). Instead of using the absolute path coming from the Python API this should modify the paths to be relative to the CMake dir of the current package.

For Windows and OpenSplice it looks like this line (https://github.com/ros2/rmw_opensplice/blob/61eaad3e54c0fe8ca040041b56cc0aaa47abd365/rmw_opensplice_cpp/CMakeLists.txt#L37) simply exports the include directories. In your case where the location of the external OpenSplice is different on the second system this path is wrong. In order to support that the package can't export a fixed path but has to find OpenSplice every time another package is being built on-top.

dirk-thomas, 2016-03-24:
@bbaillio Can you please try the first case (build, move install space, and build something on-top) with the two branches of these PRs: ament/ament_cmake#63 and ament/ament_package#38

dirk-thomas, 2016-03-24:
@bbaillio Can you please try it again with the new Windows archive from http://ci.ros2.org/view/packaging/job/packaging_windows/175/ It contains the fix from ros2/rmw_opensplice#122.

bbaillio, 2016-03-24:
i'll try all that stuff today

bbaillio, 2016-03-24:
So, in trying to build with the two branch PRs, I run into build issues. Forgive me if I did something wrong, but I don't have a lot of GitHub, PR, etc experience.
This is what I did:
In my ros2_ws/src directory, I navigated to the ament/ament_cmake folder. I did a `git pull`, then `git checkout make_template_paths_relocatable`. I did the same for the ament/ament_packages folder.

Then I kicked of the ament build process again (after removing the previous build and install directories).

I tried this process on windows x86 and Linux x86 and got the same error:
`-- Could NOT find PythonExtra (missing:  PythonExtra_EXTENSION_SUFFIX)
CMake Error at C:/Asi/ros2_ws/install/share/ament_cmake_python/cmake/ament_python_install_module.cmake:32 (message):
  ament_python_install_module() called with unused arguments:
  DESTINATION;builtin_interfaces
Call Stack (most recent call first):
  C:/Asi/ros2_ws/install/share/ament_cmake_python/cmake/ament_python_install_module.cmake:26 (_ament_cmake_python_install_module)
  C:/Asi/ros2_ws/install/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:152 (ament_python_install_module)
  C:/Asi/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/Asi/ros2_ws/install/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:148 (ament_execute_extensions)
  CMakeLists.txt:12 (rosidl_generate_interfaces)`

dirk-thomas, 2016-03-24:
@bbaillio Can you please try it again with the new Windows archive from http://ci.ros2.org/view/packaging/job/packaging_windows/177/ It contains the fix from ros2/rmw_opensplice#122.

bbaillio, 2016-03-24:
When I try to source the setup.bat or local_setup.bat files, they don't actually set any environment variables.
So I can't run ament in a global sense.

dirk-thomas, 2016-03-24:
II can't reproduce your problem `Could NOT find PythonExtra`. I was able to build a workspace with all `ament*` repos, then move the install folder to a new location, and build a second workspace with the other repos on-top after sourcing the setup file from the new location.

The error message also looks unrelated to what these branches are changing. Please provide reproducible step-by-step instructions what you are doing when you run into that problem.

--

I just downloaded the archive, unpacked it and ran `local_setup.bat`. All the expected environment variables are there for me: `AMENT_PREFIX_PATH`, `OSPL_*`, extended `Path`, `PYTHONPATH`, etc. Please provide reproducible step-by-step instructions what you are doing when you run into that problem.

wjwwood, 2016-03-25:
> For Windows and OpenSplice it looks like this line (https://github.com/ros2/rmw_opensplice/blob/61eaad3e54c0fe8ca040041b56cc0aaa47abd365/rmw_opensplice_cpp/CMakeLists.txt#L37) simply exports the [...]

@dirk-thomas I'm sure that https://github.com/ros2/rmw_opensplice/pull/122 will change the behavior as you have described, but I'm not sure that be enough. For example, the setup file generated by `opensplice_cmake_module` will reference the old location of OpenSplice:

https://github.com/ros2/rmw_opensplice/blob/61eaad3e54c0fe8ca040041b56cc0aaa47abd365/opensplice_cmake_module/env_hook/opensplice.sh.in#L1

I think it's perfectly reasonable that the location of our install folder moves on the new computer, but moving the location of or changing the version (our zip file versus PrismTech's install package) of OpenSplice on the new computer is not something I think we can support without significant changes.

dirk-thomas, 2016-03-25:
@wjwwood The referenced line could be made more flexible by checking if `OSPL_HOME` is already set (as we already do in other rmw impl.: https://github.com/ros2/rmw_connext/blob/c9d750a4ee920d06b5b8abdef766928132c7e99b/connext_cmake_module/env_hook/connext.sh.in#L5-L8). That would allow users to source their OpenSplice release file afterwards in order to use a different one.

wjwwood, 2016-03-25:
They would need to source the OpenSplice release file before sourcing ours, but yes you could do that. It's not clear to me that this is the only place we assume that OpenSplice isn't moving after we install our files.

wjwwood, 2016-03-25:
The connext snippet you referenced only warns the user that we are changing the NDDS_HOME to be what we think it should be. It doesn't allow them to use another one, as far as I can see.

bbaillio, 2016-03-25:
@dirk-thomas Ok. I investigated the windows build you pointed me to. I did a diff of the local_setup.bat file from `http://ci.ros2.org/view/packaging/job/packaging_windows/177` and from the alpha4 release package and this line was different:
`set "ament_python_executable=C:\tools\python\python.exe"` (from alpha4)
`set "ament_python_executable=C:\ProgramData\chocolatey\lib\python3\tools\python.exe"` (from 177)

I modified the file to point back to `tools\python\python.exe` and the source-ing of the script worked, but ament failed to execute. At all.

So I performed `choco install -y python3` and I got an install in the `chocolatey\lib\python3\tools\` directory.
So I reverted the local_setup.bat back to its original state and sourced it. Everything worked.
Then I did an `ament build` on my custom package and I still get this error:
`CMake Error: File C:/J/workspace/packaging_windows/ws/install/Lib/site-packages/ament_package/template/package_level/local_setup.bat.in does not exist.
CMake Error at C:/Asi/ros2-windows/share/ament_cmake_core/cmake/environment_hooks/ament_generate_package_environment.cmake:68 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  C:/Asi/ros2-windows/share/ament_cmake_core/cmake/environment_hooks/ament_cmake_environment_hooks_package_hook.cmake:26 (ament_generate_package_environment)
  C:/Asi/ros2-windows/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  C:/Asi/ros2-windows/share/ament_cmake_core/cmake/core/ament_package.cmake:57 (ament_execute_extensions)
  CMakeLists.txt:95 (ament_package)
`

dirk-thomas, 2016-03-25:
The different location of the Python executable is expected. Since alpha 4 we have updated the Windows build machine to install Python using Chocolatey rather then a custom installation. It is a known limitation the Python executable needs to be in the same location.

The Windows packaging build only includes the fix from ros2/rmw_opensplice#122. It does *not* include the changes from ament/ament_cmake#63 and ament/ament_package#38. So that the templates are still looked up in the old install location is expected. On Linux we confirmed that the two patches fix the problem. Once these changes are merged the Windows package will also benefit from that.

dirk-thomas, 2016-03-25:
After merging the two ament related PRs I have triggered another packaging job. The artifact of the following build will get you past the template location problem: http://ci.ros2.org/view/packaging/job/packaging_windows/179/

bbaillio, 2016-03-25:
Great! I tested with that build and I was able to build my custom package using the 179 artifact as the ROS2 base.

closed 2016-04-26 by bbaillio