ros2/ros2 #212: Alpha 6 (2016-05-18)

gerkey, 2016-04-06:
* [ ] Fix all test failures (or disable flaky tests)
* [ ] Get first version of rostopic-like functionality, including some graph API, identifying rclpy improvements along the way. @vmayoral
* [ ] Support nested messages in Python (1 week) @mikaelarguedas
* [ ] Make arm (32 and 64) CI run nightly without hanging (0.5 week) @gerkey
* [ ] Refactor rclcpp to use rcl (1 week) ros2/rclcpp#204 @jacquelinekay 
* [ ] Follower demo / tutorial using TurtleBot 2(.5), focusing on ROS 1 vs. ROS 2 code, structure, and usage
  * [ ] ROS 2 wrapper for Kobuki
    * [ ] Address use of dynamic_reconfigure somehow
  * [ ] ROS 2 wrapper for depth camera (e.g., Astra)
  * [ ] ROS 2 wrapper for joystick input
  * [ ] ROS 2 implementation of follower node using depth image, perhaps using OpenCV and/or PCL (probably already done by @tfoote)
  * [ ] Consider multi-robot examples, bad wifi, and other ways to demonstrate differences in ROS 2
  * [ ] Implement wait_for_service.
  * [ ] Improvement for launch (as identified during use).
  * [ ] rosconsole-like replacement for printf (not necessarily wire transport version or other outputs)
  * [ ] Use bridge to talk to rviz and other tools as needed
* [ ] Update build farm to use newer VS2015.

Stretch goals:
* [ ] FreeRTPS improvements: RMW support @codebot (and maybe @jacquelinekay)
  * [ ] Merge history cache and make it work (2 weeks)
  * [ ] Testing/CI for linux case (1 weeks)
    * [ ] Add testing and test reporting to the existing CI job
* [ ] Add doc job at ci.ros2.org (call doxygen, sphinx, whatever)
* [ ] Support parameters in C ros2/rcl#28 (2 weeks) @jacquelinekay 
* [ ] Support Services in Python @wjwwood

jacquelinekay, 2016-04-15:
Thinking of also making a general (not necessarily Turtlebot-specific) `launch` tutorial and a parameters tutorial as a goal of this alpha.

dirk-thomas, 2016-05-17:
We talked about the logging subsystem in todays meeting and decided to include a namespace hierarchy in the core logging API. Therefore the current `console_bridge` API won't work as-is since it doesn't support namespacing. It either needs to be revised with a new API or a similar replacement is required.

dirk-thomas, 2016-06-01:
Released today.

closed 2016-06-01 by dirk-thomas