ros2/ros2 #217: Switch to Xenial

dirk-thomas, 2016-04-26:
This ticket collects everything necessary to switch to Xenial. All branches should be named `xenial` to enable CI testing of all changes together.

* [x] The CI jobs need to be updated to use Xenial instead of Trusty ros2/ros2#216
  * [ ] manually install new dependencies (`pydocstyle`) on Windows slaves
* [ ] create Debian package of OpenSplice
* [x] switch to use upstream `pydocstyle` instead of embedded version and update default warnings ament/ament_lint#49
* [x] allow newer version of `clang-format` if available ament/ament_lint#49
* [ ] fix complier warnings ros2/examples#107
* [ ] fix complier warnings ros2/realtime_support#37
* [ ] update wiki pages

dirk-thomas, 2016-05-18:
Using the now available Debian package of OpenSplice the Linux job was able to build with all rmw implementations: http://ci.ros2.org/job/ci_linux/1312/

tfoote, 2016-05-23:
We need to consider how this will effect our embedded test platforms. @mikaelarguedas and I were talking at the end of last week and realized if we jump forward to Xenial we won't have any ARM test platforms we can test on. 

Almost all of them are on Ubuntu Trusty or Debian Jessie. Xenial support is supposed to come soon, but is not ready yet.

tfoote, 2016-05-24:
We'll plan to move forward with dropping trusty support in Alpha 7. We expect new base images for the boards in the next 6 months with the newer versions which will allow us to use the newer functionality. We do have the ability to test on arm and arm64 in docker/qemu.

dirk-thomas, 2016-06-10:
All pull requests are ready to be merged. Waiting for review.

dirk-thomas, 2016-06-10:
All pull requests are ready to be merged. Waiting for review.

Latest builds show no regressions:
* http://ci.ros2.org/job/ci_linux/1432/
* http://ci.ros2.org/job/ci_osx/1115/
* http://ci.ros2.org/job/ci_windows/1448/

dirk-thomas, 2016-06-15:
Another set of builds after all PRs have been merged:

* http://ci.ros2.org/job/ci_linux/1450/
* http://ci.ros2.org/job/ci_osx/1132/
* http://ci.ros2.org/job/ci_windows/1465/

dirk-thomas, 2016-06-15:
@wjwwood Can you please install CMake 3.5 on *dosa*.

wjwwood, 2016-06-15:
It's done. You know the credentials for dosa are on the internal wiki though...

dirk-thomas, 2016-06-15:
Sure, but I have no idea how to install a new version of CMake on OS X :wink:

dirk-thomas, 2016-06-15:
Sure, but I have no idea how to install a new version of CMake on OS X :wink: Thanks

jacquelinekay, 2016-06-15:
homebrew is still a mystery to me. Might be good to take time to learn about it though. Thanks William :beers:

wjwwood, 2016-06-15:
https://github.com/Linuxbrew/linuxbrew?

wjwwood, 2016-06-15:
Ack, that's a stale page: http://linuxbrew.sh/

dirk-thomas, 2016-06-16:
I have created PRs for all repos regarding CMake 3.5. Since they touch a lot of CMake code I would like to get them merged asap. Please comment on ament/ament_cmake#72 with a plus one after reviewing all the PRs referenced there.

The latest CI builds look good:
* http://ci.ros2.org/job/ci_linux/1466/
* http://ci.ros2.org/job/ci_osx/1141/
* http://ci.ros2.org/job/ci_windows/1476/

wjwwood, 2016-06-16:
I had a few comments, but I looked over all the pr's and they look good to me.

closed 2016-06-16 by dirk-thomas