ros2/ros2 #219: Launch test file references non-existent data member

bbaillio, 2016-05-03:
In the test directory of the launch source [https://github.com/ros2/launch/tree/master/launch/test], the launch_multiple_counter.py example references the data member process_descriptors of class LaunchDescriptor. This doesn't exist. I think the example should reference task_descriptors instead.

bbaillio, 2016-05-03:
There also appears to be something broken if using task_descriptor as a replacement, when a process completes. It tries to access something that doesn't exist:
`Traceback (most recent call last):
  File "C:\Asi\5alpha_ros2_ws\install\bin\launch-script.py", line 9, in <module>
    load_entry_point('launch==0.0.0', 'console_scripts', 'launch')()
  File "C:\Asi\5alpha_ros2_ws\install\Lib\site-packages\launch\main.py", line 47, in main
    rc = launcher.launch()
  File "C:\Asi\5alpha_ros2_ws\install\Lib\site-packages\launch\launcher.py", line 90, in launch
    returncode = loop.run_until_complete(self._run())
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\asyncio\base_events.py", line 337, in run_until_complete
    return future.result()
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\asyncio\futures.py", line 274, in result
    raise self._exception
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\asyncio\tasks.py", line 239, in _step
    result = coro.send(None)
  File "C:\Asi\5alpha_ros2_ws\install\Lib\site-packages\launch\launcher.py", line 203, in _run
    p.task_states[index].restart_count += 1
AttributeError: 'ProcessDescriptor' object has no attribute 'task_states'`

dirk-thomas, 2016-05-03:
Can you please clarify what commands you executed when you ran into this. I am not sure that file is currently used anywhere.

bbaillio, 2016-05-03:
Sure thing. I am playing with launching my own custom apps using **launch**, and wanted to use the aggregation technique shown in launch_multiple_counter.py. Before writing my own launch file, I wanted to run launch_multiple_counter.py. So I sourced the ROS2 setup file and navigated to src\ros2\launch\launch\test\ and ran `launch launch_multiple_counter.py`

dirk-thomas, 2016-05-03:
ros2/launch#30 should address both problems.

closed 2016-05-03 by dirk-thomas