ros2/ros2 #223: FastRTPS examples fail to build on the farm

wjwwood, 2016-05-21:
All jobs that build for FastRTPS are failing, but for example see: http://ci.ros2.org/view/nightly/job/nightly_linux_debug/79/

```
11:02:00 CMake Error at CMakeLists.txt:6 (find_package):
11:02:00   By not providing "Findfastcdr.cmake" in CMAKE_MODULE_PATH this project has
11:02:00   asked CMake to find a package configuration file provided by "fastcdr", but
11:02:00   CMake did not find one.
11:02:00 
11:02:00   Could not find a package configuration file provided by "fastcdr" with any
11:02:00   of the following names:
11:02:00 
11:02:00     fastcdrConfig.cmake
11:02:00     fastcdr-config.cmake
11:02:00 
11:02:00   Add the installation prefix of "fastcdr" to CMAKE_PREFIX_PATH or set
11:02:00   "fastcdr_DIR" to a directory containing one of the above files.  If
11:02:00   "fastcdr" provides a separate development package or SDK, be sure it has
11:02:00   been installed.
11:02:00 
11:02:00 
11:02:00 -- Configuring incomplete, errors occurred!
```

The issue is that the `fastcdr` install directory is not on the `CMAKE_PREFIX_PATH`, and the reason for that is that `ament_tools` did not find an instance of `find_package(fastcdr)` in the `CMakeLists.txt` of Fast-RTPS.

Originally I thought the issue was that eProsima moved their `find_package(fastcdr)` into a `CMakeLists.txt` of a subdirectory, so I did this change set in `ament_tools`:

https://github.com/ament/ament_tools/pull/95

However, that didn't fix the issue either, because the real problem is that none of the `CMakeLists.txt` in Fast-RTPS's build need to do `find_package(fastcdr)`. There are examples in the Fast-RTPS repository that use both `fastcdr` and `fastrtps`, but they are not included in the Fast-RTPS build. Instead the Fast-RTPS build invokes cmake again for each example:

https://github.com/eProsima/Fast-RTPS/commit/04de00039ab9b928de25216b83c5f34e1f572214 (see `cmake/dev/compile_example.cmake`)

So my above pull request will not address this problem. I think what they're doing is strange, but I don't think we should ask them to change their behavior for our tool.

These are the options I think we have:

- Ask them to `find_package(fastcdr)` somewhere in their Fast-RTPS code (maybe inside the "if build examples" code)
- Add a package.xml with the explicit dependencies and inject it in our build process some how
- Disable building of examples in Fast-RTPS (this does work) by injecting the option into our farm somehow
- Have `ament_tools` look at all `CMakeLists.txt` in the project recursively (regardless of whether or not the toplevel `CMakeLists.txt` references it)

I took the liberty of doing the last option up as a pr as well, since it was easy:

https://github.com/ament/ament_tools/pull/96

What should we do?

wjwwood, 2016-05-21:
CI for the latter solution: http://ci.ros2.org/view/nightly/job/nightly_linux_debug/80/

gerkey, 2016-05-21:
I'm not sure that I fully understand what's happening here, but: would this issue disappear in a non-isolated install? If so, then another option is to change ament to not isolate the installation of vanilla cmake projects. For vanilla projects, instead of attempting to infer cross-project dependencies by parsing their `CMakeLists.txt` files, ament could just install them all into one prefix.

wjwwood, 2016-05-21:
In this case, fastcdr gets built before fastrtps because of alphabetical ordering. Without having the dependency though that would not necessarily be true. So doing non isolated for the cmake projects on fixes half the problem, because you still need to know the right order to build them in.

wjwwood, 2016-05-24:
@dirk-thomas when you get a chance can you comment on which solution you would prefer.

tfoote, 2016-05-24:
From discussion:

For the short term. Add the hack to walk all CMakeLists.txt  recursively will get us building again. #96 

We should in parallel request to package the examples separately or default them to not build w/o an active option. 

We could also ask for a package.xml and see if they're willing to add it.

wjwwood, 2016-05-24:
@tfoote I think you mean ament/ament_tools#96

tfoote, 2016-05-24:
Yes, fixed.

wjwwood, 2016-05-26:
Closed with ament/ament_tools#96

dirk-thomas, 2016-05-27:
As discussed in https://github.com/ament/ament_tools/pull/96#issuecomment-222003792 the hack in ament should be reverted.

mikaelarguedas, 2017-05-24:
closing this given that the issue is fixed and ament/ament_tools#96 has been reverted in ament/ament_tools#148

closed 2017-05-24 by mikaelarguedas

gaoyadianta, 2019-05-05:
i have encountered the exactly same question, anybody can tell me what to do to fix this problem?