ros2/ros2 #225: Alpha 7

tfoote, 2016-05-31:
* [ ] Fix all test failures (or disable flaky tests) (everybody)
  * [ ] retry for flaky tests #224 
  * [x] continue downscoping as needed
* Infrastructure items
  * [ ] Update build farm to use newer VS2015 @codebot 
  * [ ] code coverage for C/C++ using Bullseye or other tools, and Jenkins plugin(s) (like for Gazebo buildfarm)
  * [ ] migrate to Xenial
* [ ] FastRTPS Python and C support  https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/pull/27
* [ ] interoperate system test for FastRTPS and OpenSplice (first cover languages, then cross-languages)
* [ ] Implement wait_for_service @wjwwood https://github.com/ros2/ros2/issues/215
  * [ ] fastrtps
  * [x] opensplice
  * [ ] apply to rclcpp
* [ ] Support parameters in C ros2/rcl#28 (2 weeks) @jacquelinekay 
  * Various parameter improvements that follow (rclcpp, parameter design, namespace demos, ...)
* [ ] Follower demo / tutorial using TurtleBot 2(.5), focusing on ROS 1 vs. ROS 2 code, structure, and usage, which requires (@ernestmc to help):
  * [ ] ROS 2 wrapper for Kobuki @gerkey
    * [ ] Address use of dynamic_reconfigure somehow
  * [ ] ROS 2 wrapper for depth camera (Kinect1 via freenect) @tfoote maybe Astra if things go incredibly well incredibly quickly
  * [ ] ROS 2 implementation of follower node on x86 @tfoote
  * [ ] Test multi-robot examples, bad wifi, and other ways to demonstrate differences in ROS 2
* [ ] Features that are likely required to make the TurtleBot demo better:
  * [ ] Improvements for launch (as identified during use).
    * [ ] perhaps including parameters in the YAML/python, other types of static configuration files, etc.
  * [ ] command-line dynamic parameter tool (ros2param)
  * [ ] rosconsole-like replacement for printf (not necessarily wire transport version or other outputs) @dirk-thomas https://docs.google.com/document/d/1ygcs7mPe_O53rXfVHhJC6BKwWOZIwaWIW_8OTmP8LIE
  * [ ] Make arm (32 and 64) CI run nightly without hanging (0.5 week) @gerkey 

STRETCH GOALS
* [ ] Get first version of rostopic-like functionality, including some graph API, identifying rclpy improvements along the way. @vmayoral https://github.com/ros2/ros2/issues/197
* [ ] ARM SBC-friendly version of follower using depth image, perhaps using OpenCV and/or PCL (probably already done by @tfoote)
* [ ] FreeRTPS improvements: RMW support rename to CMBML migrate to C++999 @codebot (and maybe @jacquelinekay)
  * [ ] Merge history cache and make it work (2 weeks)
  * [ ] Testing/CI for linux case (1 weeks)
    * [ ] Add testing and test reporting to the existing CI job
* [ ] Add doc job at ci.ros2.org (call doxygen, sphinx, whatever)
* [ ] Support Services in Python @wjwwood
* [ ] Refactor type support to have less duplication and easier reuse
* [ ] Refactor rmw to be more like rcl (3 weeks)

kartikmohta, 2016-09-13:
Seems like Alpha 7 is already released, are the incomplete items left for next release?

Karsten1987, 2016-10-18:
can we close this with a remark for alpha-8?

mikaelarguedas, 2016-10-18:
good point, closing this in favor of #281

closed 2016-10-18 by mikaelarguedas