ros2/ros2 #226: ROS2 and HLA

smamessier, 2016-06-01:
I just got aware of the ros2 project. I've been working a lot with ROS and the High Level Architecture (HLA) so I have a couple of questions. Over the years, I realized that ROS is a very interesting middleware for fast-time simulations, however such distributed simulations lack determinism as the timing depends on network fluctuations. Unless you use a framework such as HLA which includes time-synchronization interfaces. I see that you are already using DDS for the publish/subscribe paradigm, would it be possible to consider part of HLA specs ( IEEE 1516 High Level Architecture for Modeling and Simulation ) for built-in time synchronization services ? HLA has an open-source C++ implementatino (CERTI) which could maybe be leveraged.

wjwwood, 2016-06-01:
> however such distributed simulations lack determinism as the timing depends on network fluctuations.

Do you have any ideas how that would be addressed in an asynchronous message passing system?

In ROS 1 we have ROSTime which can be used to coordinate "federates" in a "federation" (to put it into HLA terminology) to some degree by using a clock topic. We also have "Services" which are request-reply, which could be used to coordinate each "step" of the simulation, but I'm not familiar enough with HLA to know if that's sufficient.

You'd have to champion a proposal of how we could change what we do in ROS 1 to be more accommodating to HLA as we're not experts in that scheme. Presumably that would involve some time synchronized orchestration of the various parts of the system, but it's not clear to me what parts of ROS prevent you from doing so as easily as you'd like.

Looking at CERTI briefly, I don't see how we would leverage that in ROS 2, since it serves as the "RTI" for an HLA system which deals with brokering data between federates. That's already accomplished by ROS using topics and services, though maybe not to the standards or matching the behavior required by HLA. However, I don't think it makes sense to abandon our publish/subscribe topics and request/reply services for an HLA "RTI" like CERTI because it probably won't satisfy the requirements of our other, existing users.

I guess what I have a hard time seeing is why ROS and CERTI need to be integrated? Could they not just be used together by a system integrator?

smamessier, 2016-06-01:
You're right, actually ROS already takes care of most of what CERTI does. My thought was more to add more built-in time-synchronization features. And taking some of the HLA specs could be the way to go. As you said il could be handled in an ad-hoc fashion using services but providing a standardized - more integrated - way would maybe lead to a broader adoption in the distributed simulation community.

smamessier, 2016-06-01:
Regarding reusing some of CERTI, I was talking about the part that `orchestrates` the synchronization, it could be a Ros package running on the RosMaster based on Certi's existing time management submodule. It seems pretty easy to implement `conservative time management` using ROS existing API (Although it would be convenient to have something like ros:SpinHLA which waits until allowed to advance in time, even better would be ros:SpinOnceHLA(lookeahead_time) which would allow for adaptive time step simulations). Again you're right this could also be part of an external package - which I planned to write for ROS 1 anyway - but since ROS2 is still in its early stages, I was wondering if someone was maybe interested in integrating proper time management this directly in the core.

gbiggs, 2016-06-03:
I think that something that waits until it is allowed to advance in time or adaptively steps in time is a perfect place for using a custom executor. It would wait for some message (via topic, API call, shared memory, or whatever) before executing its contained nodes the desired number of steps. Indeed, we have exactly that in OpenRTM for use with simulators. It works well for synchronising execution of components with simulation steps for users who want that sort of rigid simulation.

So what you want to achieve should, I think, be easy to achieve in the ROS 2 architecture.

wjwwood, 2016-06-10:
We have had some ideas in the past for how to efficiently describe and construct synchronous systems out of the asynchronous primitives in ROS. We'll keep these suggestions in mind as we look to prototype those patterns in the future. If you chance to notice any concrete points of concern along the way, please let us know. Thanks.

I'm going to close this issue, but please feel free to continue the discussion here or on our mailing list (in fact you may get more views and feedback there https://groups.google.com/forum/#!forum/ros-sig-ng-ros). I can reopen the issue if necessary.

closed 2016-06-10 by wjwwood