ros2/ros2 #227: CI: Windows build fails

jacquelinekay, 2016-06-02:
On windshield:

http://ci.ros2.org/job/ci_windows/1384/consoleText

```
    copying vcstool-completion\vcs.bash -> C:\ProgramData\chocolatey\lib\python3\tools\share\vcstool-completion
    error: could not delete 'C:\ProgramData\chocolatey\lib\python3\tools\share\vcstool-completion\vcs.bash': Access is denied
```

jacquelinekay, 2016-06-02:
http://ci.ros2.org/job/ci_windows/1385/

dirk-thomas, 2016-06-02:
There is a new `vcstool` release and it looks like the job tries to install it to the system location without having admin permission. For a short term workaround you can just install the new package manually. For the long term the Windows jobs shouldn't even try to install into system directories or have the necessary permissions.

codebot, 2016-06-02:
i'll try to run that choco command by hand from a Super Administrator terminal with Filesystem and Internet Permissions 13.0+ and see what happens

codebot, 2016-06-02:
the pip installation worked fine in a Super Administrator terminal with Filesystem and Internet Permissions 13.0+.  :chicken:  hopefully CI will work now

jacquelinekay, 2016-06-02:
Thanks :hamburger:

jacquelinekay, 2016-06-02:
I'm going to cancel the job you started to test this since it seems to be successful so that I can test another branch, if that's ok.

jacquelinekay, 2016-06-02:
Oh nevermind, it's running on eatable.

dirk-thomas, 2016-06-02:
I would suggest to keep this ticket open since the actual problem is not resolved and will happen again every time a dependency has a newer version.

jacquelinekay, 2016-06-02:
ok, I've changed the name to reflect the underlying problem, feel free to change it if you don't think that's accurate,

tfoote, 2018-03-08:
Resolved a while ago. the machines still auto update in the background and bhe build cop periodically checks on them.

closed 2018-03-08 by tfoote