ros2/ros2 #233: Problem with QoS settings

rafal-kozik, 2016-06-09:
Hi,

I am working on evaluation of ROS2 and I prepared some benchmarks to check how ROS2 behaves in an unreliable environment where packets are lost from time to time. I have three types of messages with different QoS:

```cpp
// Reliable communication with a queue size up to resource limits
// Keep data for late-joiners up to resource limits
static const rmw_qos_profile_t rmw_qos_profile_alarm =
{
	RMW_QOS_POLICY_KEEP_ALL_HISTORY,
	100, // Should be ignored
	RMW_QOS_POLICY_RELIABLE,
	RMW_QOS_POLICY_TRANSIENT_LOCAL_DURABILITY
};

// Reliable communication with a queue size of 10 samples
// Do not keep data for late-joiners
static const rmw_qos_profile_t rmw_qos_profile_control =
{
	RMW_QOS_POLICY_KEEP_LAST_HISTORY,
	10,
	RMW_QOS_POLICY_RELIABLE,
	RMW_QOS_POLICY_VOLATILE_DURABILITY
};

// Send and forget
static const rmw_qos_profile_t rmw_qos_profile_sensor =
{
	RMW_QOS_POLICY_KEEP_LAST_HISTORY,
	0,
	RMW_QOS_POLICY_BEST_EFFORT,
	RMW_QOS_POLICY_VOLATILE_DURABILITY
};
```
Those graphs show how many packets were not delivered as a function of percentage of lost packets.
[image]
[image]
[image]
It looks as if there was a reliable channel for all types of messages with a limited history. To verify this I ran the same benchmarks on nodes implemented using only OpenSplice library, here are the results:
[image]
[image]
[image]
Those results look fine:
- Sensor: linear correlation between lost packets and packets delivered
- Control: channel is reliable so more packets reach the subscriber
- Alarm: as long as it possible to connect, all packets are delivered

When I used Wireshark to verify QoS settings, information in RTPS packets didn’t match ROS2 settings. It seemed ok for OpenSplice.

I’ve also tested QoS settings using listener and talker aplications from examples: [listener.cpp](https://github.com/ros2/examples/blob/master/rclcpp_examples/src/topics/listener.cpp) and [talker.cpp](https://github.com/ros2/examples/blob/master/rclcpp_examples/src/topics/talker.cpp). I gathered  packets for two profiles: rmw_qos_profile_sensor_data and rmw_qos_profile_parameter_events ([qos_profiles.h](https://github.com/ros2/rmw/blob/master/rmw/include/rmw/qos_profiles.h)), which have different Reliability and History settings. When I examined them with Wireshark I didn’t see any differences. For example, packets with data related to „chatter” topic (filter rtps.param.topicName == chatter) have the same Reliability setting and no information on History:
```
submessageId: DATA (0x15)
    Flags: 0x05, Data present, Endianness bit
    octetsToNextHeader: 316
    0000 0000 0000 0000 = Extra flags: 0x0000
    Octets to inline QoS: 16
    readerEntityId: ENTITYID_BUILTIN_SUBSCRIPTIONS_READER (0x000004c7)
    writerEntityId: ENTITYID_BUILTIN_SUBSCRIPTIONS_WRITER (0x000004c2)
    writerSeqNumber: 1
    serializedData
        encapsulation kind: PL_CDR_LE (0x0003)
        encapsulation options: 0x0000
        serializedData:
            PID_UNICAST_LOCATOR
            PID_UNICAST_LOCATOR
            PID_EXPECTS_INLINE_QOS
            PID_PARTICIPANT_GUID
            PID_TOPIC_NAME
                parameterId: PID_TOPIC_NAME (0x0005)
                parameterLength: 12
                topic: chatter
            PID_TYPE_NAME
            PID_KEY_HASH
            PID_ENDPOINT_GUID
            PID_PROTOCOL_VERSION
            PID_VENDOR_ID
            PID_DURABILITY
            PID_DEADLINE
            PID_LATENCY_BUDGET
            PID_LIVELINESS
            PID_RELIABILITY
                parameterId: PID_RELIABILITY (0x001a)
                parameterLength: 12
                Kind: BEST_EFFORT_RELIABILITY_QOS (0x00000001)
            PID_LIFESPAN
            PID_OWNERSHIP
            PID_DESTINATION_ORDER
            PID_SENTINEL

submessageId: DATA (0x15)
	Flags: 0x05, Data present, Endianness bit
	octetsToNextHeader: 324
	0000 0000 0000 0000 = Extra flags: 0x0000
	Octets to inline QoS: 16
	readerEntityId: ENTITYID_UNKNOWN (0x00000000)
	writerEntityId: ENTITYID_BUILTIN_PUBLICATIONS_WRITER (0x000003c2)
	writerSeqNumber: 2
	serializedData
		encapsulation kind: PL_CDR_LE (0x0003)
		encapsulation options: 0x0000
		serializedData:
			PID_UNICAST_LOCATOR
			PID_UNICAST_LOCATOR
			PID_PARTICIPANT_GUID
			PID_TOPIC_NAME
				parameterId: PID_TOPIC_NAME (0x0005)
				parameterLength: 12
				topic: chatter
			PID_TYPE_NAME
			PID_KEY_HASH
			PID_ENDPOINT_GUID
			PID_TYPE_MAX_SIZE_SERIALIZED
			PID_PROTOCOL_VERSION
			PID_VENDOR_ID
			PID_DURABILITY
			PID_DEADLINE
			PID_LATENCY_BUDGET
			PID_LIVELINESS
			PID_RELIABILITY
				parameterId: PID_RELIABILITY (0x001a)
				parameterLength: 12
				Kind: RELIABLE_RELIABILITY_QOS (0x00000002)
			PID_LIFESPAN
			PID_OWNERSHIP
			PID_OWNERSHIP_STRENGTH
			PID_DESTINATION_ORDER
			PID_SENTINEL
```
Other packets don’t contain History settings either (filters rtps.history.kind == 0x0 and rtps.history.kind == 0x1).

It seems that there might be a problem with passing QoS settings to a DDS implementation. If you need more information, let me know.

OS: Ubuntu 14.04.4 LTS in Docker version 1.10.3
ROS2: alpha6
DDS: libopensplice64 6.4.0-3~trusty

vmayoral, 2016-06-09:
Thanks for sharing this @rafal-kozik. Any chance you could share the sources used to generate all the data and plot it? I'd be interested to see how different DDS implementations compare to each other.

dirk-thomas, 2016-06-09:
Can you please share more information about your example. Without information like message size and publishing frequency the results are difficult to interpret.

> When I used Wireshark to verify QoS settings, information in RTPS packets didn’t match ROS2 settings. It seemed ok for OpenSplice.

You might want to clarify what exact values don't match. It could be a bug but it could also be that your expectation is incorrect. E.g. some policies are local to the communication endpoint and not communicated over the wire.

rafal-kozik, 2016-06-10:
We will upload the project to GitHub today and I will let you know.

adamdbrw, 2016-06-10:
The github project is ready:
https://github.com/piappl/ros2_benchmarking

The ROS2 evaluation is a part of our tasks in R5-Cop project (http://r5-cop.eu/en/). We aim to create an independent assessment of the current ROS2 state in terms of whether it looks promising, shows some gains on ROS1 already, whether it will be worth migrating to anytime soon, etc.

For this task, we could also benefit from any other evaluations or comparisons that you have done or are aware of. I saw a task planned for alpha7 (previously for alpha6): "Test multi-robot examples, bad wifi, and other ways to demonstrate differences in ROS 2", the outcome of which we are also interested in.

The licence we chose is LGPL3, let me know if that's ok for you. Feel free to give feedback and contribute. The project is of course still under development. We currently have ROS1, ROS2 and OpenSplice DDS nodes for testing, with configurable QoS and message parameters (frequency, size etc).

The project has additional value of providing an example implementation of a simple communication layer based on ROS2 (there is some architecture to it that might fit into some applications, not just a simple main.cpp stuff). The ros2node library might be an useful tool to anyone wanting to implement ROS2 for their robotic applications.

The authors of the project are @adamdbrw (me), @haueck (aka rafal-kozik) and @ursereg.

We are happy to help and keep the support for the project.

SantiagoMunoz, 2016-06-10:
Hello,

Here at eProsima we have been working at measuring performance against packet loss through a shim layer on top of our new UDPv4 implementation, and we have some interesting results. 

The methodology for these tests is as follows:
   * We modified the Hello World listener-talker example to stop after publishing the 50th time, and added a timestamp to the listener so we can measure the time difference between receiving the first and last change.
   * We introduced our shim transport layer, causing losses of udp packets for data messages in a range from 0% to 90% after the discovery phase.

Our main conclusion is that FastRTPS maintains robustness in all ranges, allowing the entirety of the messages to be delivered and surfaced to the ros2 layer even in the 90% packet drop scenario. We only observed a single message loss in the whole process (possibly a malformed packet).

[image]

Given that receiving all changes is inevitable, a more interesting metric would be the time spent in delivering those 50 changes as a function of the packet drop rate, and relative to the time spent in a zero-loss scenario (to abstract away the details of the machine):

[image]

FastRTPS is also resistant to drops in Heartbeat and ACK messages, although those increase the waiting time substantially as both must succeed in sequence. In case of dealing with heavy packet loss, decreasing the heartbeatPeriod parameter in the writer attributes has a very meaningful impact in performance. By freezing the packet loss at 90, we can observe the following relationship between the heartbeat period and the relative time:

[image]

JaimeMartin, 2016-06-10:
Hi  @adamdbrw,

It would be very interesting to include in your study ROS2 over eProsima Fast RTPS. Let me know if you need help.

haueck, 2016-06-10:
@SantiagoMunoz thank you for your results! The robustness of FastRTPS is really impressive. As you mentioned, for heavy packet losses there is a problem with correct Heartbeat and ACK sequence, so time needed to establish a connection between a publisher and a subscriber increases. In our case, OpenSplice was not able to establish a connection within 3 minutes for losses 60% and more. We need to tweak heartbeatPeriod parameter to get comparable results.

adamdbrw, 2016-06-10:
Hi Jaime,

I think we should cooperate. We would for sure want to supplement our results with eProsima FastRTPS - the only issue is whether we will be able to manage our effort and deliver more than planned in the same period of time. I can try to make it happen. Can you please take a look at the project source and estimate how much effort it would take to accomodate Fast RTPS in current architecture (add a node to comm/)?

Of course we can also refer to eProsima results in our report, but for comparison sake (with OpenSplice DDS), parameters and the testing environment must be the same and well controlled.

haueck, 2016-06-10:
@dirk-thomas It is very likely that I am wrong because I am not aware of some default settings that are not exchanged. To make sure that everything is ok I created one more test: I modified rmw_qos_profile_alarm profile:
```cpp
static const rmw_qos_profile_t rmw_qos_profile_alarm =
{   
    RMW_QOS_POLICY_KEEP_LAST_HISTORY,
    999,
    RMW_QOS_POLICY_RELIABLE,
    RMW_QOS_POLICY_TRANSIENT_LOCAL_DURABILITY
}; 
```
So now I have two types of messages with the same History setting but different depth (10 and 999). I expected ROS2 nodes to exchange that information so I looked  for RTPS packets with History parameter set to KEEP_LAST_HISTORY_QOS (Wireshark filter: rtps.history.kind == 0x0). But I didn't get any packets. For OpenSplice nodes I was able to find those packets:
```
serializedData
    encapsulation kind: PL_CDR_LE (0x0003)
    encapsulation options: 0x0000
    serializedData:
        PID_PROTOCOL_VERSION
        PID_VENDOR_ID
        PID_ENDPOINT_GUID
        PID_TOPIC_NAME
            parameterId: PID_TOPIC_NAME (0x0005)
            parameterLength: 20
            topic: robot_cmd_vel
        PID_TYPE_NAME
        PID_HISTORY
            parameterId: PID_HISTORY (0x0040)
            parameterLength: 8
            Kind: KEEP_LAST_HISTORY_QOS (0x00000000)
            Depth: 10
        PID_SENTINEL

serializedData
    encapsulation kind: PL_CDR_LE (0x0003)
    encapsulation options: 0x0000
    serializedData:
        PID_PROTOCOL_VERSION
        PID_VENDOR_ID
        PID_ENDPOINT_GUID
        PID_TOPIC_NAME
            parameterId: PID_TOPIC_NAME (0x0005)
            parameterLength: 24
            topic: robot_robot_control
        PID_TYPE_NAME
        PID_DURABILITY
        PID_HISTORY
            parameterId: PID_HISTORY (0x0040)
            parameterLength: 8
            Kind: KEEP_LAST_HISTORY_QOS (0x00000000)
            Depth: 999
        PID_SENTINEL
```
Topic robot_cmd_vel: rmw_qos_profile_control. Topic robot_robot_control: profile rmw_qos_profile_alarm. Packet traces:

[ros2.pcap.zip](https://github.com/ros2/ros2/files/309101/ros2.pcap.zip)
[opensplice.pcap.zip](https://github.com/ros2/ros2/files/309103/opensplice.pcap.zip)

If you have any idea what we can check so we get the same results as for pure OpenSplice nodes, please let us know.

adamdbrw, 2016-06-14:
Thanks Jaime,
@SantiagoMunoz, could you please share how your tests were conducted? Were they on ROS2 alpha6?

JaimeMartin, 2016-06-16:
hi @adamdbrw,

we (@santiagomunoz is part of eprosima's team) used ROS2 alpha6.

SantiagoMunoz, 2016-06-16:
Hi @adamdbrw 
We used the built in talker and listener example from ROS, making modifications to stop after 50 messages and report the total time the listener takes to receive them. 
We also forced FastRTPS to use a test transport layer we have been working that we configured to lose data packets (it is added to the upcoming release that is coming out in the next few days).
It would be good to normalize the test using the same envirnonment and conditions, what tool are you using to lose packets?

haueck, 2016-06-16:
Hi @SantiagoMunoz 

We use two Docker containers with ROS nodes that talk to each other. It is possible to read a bridge configuration using brctl command and get interfaces assigned to any Docker network. When we have those two interfaces for those nodes we change their characteristics with tc-netem tool. For packet loss we run this command twice: `/sbin/tc qdisc replace dev <interface> root netem loss 5%`.

adamdbrw, 2016-06-21:
@dirk-thomas, @vmayoral, guys,

I've noticed that in current ROS2 version in rcl there qos parameters are ignored when creating subscribtion and publisher (rmw_create_subscription and rmw_create_publisher calls in rcl_subscription_init and rcl_publisher_init, in subscription.c and publisher.c respectively (in rcl).

The simple patch would be to replace &rmw_qos_profile_default with &options->qos in the parameter lists for both calls. Does this sound about right or is there a special reason why qos profiles are not supported yet?

We are just testing a patch (with opensplice).

dirk-thomas, 2016-06-21:
@wjwwood Since you wrote this code maybe you can comment. It looks like the passed in options are being ignored and only the default QoS options are set (https://github.com/ros2/rcl/blob/8cd81a5a9a35f89fbe294b3ab92549a39c369e05/rcl/src/rcl/publisher.c#L79).

wjwwood, 2016-06-21:
Yeah, looks like a bug. I have a test to make sure the given options (including the QoS) are stored in the publisher, but there is no test to assert that the QoS has been applied properly. I don't know of a good way to test that other than some sort of mocking. At the very least I can open a pr that fixes the problem:

https://github.com/ros2/rcl/pull/65

@adamdbrw Thanks for pointing that out, and sorry for any problems it caused you, hopefully we didn't waste too much of your time with this 😄. Switching to use `rcl` and not `rmw` directly only happen in the last couple of alphas, so we didn't notice it in our acceptance testing that we did before that.

I'd really like to get a test to cover this, but unfortunately, the only things I can think of is some kind of qualitative measure (which is likely to be flaky) or using something like wireshark to ensure the network data contains what we expect (which sounds messy and complicated).

adamdbrw, 2016-06-22:
@wjwwood Happy to help. As me and @haueck  are still working on ROS 2.0 evaluation, there is a chance some more patches will be proposed - should we submit them directly?

wjwwood, 2016-06-22:
@adamdbrw you're welcome to submit pull requests if you find anything you think is wrong or could be improved.

haueck, 2016-06-24:
Hi,

@SantiagoMunoz could you tell us about QoS settings that you used to get your results? This is QoS for our RobotAlarm message (the same for DataReader, DataWriter and Topic) (source code: [ddsqos.cpp](https://github.com/piappl/ros2_benchmarking/blob/master/comm/ddsnode/ddsqos.cpp)):
```cpp
dds::(pub|sub|topic)::qos::(DataWriterQos|DataReaderQos|TopicQos) qos;
qos << policy::Reliability::Reliable();
qos << policy::Durability::Transient();
qos << policy::History::KeepAll;
```
And this is a message definition (source code: [RobotMessages.idl](https://github.com/piappl/ros2_benchmarking/blob/master/comm/ddsnode/RobotMessages.idl)):
```
struct RobotAlarm
{
    long id;
    long alarm1;
    long alarm2;
};
#pragma keylist RobotAlarm
```
We removed a key because ROS2 does not support it at the moment. For those settings we start loosing packets event at the 20% loss rate. Interestingly, when we used id field as a key (unique for each message) we did not loose any packets for up 50% loss rate.

Thanks in advance!

rafal-kozik, 2016-07-04:
The original problem has been resolved by @wjwwood patch.

closed 2016-07-04 by rafal-kozik