ros2/ros2 #237: optionally use a daemon to make queries about the graph

wjwwood, 2016-06-23:
Currently the few graph query functions we have, e.g. `count_publishers(topic)`, `count_subscribers(topic)`, and `client->service_is_available()`, are based on discovery information collected by the node. An advantage to this is that nodes do not rely on a single point of failure to answer graph queries. Two downsides, however, are that all nodes need to duplicate this information and that the distributed discovery (default method of discover for DDS) takes time to propagate.

Providing an option to use a centralized daemon instead of locally collected information can save on resources when appropriate and more importantly can give timely and correct answers. The latter point is especially important for command line tools which need to get the state of the graph, report it, and close quickly. Without a daemon these tools will either have to wait an unknown amount of time for discovery information to propagate or risk reporting early but missing information.

For long running systems this doesn't matter as much, and the daemon would only be useful for reducing the footprint of each ROS process.

It's worth mentioning that this daemon would not be required for communication and normal discovery to occur. It would only allow programs to get information about the ROS graph quickly, potentially without even needing to create a node (it could ask over http and the daemon could return JSON as an example).

Mentioned here: https://github.com/ros2/ros2/issues/215#issuecomment-226589984
Also relevant: https://github.com/ros2/ros2/issues/215#issuecomment-213643708

wjwwood, 2016-06-23:
I put this issue here because it would probably touch `rmw_*`, `rmw`, `rcl`, `rclcpp`, etc...

gbiggs, 2016-06-24:
A good approach could be a graph query API that can switch between collected information or contacting the daemon for the information. Switching could be automatic (couldn't find the well-known topics, assume the daemon is not running, so use collected information), or at the request of the user. I think that this would require some initialisation when a node is first launched to ensure that it doesn't give unpredictable delays while looking for a daemon and suddenly finding it needs to collect information before it can answer.

I was originally going to suggest two APIs, one that works with collected information (or collects information on request), and one that works with a daemon (which would be built on the first API). But on thinking about this, the user-facing API would be nearly identical and I couldn't see a benefit to having two nearly identical APIs.

Either way, giving the user the option of using a daemon or not, or even no daemon and no collecting, would provide flexibility to system integrators.

wjwwood, 2016-06-24:
Yeah, I was imagining the same. Hopefully the default behavior could be use the daemon, but fallback to local discovery info if it's not there. We just need to see if that works out in practice, as you said not introducing undo system delays.

We'll definitely have a way to say "only use the daemon" or "do not use the daemon" in the API. I was imagining that as an option rather than a separate API too. I was thinking it might be done with dependency injection, where you can pass in "DaemonOnly", "LocalOnly", or "Automatic" object and the interface would use what you gave it. That also makes it easier to mock up the queries for testing.

dirk-thomas, 2017-08-03:
Just as a reference [ros2cli](https://github.com/ros2/ros2cli) implements a daemon in Python which at least exposes some of the API.

tfoote, 2018-03-08:
The daemon has been implemented.

closed 2018-03-08 by tfoote