ros2/ros2 #239: C++ coverage does not report on intra_process_demo

gerkey, 2016-06-23:
The recently added C++ coverage support doesn't report on the `intra_process_demo` package (e.g., http://ci.ros2.org/job/coverage_test/10/). I just audited that build pretty carefully and as far as I can tell, `intra_process_demo` is the only package missing that: contains C++ code, was compiled with our flags (as opposed to being third-party code that we're building), and is exercised during the test step.

Looking at the console output, it appears to be built just like the others, and the tests appear to be run. They're running indirectly, via `nosetests` and a wrapper Python script, but there are other packages doing the same thing, and they're not missing from the report.

Looking into the Jenkins workspace, I can see the `.gcno` files that are produced at build time:
```
$ find . -name "*.gcno" | grep intra_process_demo
./build/intra_process_demo/CMakeFiles/image_pipeline_with_two_image_view__rmw_opensplice_cpp.dir/src/image_pipeline/image_pipeline_with_two_image_view.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_view_node__rmw_opensplice_cpp.dir/src/image_pipeline/image_view_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/camera_node__rmw_opensplice_cpp.dir/src/image_pipeline/camera_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/camera_node__rmw_fastrtps_cpp.dir/src/image_pipeline/camera_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_view_node__rmw_fastrtps_cpp.dir/src/image_pipeline/image_view_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/cyclic_pipeline.dir/src/cyclic_pipeline/cyclic_pipeline.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_view_node.dir/src/image_pipeline/image_view_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/two_node_pipeline__rmw_opensplice_cpp.dir/src/two_node_pipeline/two_node_pipeline.cpp.gcno
./build/intra_process_demo/CMakeFiles/watermark_node__rmw_opensplice_cpp.dir/src/image_pipeline/watermark_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_pipeline_all_in_one.dir/src/image_pipeline/image_pipeline_all_in_one.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_pipeline_with_two_image_view__rmw_fastrtps_cpp.dir/src/image_pipeline/image_pipeline_with_two_image_view.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_pipeline_with_two_image_view.dir/src/image_pipeline/image_pipeline_with_two_image_view.cpp.gcno
./build/intra_process_demo/CMakeFiles/watermark_node.dir/src/image_pipeline/watermark_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/two_node_pipeline.dir/src/two_node_pipeline/two_node_pipeline.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_pipeline_all_in_one__rmw_fastrtps_cpp.dir/src/image_pipeline/image_pipeline_all_in_one.cpp.gcno
./build/intra_process_demo/CMakeFiles/cyclic_pipeline__rmw_opensplice_cpp.dir/src/cyclic_pipeline/cyclic_pipeline.cpp.gcno
./build/intra_process_demo/CMakeFiles/3.5.1/CompilerIdCXX/CMakeCXXCompilerId.gcno
./build/intra_process_demo/CMakeFiles/3.5.1/CompilerIdC/CMakeCCompilerId.gcno
./build/intra_process_demo/CMakeFiles/cyclic_pipeline__rmw_fastrtps_cpp.dir/src/cyclic_pipeline/cyclic_pipeline.cpp.gcno
./build/intra_process_demo/CMakeFiles/camera_node.dir/src/image_pipeline/camera_node.cpp.gcno
./build/intra_process_demo/CMakeFiles/two_node_pipeline__rmw_fastrtps_cpp.dir/src/two_node_pipeline/two_node_pipeline.cpp.gcno
./build/intra_process_demo/CMakeFiles/image_pipeline_all_in_one__rmw_opensplice_cpp.dir/src/image_pipeline/image_pipeline_all_in_one.cpp.gcno
./build/intra_process_demo/CMakeFiles/watermark_node__rmw_fastrtps_cpp.dir/src/image_pipeline/watermark_node.cpp.gcno
```
but there aren't any `.gcda` files that should have accumulated during the test step:
```
$ find . -name "*.gcda" | grep intra_process_demo
```

jacquelinekay, 2016-06-24:
I'm looking into this right now; I have an idea of what might have happened.

jacquelinekay, 2016-06-24:
When I run an executable from this package manually with the gcov compile flags it generates a .gcda file, but when I use nose on the corresponding Python test for the executable it does not generate a gcda file, which makes me suspect the demo is not being run on CI.

mikaelarguedas, 2017-04-05:
@gerkey bump (should we move this in backlog?)

dhood, 2017-08-08:
I think https://github.com/ros2/launch/pull/44 fixed this. The test executables for intra_process_demo have coverage output:

http://ci.ros2.org/view/nightly/job/nightly_linux_coverage/468/cobertura/src_ros2_demos_intra_process_demo_src_two_node_pipeline/

```
$ find . -name "*.gcda" | grep intra_process_demo
./ws/build/intra_process_demo/CMakeFiles/two_node_pipeline.dir/src/two_node_pipeline/two_node_pipeline.cpp.gcda
./ws/build/intra_process_demo/CMakeFiles/cyclic_pipeline.dir/src/cyclic_pipeline/cyclic_pipeline.cpp.gcda
```

From what I understand they're the only .gcda files we're expecting (they're the only tested executables) so I'll close this. If there's work left to be done here please re-open.

closed 2017-08-08 by dhood