ros2/ros2 #240: C++ coverage report uses long names

gerkey, 2016-06-23:
The C++ coverage report (#234) would be more readable if the long package names were shortened (e.g., http://ci.ros2.org/job/coverage_test/10/cobertura/). It will probably require some manually post-processing before giving the data to Cobertura.

dirk-thomas, 2016-10-17:
ros2/ci#2 removes the prefix path of the workspace. The remaining relative path is already shorter but shill not short.

Not sure how those could be trimmed further though...

clalancette, 2021-01-07:
I think these are as short as they reasonably can be at this point.  I'm going to close this out for now, but do feel free to reopen if you think there is a way to shorten these further.

closed 2021-01-07 by clalancette