ros2/ros2 #244: Jobs hanging on errors

dhood, 2016-06-24:
buildfarm jobs seem to be hanging on errors lately.

As an example, when an error occurred in http://ci.ros2.org/job/ci_linux/1551/console, I saw:

```
# BEGIN SUBSECTION: import repositories
22:58:39 (venv)
22:58:39 ==> curl -sk ci_scripts_rti_connext_binary -o ros2.repos
22:58:39 <== '. ../venv/bin/activate && curl -sk ci_scripts_rti_connext_binary -o ros2.repos' exited with return code '6'
22:58:41 Build step 'Execute shell' marked build as failure
22:58:41 [ssh-agent] Stopped.
22:58:41 [WARNINGS] Skipping publisher since build result is FAILURE
22:58:41 Publish Cobertura Coverage Report is waiting for a checkpoint on ci_linux #1547
```

and then once the job was canceled:
```
# BEGIN SUBSECTION: import repositories
22:58:39 (venv)
22:58:39 ==> curl -sk ci_scripts_rti_connext_binary -o ros2.repos
22:58:39 <== '. ../venv/bin/activate && curl -sk ci_scripts_rti_connext_binary -o ros2.repos' exited with return code '6'
22:58:41 Build step 'Execute shell' marked build as failure
22:58:41 [ssh-agent] Stopped.
22:58:41 [WARNINGS] Skipping publisher since build result is FAILURE
22:58:41 Publish Cobertura Coverage Report is waiting for a checkpoint on ci_linux #1547
23:06:37 ERROR: Step ‘Publish Cobertura Coverage Report’ aborted due to exception:
23:06:37 java.lang.InterruptedException
23:06:37 at java.lang.Object.wait(Native Method)
23:06:37 at java.lang.Object.wait(Object.java:503)
23:06:37 at hudson.model.Run$RunExecution$CheckpointSet.waitForCheckPoint(Run.java:1588)
23:06:37 at hudson.model.Run.waitForCheckpoint(Run.java:1541)
23:06:37 at hudson.model.CheckPoint.block(CheckPoint.java:157)
23:06:37 at hudson.tasks.BuildStepMonitor$3.perform(BuildStepMonitor.java:41)
23:06:37 at hudson.model.AbstractBuild$AbstractBuildExecution.perform(AbstractBuild.java:782)
23:06:37 at hudson.model.AbstractBuild$AbstractBuildExecution.performAllBuildSteps(AbstractBuild.java:723)
23:06:37 at hudson.model.Build$BuildExecution.post2(Build.java:185)
23:06:37 at hudson.model.AbstractBuild$AbstractBuildExecution.post(AbstractBuild.java:668)
23:06:37 at hudson.model.Run.execute(Run.java:1763)
23:06:37 at hudson.model.FreeStyleBuild.run(FreeStyleBuild.java:43)
23:06:37 at hudson.model.ResourceController.execute(ResourceController.java:98)
23:06:37 at hudson.model.Executor.run(Executor.java:410)
23:06:37 [xUnit] [INFO] - Starting to record.
23:06:37 [xUnit] [INFO] - Processing CTest-Version N/A (default)
23:06:37 [xUnit] [INFO] - [CTest-Version N/A (default)] - No test report file(s) were found with the pattern 'ws/build/*/Testing/*/Test.xml' relative to '/home/jenkins/workspace/ci_linux' for the testing framework 'CTest-Version N/A (default)'. Did you enter a pattern relative to the correct directory? Did you generate the result report(s) for 'CTest-Version N/A (default)'?
23:06:37 [xUnit] [WARNING] - No test reports found for the metric 'CTest' with the resolved pattern 'ws/build/*/Testing/*/Test.xml'.
23:06:37 [xUnit] [INFO] - Skipping the metric tool processing.
23:06:37 [xUnit] [INFO] - Processing GoogleTest-1.6
23:06:37 [xUnit] [INFO] - [GoogleTest-1.6] - No test report file(s) were found with the pattern 'ws/build/*/test_results/**/*.gtest.xml' relative to '/home/jenkins/workspace/ci_linux' for the testing framework 'GoogleTest-1.6'. Did you enter a pattern relative to the correct directory? Did you generate the result report(s) for 'GoogleTest-1.6'?
23:06:37 [xUnit] [WARNING] - No test reports found for the metric 'GoogleTest' with the resolved pattern 'ws/build/*/test_results/**/*.gtest.xml'.
23:06:37 [xUnit] [INFO] - Skipping the metric tool processing.
23:06:37 [xUnit] [INFO] - Processing JUnit
23:06:37 [xUnit] [INFO] - [JUnit] - No test report file(s) were found with the pattern 'ws/build/*/test_results/**/*.xunit.xml' relative to '/home/jenkins/workspace/ci_linux' for the testing framework 'JUnit'. Did you enter a pattern relative to the correct directory? Did you generate the result report(s) for 'JUnit'?
23:06:37 [xUnit] [WARNING] - No test reports found for the metric 'JUnit' with the resolved pattern 'ws/build/*/test_results/**/*.xunit.xml'.
23:06:37 [xUnit] [INFO] - Skipping the metric tool processing.
23:06:37 [xUnit] [INFO] - Processing CTest-Version N/A (default)
23:06:37 [xUnit] [INFO] - [CTest-Version N/A (default)] - No test report file(s) were found with the pattern 'work space/build space/*/Testing/*/Test.xml' relative to '/home/jenkins/workspace/ci_linux' for the testing framework 'CTest-Version N/A (default)'. Did you enter a pattern relative to the correct directory? Did you generate the result report(s) for 'CTest-Version N/A (default)'?
23:06:37 [xUnit] [WARNING] - No test reports found for the metric 'CTest' with the resolved pattern 'work space/build space/*/Testing/*/Test.xml'.
23:06:37 [xUnit] [INFO] - Skipping the metric tool processing.
23:06:37 [xUnit] [INFO] - Processing GoogleTest-1.6
23:06:37 [xUnit] [INFO] - [GoogleTest-1.6] - No test report file(s) were found with the pattern 'work space/build space/*/test_results/**/*.gtest.xml' relative to '/home/jenkins/workspace/ci_linux' for the testing framework 'GoogleTest-1.6'. Did you enter a pattern relative to the correct directory? Did you generate the result report(s) for 'GoogleTest-1.6'?
23:06:37 [xUnit] [WARNING] - No test reports found for the metric 'GoogleTest' with the resolved pattern 'work space/build space/*/test_results/**/*.gtest.xml'.
23:06:37 [xUnit] [INFO] - Skipping the metric tool processing.
23:06:37 [xUnit] [INFO] - Processing JUnit
23:06:37 [xUnit] [INFO] - [JUnit] - No test report file(s) were found with the pattern 'work space/build space/*/test_results/**/*.xunit.xml' relative to '/home/jenkins/workspace/ci_linux' for the testing framework 'JUnit'. Did you enter a pattern relative to the correct directory? Did you generate the result report(s) for 'JUnit'?
23:06:37 [xUnit] [WARNING] - No test reports found for the metric 'JUnit' with the resolved pattern 'work space/build space/*/test_results/**/*.xunit.xml'.
23:06:37 [xUnit] [INFO] - Skipping the metric tool processing.
23:06:37 [xUnit] [INFO] - There are errors when processing test results.
23:06:37 [xUnit] [INFO] - Skipping tests recording.
23:06:37 Finished: FAILURE 
```

this has also happened on OS X yesterday: http://ci.ros2.org/job/ci_osx/1192/console#console-section-34

dirk-thomas, 2016-06-24:
Your job finished before another earlier build. In that case the Cobertura Coverage Report needs to wait for the older builds to finish first. If you abort the job you basically cancel the coverage report. So what you describe is expected behavior.

The OS X build from yesterday was hanging on a test which ran infinitely (which has been fixed upstream in the meantime) and is unrelated to the other case.

dhood, 2016-06-24:
ok, that's clearer now. my job was waiting for http://ci.ros2.org/job/ci_linux/1547/ I guess, which was running when my job started. But that job was also hanging/running infinitely, and you aborted it. normally this would let mine finish, but I had already canceled it in the meantime.

thank you

closed 2016-06-24 by dhood

mikaelarguedas, 2016-06-25:
Maybe it's not related but I noticed that happening whenever another
(previous) unfinished CI job is running. Jenkins is waiting for the
previous job to finish to compute statistics. Maybe that's what you noticed
and is not link to job failure or success ?

mikaelarguedas, 2016-06-25:
Just received the follow-up email. Sorry for the noise

gerkey, 2016-06-27:
I was also confused by that behavior of the Cobertura plugin. Turns out that it's a known bug:

https://issues.jenkins-ci.org/browse/JENKINS-26823

dirk-thomas, 2016-06-27:
I updated the Jenkins plugin to the latest release (1.9.8) which seems to have fixed the problem:

* http://ci.ros2.org/job/coverage_test/12/
finished before
* http://ci.ros2.org/job/coverage_test/11/

(see https://github.com/jenkinsci/cobertura-plugin # 42)