ros2/ros2 #246: Demos not working with connext binaries off website

dhood, 2016-06-25:
Builds using Connext binaries off the RTI [website](http://www.rti.com/downloads/connext-files.html) (labeled `rti_connext_dds-5.2.3-eval-x64Linux3gcc4.8.2.run`) aren't currently working

[Linking issues](https://github.com/ros2/rcl/issues/66) are fixed, but there must be more differences between the version on the website and our debian packages.

`listener` just says: `I heard: []`, for example, while other outputs are a bit more verbose:
```
dhood@osrf-esteve:~/ros2_ws [ros2_ws]$ ./build_isolated/rclcpp_examples/set_and_get_parameters__rmw_connext_dynamic_cpp 
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 02-Jul-2016 See www.rti.com for more information.
*** Error in `./build_isolated/rclcpp_examples/set_and_get_parameters__rmw_connext_dynamic_cpp': munmap_chunk(): invalid pointer: 0x0000000002ebd5f0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x7f8cc4632725]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x1a8)[0x7f8cc463ec18]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN9__gnu_cxx13new_allocatorIN14rcl_interfaces3msg10Parameter_ISaIvEEEE10deallocateEPS5_m+0x20)[0x7f8cc5230e18]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZNSt16allocator_traitsISaIN14rcl_interfaces3msg10Parameter_ISaIvEEEEE10deallocateERS5_PS4_m+0x2b)[0x7f8cc522dd36]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZNSt12_Vector_baseIN14rcl_interfaces3msg10Parameter_ISaIvEEESaIS4_EE13_M_deallocateEPS4_m+0x32)[0x7f8cc522a446]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZNSt12_Vector_baseIN14rcl_interfaces3msg10Parameter_ISaIvEEESaIS4_EED1Ev+0x52)[0x7f8cc5235fa6]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZNSt6vectorIN14rcl_interfaces3msg10Parameter_ISaIvEEESaIS4_EED1Ev+0x41)[0x7f8cc52346e1]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN14rcl_interfaces3srv22SetParameters_Request_ISaIvEED1Ev+0x18)[0x7f8cc523db38]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZNSt15_Sp_counted_ptrIPN14rcl_interfaces3srv22SetParameters_Request_ISaIvEEELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv+0x22)[0x7f8cc525c5c8]
./build_isolated/rclcpp_examples/set_and_get_parameters__rmw_connext_dynamic_cpp(_ZNSt16_Sp_counted_baseILN9__gnu_cxx12_Lock_policyE2EE10_M_releaseEv+0x42)[0x409d72]
./build_isolated/rclcpp_examples/set_and_get_parameters__rmw_connext_dynamic_cpp(_ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE2EED2Ev+0x27)[0x4095bd]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZNSt12__shared_ptrIvLN9__gnu_cxx12_Lock_policyE2EED1Ev+0x1c)[0x7f8cc51d913c]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZNSt10shared_ptrIvED1Ev+0x18)[0x7f8cc51d9158]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN6rclcpp8executor8Executor15execute_serviceESt10shared_ptrINS_7service11ServiceBaseEE+0x1ae)[0x7f8cc51dc278]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN6rclcpp8executor8Executor22execute_any_executableESt10shared_ptrINS0_13AnyExecutableEE+0x1a5)[0x7f8cc51dbcaf]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN6rclcpp8executor8Executor9spin_onceENSt6chrono8durationIlSt5ratioILl1ELl1000000000EEEE+0xd8)[0x7f8cc51db944]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN6rclcpp8executor8Executor26spin_until_future_completeISt6vectorIN14rcl_interfaces3msg20SetParametersResult_ISaIvEEESaIS8_EESt5ratioILl1ELl1000EEEENS0_16FutureReturnCodeERSt13shared_futureIT_ENSt6chrono8durationIlT0_EE+0xe2)[0x7f8cc5217fec]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN6rclcpp9executors31spin_node_until_future_completeISt6vectorIN14rcl_interfaces3msg20SetParametersResult_ISaIvEEESaIS7_EESt5ratioILl1ELl1000EEEENS_8executor16FutureReturnCodeERNSC_8ExecutorESt10shared_ptrINS_4node4NodeEERSt13shared_futureIT_ENSt6chrono8durationIlT0_EE+0x81)[0x7f8cc5213328]
/home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_dynamic_cpp.so(_ZN6rclcpp16parameter_client20SyncParametersClient14set_parametersERKSt6vectorINS_9parameter16ParameterVariantESaIS4_EE+0xd5)[0x7f8cc5208eaf]
./build_isolated/rclcpp_examples/set_and_get_parameters__rmw_connext_dynamic_cpp[0x408d16]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f8cc45db830]
./build_isolated/rclcpp_examples/set_and_get_parameters__rmw_connext_dynamic_cpp[0x4088e9]
```

I've made the `ci_scripts_rti_connext_binary` branch which uses connext off the website (https://github.com/ros2/ros2/pull/245). here's an example job with branches at master: http://ci.ros2.org/job/ci_linux/1554/

I ran a job with branches at alpha-6 to check if it was changes that we'd made that were responsible, but similar errors occur: http://ci.ros2.org/job/ci_linux/1555/consoleFull (alpha-6 was working fine with both `rti_connext_dds-5.2.0-eval-x64Linux3gcc4.8.2.run` and `rti_connext_dds-5.2.0-eval-x64Linux3.xgcc4.6.3.run` off the website)

dhood, 2016-06-28:
this might be related to the xenial switch.
http://ci.ros2.org/job/ci_linux/1581/ does not give these errors and uses 5.2.3 off the website with alpha 6 code (as did http://ci.ros2.org/job/ci_linux/1555/), difference is it's using trusty instead of xenial

jacquelinekay, 2016-06-29:
I still suspect this is due to the C++ ABI differences between gcc 4 and gcc 5, especially since the error. (https://gcc.gnu.org/onlinedocs/libstdc++/manual/using_dual_abi.html)

Shouldn't the following sources use the `Connext_GLIBCXX_USE_CXX11_ABI_ZERO` switch since they compile to libraries that link against Connext?

I believe the only headers that would affect Connext Dynamic are the rmw_connext_shared_cpp headers. But using the switch in the sources for all libraries that link against Connext seems like the safest way to go to prevent ABI leakage.

https://github.com/ros2/rmw_connext/blob/master/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/types.hpp
https://github.com/ros2/rmw_connext/blob/master/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/shared_functions.hpp
https://github.com/ros2/rmw_connext/blob/master/rmw_connext_shared_cpp/src/shared_functions.cpp
https://github.com/ros2/rmw_connext/blob/master/rmw_connext_cpp/src/functions.cpp
https://github.com/ros2/rmw_connext/blob/master/rosidl_typesupport_connext_cpp/resource/msg__type_support.cpp.em

dirk-thomas, 2016-06-29:
Only very few code in Connext actually uses API which is affected by the dual ABI. Can you argue why you think these specific five files need the same flag? For some I don't see anything in the file which would be affected by the flag.

E.g. I don't see why `rosidl_typesupport_connext_cpp/resource/msg__type_support.cpp.em` would need this. It is not dealing with service at all.

dhood, 2016-06-30:
this seems to only affect connext_dynamic. 

there don't seem to be any new errors for connext_static when using the binaries off the RTI website: http://ci.ros2.org/job/ci_linux/1594/ (only connext static, using debs) vs 
http://ci.ros2.org/job/ci_linux/1596/ (only connext static, using connext off RTI website)

since we're not supporting connext_dynamic, I'm not going to investigate this any further unless there's motivation to continue looking into it

dirk-thomas, 2016-06-30:
I think it would be very valuable to find and fix the reason why many but not all of the `rmw_connext_dynamic_cpp` related tests fail on Windows (they account for ~75% of our failing tests). But I might be alone with that opinion.

wjwwood, 2016-06-30:
> I think it would be very valuable to find and fix the reason why many but not all of the `rmw_connext_dynamic_cpp` related tests fail on Windows (they account for ~75% of our failing tests). 

I'm not sure that's the best use of time at the moment. However, it's good that we know about it and that we have a detailed description here.

@dhood can you change the title to indicate that this only affects connext dynamic?

> But I might be alone with that opinion.

I think it's valuable, but not enough to justify doing it right now. Later, even at the risk of potentially never resolving it, is okay by me.

dhood, 2016-09-21:
This has been fixed by https://github.com/ros2/rmw_connext/pull/192.

Linux job using connext dynamic binaries off the rti website: [image]

closed 2016-09-21 by dhood