ros2/ros2 #253: Ordering of ament_target_dependencies influences typesupport selected

dhood, 2016-07-01:
Typesupport selection is currently sensitive to the ordering of the dependencies in `ament_target_dependencies`.

For example, if 
```
  ament_target_dependencies(${target}${target_suffix}
    "rclcpp${target_suffix}"
    "std_msgs"
    "example_interfaces")
```
is changed to 
```
  ament_target_dependencies(${target}${target_suffix}
    "example_interfaces"
    "rclcpp${target_suffix}"
    "std_msgs")
```
[here](https://github.com/ros2/examples/blob/035c2e923dbecf0bb0238b7b9887b354ffb0af10/rclcpp_examples/CMakeLists.txt#L23), the following error occurs:

```
dhood@osrf-esteve:~/ros2_ws [ros2_ws]$ ./build_isolated/rclcpp_examples/talker__rmw_connext_cpp 
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 02-Jul-2016 See www.rti.com for more information.
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not create publisher: type support handle implementation 'introspection_cpp' (0x7fe8e86b8669) does not match valid type supports ('connext_static_cpp' (0x7fe8e79e0649), 'connext_static_c' (0x7fe8e7be26e9)), at /home/dhood/ros2_ws/src/ros2/rmw_connext/rmw_connext_cpp/src/functions.cpp:161, at /home/dhood/ros2_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:81
Aborted (core dumped)
```

----

In this case, the `rosidl_typesupport_introspection_cpp` include paths occur before the `rosidl_typesupport_connect_cpp` include paths (as of https://github.com/ros2/rosidl/pull/158/commits/1e4b0741cbd82da5e62b0822ab24f293ede12ba4):

With `example_interfaces` package listed as dependency before `rclcpp*`:
```
Scanning dependencies of target talker__rmw_connext_cpp
make[2]: Leaving directory '/home/dhood/ros2_ws/build_isolated/rclcpp_examples'
/usr/bin/make -f CMakeFiles/talker__rmw_connext_cpp.dir/build.make CMakeFiles/talker__rmw_connext_cpp.dir/build
make[2]: Entering directory '/home/dhood/ros2_ws/build_isolated/rclcpp_examples'
[ 99%] Building CXX object CMakeFiles/talker__rmw_connext_cpp.dir/src/topics/talker.cpp.o
/usr/bin/c++   -DRTI_LINUX -DRTI_UNIX -I/home/dhood/ros2_ws/install_isolated/builtin_interfaces/include -I/home/dhood/ros2_ws/install_isolated/rosidl_generator_cpp/include -I/home/dhood/ros2_ws/install_isolated/rosidl_generator_c/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/include/rosidl_typesupport_introspection_c/impl -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/include/rosidl_typesupport_introspection_cpp/impl -I/home/dhood/ros2_ws/install_isolated/example_interfaces/include -I/home/dhood/ros2_ws/install_isolated/rclcpp/include -I/home/dhood/ros2_ws/install_isolated/rcl_interfaces/include -I/home/dhood/ros2_ws/install_isolated/rcl/include -I/home/dhood/ros2_ws/install_isolated/rmw/include -I/home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/include -I/home/dhood/rti_connext_dds-5.2.3/include -I/home/dhood/rti_connext_dds-5.2.3/include/ndds -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp/include/rosidl_typesupport_connext_cpp/impl -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_c/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_c/include/rosidl_typesupport_connext_c/impl -I/home/dhood/ros2_ws/install_isolated/std_msgs/include  -std=c++11 -Wall -Wextra   -o CMakeFiles/talker__rmw_connext_cpp.dir/src/topics/talker.cpp.o -c /home/dhood/ros2_ws/src/ros2/examples/rclcpp_examples/src/topics/talker.cpp
[100%] Linking CXX executable talker__rmw_connext_cpp
/usr/bin/cmake -E cmake_link_script CMakeFiles/talker__rmw_connext_cpp.dir/link.txt --verbose=1
/usr/bin/c++    -std=c++11 -Wall -Wextra   CMakeFiles/talker__rmw_connext_cpp.dir/src/topics/talker.cpp.o  -o talker__rmw_connext_cpp  -L/home/dhood/rti_connext_dds-5.2.3/lib/x64Linux3gcc4.8.2 /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rclcpp/lib/librclcpp__rmw_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rcl/lib/librcl__rmw_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib/librmw_connext_shared_cpp.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_c/lib/librosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/rmw/lib/librmw.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp/lib/librosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rmw_connext_cpp/lib/librmw_connext_cpp.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib/librosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/lib/librosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib/librosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/std_msgs/lib/libstd_msgs__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/std_msgs/lib/libstd_msgs__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/std_msgs/lib/libstd_msgs__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/std_msgs/lib/libstd_msgs__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/std_msgs/lib/libstd_msgs__rosidl_typesupport_connext_cpp.so -lnddsc -lnddscore -lnddscpp -lrticonnextmsgcpp -lpthread /usr/lib/x86_64-linux-gnu/libdl.so -Wl,-rpath,/home/dhood/ros2_ws/install_isolated/example_interfaces/lib:/home/dhood/ros2_ws/install_isolated/rclcpp/lib:/home/dhood/ros2_ws/install_isolated/rcl/lib:/home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib:/home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_c/lib:/home/dhood/ros2_ws/install_isolated/rmw/lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp/lib:/home/dhood/ros2_ws/install_isolated/rmw_connext_cpp/lib:/home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib:/home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib:/home/dhood/ros2_ws/install_isolated/std_msgs/lib:/home/dhood/rti_connext_dds-5.2.3/lib/x64Linux3gcc4.8.2: 
make[2]: Leaving directory '/home/dhood/ros2_ws/build_isolated/rclcpp_examples'
[100%] Built target talker__rmw_connext_cpp
```
With `example_interfaces` package listed as dependency after `rclcpp*`:
```
Scanning dependencies of target talker__rmw_connext_cpp
[... cut ...] /lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib:/home/dhood/ros2_ws/install_isolated/example_interfaces/lib:/home/dhood/rti_connext_dds-5.2.3/lib/x64Linux3gcc4.8.2: 
make[2]: Leaving directory '/home/dhood/ros2_ws/build_isolated/rclcpp_examples'
[100%] Built target talker__rmw_connext_cpp
```

Also affected: https://github.com/ros2/demos/blob/ecd4ef52f6537da5fb08289d93634b4b4e3511fd/pendulum_control/CMakeLists.txt#L36 (see https://github.com/ros2/demos/pull/74)

wjwwood, 2016-07-01:
Thanks for the detailed issue.

dhood, 2016-09-14:
Explanation of the issue at https://github.com/ros2/ros1_bridge/pull/27#issuecomment-233327633

dirk-thomas, 2016-09-14:
The ticket describes the current behavior of the system. Just to clarify: can you please describe what the goal of the ticket is. Is it to make the dependency order not have any side effect?

dhood, 2016-09-14:
This ticket is just to document this behaviour, not necessarily to propose a way forward. I was investigating the linking in July because I was fixing a bug (https://github.com/ros2/demos/pull/74) that was introduced because the effect of the dependency order was not anticipated. As such this ticket is to track the issue until it is resolved either by changing the behaviour or documenting it sufficiently

dhood, 2016-09-14:
(I mean unexpected in the 'user' sense, not the linking sense)

dirk-thomas, 2017-01-03:
This should not be the case anymore with the new typesupport (ros2/ros2#302) since there is no include directory based selection happening anymore.

closed 2017-01-03 by dirk-thomas

mikaelarguedas, 2017-01-05:
FYI: ros1_bridge also needed an update https://github.com/ros2/ros1_bridge/commit/ffe97cb6b40bd34a236e6a92ad6344d62dfa1d53