ros2/ros2 #258: rostopic like tools

shengwen1997, 2016-08-12:
Hello, I am interested about this ROS2 project, and hope to investigate more.
I would like to know if there is any tools like rostopic for ROS2?

Thanks!

dirk-thomas, 2016-08-12:
Currently none of these helpful tools exist yet. The goal is to write them in Python (since command line tools in C++ are much more effort) but the Python client library doesn't yet provide all the required functions yet. Also some rmw implementations lack the necessary introspection features atm.

But work on at least some of these tools has started. Please see #197 for the ongoing discussion as well as references to pull requests.

For further discussions and questions like this the [mailing list](https://groups.google.com/forum/#!forum/ros-sig-ng-ros) is probably better suited.

closed 2016-08-12 by dirk-thomas

dirk-thomas, 2017-06-14:
Please find the latest tools in this repo: https://github.com/ros2/ros2cli