ros2/ros2 #259: alpha 7 binary release includes bad python hashbang lines

jonbinney, 2016-08-21:
I downloaded ros2-alpha7-package-linux-opensplice.tar.bz2 from https://github.com/ros2/ros2/releases

Looks like it is hardcoded to use the python in the buildfarm's virtualenv, which means that the python scripts (such as ament) from the binary can't be used as is.

For example in `/home/jbinney/ws/bridge/ros2-linux/bin/ament`, the first line is:
```
#!/home/rosbuild/ci_scripts/venv/bin/python
```

Manually hanging that to `/usr/bin/env python3` appears to fix the problem. Are the python scripts in the binary installation supposed to be usable?

wjwwood, 2016-08-21:
Yeah, we're aware of this issue. It has to do with our "packaging":

https://github.com/ros2/ros2/issues/141

The tutorials note that you need to invoke the python scripts with `python3` if you are using the binary:

https://github.com/ros2/ros2/wiki/Python-Programming#run-the-examples

If you write a new Python package and build it from source then this won't be an issue, or if you were to build ros2 from source. I'm not sure if it is worth fixing for now, since this problem is handled by package managers usually, for example Debian packaging addresses this problem some how.

dirk-thomas, 2016-08-22:
I will close this a duplicate of #141.

closed 2016-08-22 by dirk-thomas