ros2/ros2 #261: internal compiler error: Killed (program cc1plus)

Harin55, 2016-08-25:
While building the entire ros_ws on a Raspberry 2 the following error occours:

```
[ 92%] Building C object CMakeFiles/test_communication__rosidl_typesupport_opensplice_c__pyext.dir/rosidl_generator_py/test_communication/msg/_u_int32_s.c.o
[ 92%] Linking C shared library rosidl_generator_py/test_communication/test_communication_s__rosidl_typesupport_opensplice_c.cpython-35m-arm-linux-gnueabihf.so
[ 92%] Built target test_communication__rosidl_typesupport_opensplice_c__pyext
Scanning dependencies of target test_subscription_valid_data_cpp__rmw_opensplice_cpp
[ 92%] Building CXX object CMakeFiles/test_subscription_valid_data_cpp__rmw_opensplice_cpp.dir/test/test_subscription_valid_data.cpp.o
[ 93%] Linking CXX executable test_replier_cpp__rmw_opensplice_cpp
[ 93%] Built target test_replier_cpp__rmw_opensplice_cpp
Scanning dependencies of target test_subscription_valid_data_cpp__rmw_fastrtps_cpp
[ 94%] Building CXX object CMakeFiles/test_subscription_valid_data_cpp__rmw_fastrtps_cpp.dir/test/test_subscription_valid_data.cpp.o
[ 95%] Linking CXX executable test_subscription_valid_data_cpp__rmw_opensplice_cpp
[ 95%] Built target test_subscription_valid_data_cpp__rmw_opensplice_cpp
Scanning dependencies of target test_publisher_subscriber_cpp__rmw_fastrtps_cpp
[ 95%] Linking CXX executable test_subscription_valid_data_cpp__rmw_fastrtps_cpp
[ 95%] Building CXX object CMakeFiles/test_publisher_subscriber_cpp__rmw_fastrtps_cpp.dir/test/test_publisher_subscriber.cpp.o
[ 95%] Built target test_subscription_valid_data_cpp__rmw_fastrtps_cpp
Scanning dependencies of target test_replier_cpp__rmw_fastrtps_cpp
[ 96%] Building CXX object CMakeFiles/test_replier_cpp__rmw_fastrtps_cpp.dir/test/test_replier.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
CMakeFiles/test_subscriber_cpp__rmw_opensplice_cpp.dir/build.make:62: recipe for target 'CMakeFiles/test_subscriber_cpp__rmw_opensplice_cpp.dir/test/test_subscriber.cpp.o' failed
make[2]: *** [CMakeFiles/test_subscriber_cpp__rmw_opensplice_cpp.dir/test/test_subscriber.cpp.o] Error 4
CMakeFiles/Makefile2:106: recipe for target 'CMakeFiles/test_subscriber_cpp__rmw_opensplice_cpp.dir/all' failed
make[1]: *** [CMakeFiles/test_subscriber_cpp__rmw_opensplice_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 96%] Linking CXX executable test_replier_cpp__rmw_fastrtps_cpp
[ 96%] Built target test_replier_cpp__rmw_fastrtps_cpp
[ 96%] Linking CXX executable test_publisher_cpp__rmw_opensplice_cpp
[ 96%] Built target test_publisher_cpp__rmw_opensplice_cpp
[ 97%] Linking CXX executable test_publisher_subscriber_cpp__rmw_fastrtps_cpp
[ 97%] Built target test_publisher_subscriber_cpp__rmw_fastrtps_cpp
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```
The examination shows, that the Rasp runs out of RAM. After following the second method described here http://www.thegeekstuff.com/2010/08/how-to-add-swap-space/ to add more 4 GB swap, the build runs without this error. Looking with `free` show in worst case this:
```
pi:~$ free --mega
              total        used        free      shared  buff/cache   available
Mem:            947         919          14           0          13          11
Swap:          4296         451        3845
```
The build needs round about 4,5 h.

May you can suggest these step on your wiki for other ones who interested to build ROS2 on a Raspberry.

dirk-thomas, 2016-08-25:
Ca you please appending a subsection with these information to https://github.com/ros2/ros2/wiki/Linux-Development-Setup Thanks!

gerkey, 2016-08-25:
I also recommend adding `--make-flags -j1 --` to your invocation of `ament.py`.

Harin55, 2016-08-25:
@dirk-thomas Do I understand you right that I should write something on your page? If yes - ok I will do it.

@gerkey I don't understand this. I'm a newbie to `ament` and Python. For me it's really hard to understand what's happen there. I'm usually working with `make` :D

gerkey, 2016-08-25:
Ah, sorry for the opaque advice. Those flags are just telling `ament` to pass `-j1` to `make`. That way you avoid any parallel compile steps (by default, `ament` will tell `make` to run multiple compile/link steps in parallel according to how many cores you have available), which might slow down the build but will limit your potential memory consumption (and if it avoids having to dip into swap, then the build might actually be faster).

Harin55, 2016-08-26:
Build Performance on different maschines/set ups:

- - - -
__Raspberry Pi 2 - Raspbian, Quad Core (ARM Cortex-A7 @ 900MHz), 1024 MB RAM, 4096 MB swap__  
running with `--make-flags -j1`

```
	Command being timed: "src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --make-flags -j1"
	User time (seconds): 33164.02
	System time (seconds): 1560.20
	Percent of CPU this job got: 98%
	Elapsed (wall clock) time (h:mm:ss or m:ss): 9:45:12
	Average shared text size (kbytes): 0
	Average unshared data size (kbytes): 0
	Average stack size (kbytes): 0
	Average total size (kbytes): 0
	Maximum resident set size (kbytes): 379092
	Average resident set size (kbytes): 0
	Major (requiring I/O) page faults: 354
	Minor (reclaiming a frame) page faults: 47740772
	Voluntary context switches: 1049855
	Involuntary context switches: 482091
	Swaps: 0
	File system inputs: 170240
	File system outputs: 3477064
	Socket messages sent: 0
	Socket messages received: 0
	Signals delivered: 0
	Page size (bytes): 4096
	Exit status: 0
```

- - - -
__VirtualBox - Debian, one CPU (i7-4720HQ CPU @ 2.60GHz), 2048 MB RAM__

    Command being timed: "src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install"
	User time (seconds): 1620.24
	System time (seconds): 132.91
	Percent of CPU this job got: 94%
	Elapsed (wall clock) time (h:mm:ss or m:ss): 30:48.95
	Average shared text size (kbytes): 0
	Average unshared data size (kbytes): 0
	Average stack size (kbytes): 0
	Average total size (kbytes): 0
	Maximum resident set size (kbytes): 408620
	Average resident set size (kbytes): 0
	Major (requiring I/O) page faults: 248
	Minor (reclaiming a frame) page faults: 43734584
	Voluntary context switches: 2016326
	Involuntary context switches: 2026219
	Swaps: 0
	File system inputs: 63664
	File system outputs: 3177248
	Socket messages sent: 0
	Socket messages received: 0
	Signals delivered: 0
	Page size (bytes): 4096
	Exit status: 0

dirk-thomas, 2016-10-14:
I have added a note to the wiki page: https://github.com/ros2/ros2/wiki/Linux-Development-Setup/_compare/d4b9a6dae6cb9844a76e2778cdde5eb0814baff7...fd9afee62c551ffa7bd321fed0be582d8c929f3e

closed 2016-10-14 by dirk-thomas