ros2/ros2 #262: Update install directions for Alpha 7

mkhansen-intel, 2016-08-25:
It looks like the Installation directions were not updated for Alpha 7. Since Alpha 7 now splits the binary packages into two options and includes support for Ubuntu 16.04, the install directions should be updated. Thanks.

slremy, 2016-08-29:
As a new person entering the fray, I think there'd also need to be some guidance in the installation instructions on avoiding the barrage of ` /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version GLIBCXX_3.4.21 ` messages that a noob like me is going to have to sift through.

wjwwood, 2016-09-02:
@mkhansen-intel I added a note that the file name might be different and updated our binary installation page for Linux to note that they are for 16.04 only (and that previous alpha versions were 14.04).

@slremy I'm not sure what messages you're referring to. I'd suggest opening your own issue or asking about it on our mailing list. Make sure to include more details and more text context if you post an error or warning.

closed 2016-09-02 by wjwwood