ros2/ros2 #264: Some connections between different machines can't established

Harin55, 2016-08-27:
Hello,

I build up a small network with an Raspberry Pi (Raspbian Jessie) connected to an router via LAN and an host PC (Ubuntu Gnome 16.04) connected to the same router via WLAN. Inside the host are  some virtual machines installed with Debian Jessie and connected via a bridged network.

I found that the Rasp doesn't connect to the VM's but the VM's are in contact. Can you give me some suggestions to find out what the reason is to this behaviour?

I do some studies using the talker and listener from the sources:

Talker on Raspberry Pi
----------------------

> all ros users are on virtual machines

- - -

    pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
    pi:bin$ ./talker

      > pi@192.168.1.40
      > pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > pi:bin$ ./listener                                  > yes
      > pi:bin$ ./listener__rmw_opensplice_cpp              > yes
      > pi:bin$ ./listener__rmw_fastrtps_cpp                > yes after restarting talker
      > pi:bin$ ./listener_best_effort                      > yes
      > pi:bin$ ./listener_best_effort__rmw_opensplice_cpp  > yes
      > pi:bin$ ./listener_best_effort__rmw_fastrtps_cpp    > yes

      > ros@192.168.1.20
      > ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > ros:bin$ ./listener                                 > no
      > ros:bin$ ./listener__rmw_opensplice_cpp             > no
      > ros:bin$ ./listener__rmw_fastrtps_cpp               > no
      > ros:bin$ ./listener_best_effort                     > no
      > ros:bin$ ./listener_best_effort__rmw_opensplice_cpp > no
      > ros:bin$ ./listener_best_effort__rmw_fastrtps_cpp   > no

- - -

    pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
    pi:bin$ ./talker__rmw_opensplice_cpp

      > pi@192.168.1.40
      > pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > pi:bin$ ./listener                                  > yes after restarting talker
      > pi:bin$ ./listener__rmw_opensplice_cpp              > yes
      > pi:bin$ ./listener__rmw_fastrtps_cpp                > no
      > pi:bin$ ./listener_best_effort                      > yes
      > pi:bin$ ./listener_best_effort__rmw_opensplice_cpp  > yes
      > pi:bin$ ./listener_best_effort__rmw_fastrtps_cpp    > yes

      > ros@192.168.1.20
      > ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > ros:bin$ ./listener                                 > no
      > ros:bin$ ./listener__rmw_opensplice_cpp             > no
      > ros:bin$ ./listener__rmw_fastrtps_cpp               > no
      > ros:bin$ ./listener_best_effort                     > no
      > ros:bin$ ./listener_best_effort__rmw_opensplice_cpp > no
      > ros:bin$ ./listener_best_effort__rmw_fastrtps_cpp   > no

- - -

    pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
    pi:bin$ ./talker__rmw_fastrtps_cpp

      > pi@192.168.1.40
      > pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > pi:bin$ ./listener                                  > yes
      > pi:bin$ ./listener__rmw_opensplice_cpp              > yes
      > pi:bin$ ./listener__rmw_fastrtps_cpp                > yes after restarting talker
      > pi:bin$ ./listener_best_effort                      > yes
      > pi:bin$ ./listener_best_effort__rmw_opensplice_cpp  > yes
      > pi:bin$ ./listener_best_effort__rmw_fastrtps_cpp    > yes

      > ros@192.168.1.20
      > ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > ros:bin$ ./listener                                 > no
      > ros:bin$ ./listener__rmw_opensplice_cpp             > no
      > ros:bin$ ./listener__rmw_fastrtps_cpp               > no
      > ros:bin$ ./listener_best_effort                     > no
      > ros:bin$ ./listener_best_effort__rmw_opensplice_cpp > no
      > ros:bin$ ./listener_best_effort__rmw_fastrtps_cpp   > no

Talker on Virtual Machine
----------------------

> all ros users are on virtual maschines

- - -

    ros@192.168.1.20
    ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
    ros:bin$ ./talker

      > ros@192.168.1.21
      > ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > ros:bin$ ./listener                                 > yes
      > ros:bin$ ./listener__rmw_opensplice_cpp             > yes
      > ros:bin$ ./listener__rmw_fastrtps_cpp               > yes after restarting talker
      > ros:bin$ ./listener_best_effort                     > yes
      > ros:bin$ ./listener_best_effort__rmw_opensplice_cpp > yes
      > ros:bin$ ./listener_best_effort__rmw_fastrtps_cpp   > yes

      > pi@192.168.1.40
      > pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > pi:bin$ ./listener                                  > no
      > pi:bin$ ./listener__rmw_opensplice_cpp              > no
      > pi:bin$ ./listener__rmw_fastrtps_cpp                > no
      > pi:bin$ ./listener_best_effort                      > no
      > pi:bin$ ./listener_best_effort__rmw_opensplice_cpp  > no
      > pi:bin$ ./listener_best_effort__rmw_fastrtps_cpp    > no

- - -

    ros@192.168.1.20
    ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
    ros:bin$ ./talker__rmw_opensplice_cpp

      > ros@192.168.1.21
      > ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > ros:bin$ ./listener                                 > yes
      > ros:bin$ ./listener__rmw_opensplice_cpp             > yes
      > ros:bin$ ./listener__rmw_fastrtps_cpp               > yes
      > ros:bin$ ./listener_best_effort                     > yes
      > ros:bin$ ./listener_best_effort__rmw_opensplice_cpp > yes
      > ros:bin$ ./listener_best_effort__rmw_fastrtps_cpp   > yes

      > pi@192.168.1.40
      > pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > pi:bin$ ./listener                                  > no
      > pi:bin$ ./listener__rmw_opensplice_cpp              > no
      > pi:bin$ ./listener__rmw_fastrtps_cpp                > no
      > pi:bin$ ./listener_best_effort                      > no
      > pi:bin$ ./listener_best_effort__rmw_opensplice_cpp  > no
      > pi:bin$ ./listener_best_effort__rmw_fastrtps_cpp    > no

- - -

    ros@192.168.1.20
    ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
    ros:bin$ ./talker__rmw_fastrtps_cpp

      > ros@192.168.1.20
      > ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > ros:bin$ ./listener                                 > yes after restarting talker
      > ros:bin$ ./listener__rmw_opensplice_cpp             > yes
      > ros:bin$ ./listener__rmw_fastrtps_cpp               > yes after restarting talker
      > ros:bin$ ./listener_best_effort                     > yes
      > ros:bin$ ./listener_best_effort__rmw_opensplice_cpp > yes
      > ros:bin$ ./listener_best_effort__rmw_fastrtps_cpp   > yes

      > ros@192.168.1.21
      > ros:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > ros:bin$ ./listener                                 > yes
      > ros:bin$ ./listener__rmw_opensplice_cpp             > yes
      > ros:bin$ ./listener__rmw_fastrtps_cpp               > yes after restarting talker
      > ros:bin$ ./listener_best_effort                     > yes
      > ros:bin$ ./listener_best_effort__rmw_opensplice_cpp > yes
      > ros:bin$ ./listener_best_effort__rmw_fastrtps_cpp   > yes

      > pi@192.168.1.40
      > pi:bin$ export OSPL_URI=file:///usr/etc/opensplice/config/ospl.xml
      > pi:bin$ ./listener                                  > no
      > pi:bin$ ./listener__rmw_opensplice_cpp              > no
      > pi:bin$ ./listener__rmw_fastrtps_cpp                > no
      > pi:bin$ ./listener_best_effort                      > no
      > pi:bin$ ./listener_best_effort__rmw_opensplice_cpp  > no
      > pi:bin$ ./listener_best_effort__rmw_fastrtps_cpp    > no

rohbotics, 2016-08-27:
Are you using bridged networking for your VM's? 

The default in VirtualBox (and maybe VMWare) is NAT networking, which makes all of your VM's non-contactable from outside the physical machine they are being run on.

You want Bridged Networking for running ROS(1 or 2) on VMs that need to communicate with ROS nodes  on other physical machines.

Harin55, 2016-08-27:
As described, I set all my network adapters to bridged network *wlp4s0* (I use VirtualBox). Because of this I can send data between the VM's. I can also use `ssh` between all my machines (Rasp, Host, VMs). But a talker-listener-connection between the Rasp and the VM's doesn't run.

wjwwood, 2016-09-02:
The DDS implementations require multicast UDP in order to do discovery. In the cases where communication does not work at all, a network configuration that does not allow for multicast UDP is the most likely culprit. For cases where you have to restart to get communication, I'm less certain about that.

Harin55, 2016-09-02:
I try out a another setup: Now my Rasps are connected via LAN to an router and my host pc ist conected via LAN to the same router too. Now it works without any trouble. Changing from WLAN to LAN helps. I hope we will find a solution to let it work together with WLAN too.

dirk-thomas, 2016-10-14:
Since the remaining issue seems to be related to configuration of the network to support UDP multicast I will close this for now since there is nothing to do here for us. If you find additional information about it please feel free to comment here for future readers and/or add some note to the wiki pages about it.

closed 2016-10-14 by dirk-thomas