ros2/ros2 #265: ROS2 doesn't run on Raspberry Pi 2 Model B+ V1.2

Harin55, 2016-09-01:
Hello,

Several tries to install and build ROS2 on an Raspberry Pi 2 Model B+ V1.2 fail.
Even coping the binaries from an other Raspberry Pi doesn't help. After changing
SD-Cards between different Rasps this is the result:

| Raspberry Pi     |  ROS  | ROS2 |
|:-----------------|:-----:|:----:|
| 2 Model B V1.1   |  run  |  run |
| 2 Model B+ V1.2  |  run  |  __rip__ |

## Build error

This is the error during the build of ROS2 on the Rasp V1.2
```
Process package 'rosidl_generator_py' with context:
--------------------------------------------------------------------------------
 source_space => /home/pi/ros2_ws/src/ros2/rosidl/rosidl_generator_py
  build_space => /home/pi/ros2_ws/build/rosidl_generator_py
install_space => /home/pi/ros2_ws/install
   make_flags => -j3
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'rosidl_generator_py'
==> '. /home/pi/ros2_ws/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/pi/ros2_ws/build/rosidl_generator_py'
-- Found ament_cmake: 0.0.0 (/home/pi/ros2_ws/install/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found ament_cmake_nose: 0.0.0 (/home/pi/ros2_ws/install/share/ament_cmake_nose/cmake)
-- Found rosidl_cmake: 0.0.0 (/home/pi/ros2_ws/install/share/rosidl_cmake/cmake)
-- Found rosidl_generator_c: 0.0.0 (/home/pi/ros2_ws/install/share/rosidl_generator_c/cmake)
-- Found ament_lint_auto: 0.0.0 (/home/pi/ros2_ws/install/share/ament_lint_auto/cmake)
-- Found rmw_implementation_cmake: 0.0.0 (/home/pi/ros2_ws/install/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rmw_fastrtps_cpp: 0.0.0 (/home/pi/ros2_ws/install/share/rmw_fastrtps_cpp/cmake)
-- Found OpenSplice: /usr/lib/libcmxml.so;/usr/lib/libcommonserv.so;/usr/lib/libdcpsgapi.so;/usr/lib/libdcpssac.so;/usr/lib/libddsconfparser.so;/usr/lib/libddsconf.so;/usr/lib/libddsdatabase.so;/usr/lib/libddsi2.so;/usr/lib/libddskernel.so;/usr/lib/libddsosnet.so;/usr/lib/libddsos.so;/usr/lib/libddsserialization.so;/usr/lib/libddsuser.so;/usr/lib/libddsutil.so;/usr/lib/libdurability.so;/usr/lib/libspliced.so;/usr/lib/libdcpsisocpp.so;/usr/lib/libdcpssacpp.so  
-- Found PrismTech OpenSplice: /usr/share/opensplice/cmake
-- Found OpenSplice: /usr/share/opensplice/cmake/../../../include/opensplice;/usr/share/opensplice/cmake/../../../include/opensplice/sys;/usr/share/opensplice/cmake/../../../include/opensplice/dcps/C++/SACPP  
-- Found OpenSplice: /usr/lib/libcmxml.so;/usr/lib/libcommonserv.so;/usr/lib/libdcpsgapi.so;/usr/lib/libdcpssac.so;/usr/lib/libddsconfparser.so;/usr/lib/libddsconf.so;/usr/lib/libddsdatabase.so;/usr/lib/libddsi2.so;/usr/lib/libddskernel.so;/usr/lib/libddsosnet.so;/usr/lib/libddsos.so;/usr/lib/libddsserialization.so;/usr/lib/libddsuser.so;/usr/lib/libddsutil.so;/usr/lib/libdurability.so;/usr/lib/libspliced.so;/usr/lib/libdcpsisocpp.so;/usr/lib/libdcpssacpp.so  
-- Found PrismTech OpenSplice: /usr/share/opensplice/cmake
-- Found OpenSplice: /usr/share/opensplice/cmake/../../../include/opensplice;/usr/share/opensplice/cmake/../../../include/opensplice/sys;/usr/share/opensplice/cmake/../../../include/opensplice/dcps/C++/SACPP  
-- Found rmw_opensplice_cpp: 0.0.0 (/home/pi/ros2_ws/install/share/rmw_opensplice_cpp/cmake)
-- Found OpenSplice: /usr/lib/libcmxml.so;/usr/lib/libcommonserv.so;/usr/lib/libdcpsgapi.so;/usr/lib/libdcpssac.so;/usr/lib/libddsconfparser.so;/usr/lib/libddsconf.so;/usr/lib/libddsdatabase.so;/usr/lib/libddsi2.so;/usr/lib/libddskernel.so;/usr/lib/libddsosnet.so;/usr/lib/libddsos.so;/usr/lib/libddsserialization.so;/usr/lib/libddsuser.so;/usr/lib/libddsutil.so;/usr/lib/libdurability.so;/usr/lib/libspliced.so;/usr/lib/libdcpsisocpp.so;/usr/lib/libdcpssacpp.so  
-- Found PrismTech OpenSplice: /usr/share/opensplice/cmake
-- Found OpenSplice: /usr/share/opensplice/cmake/../../../include/opensplice;/usr/share/opensplice/cmake/../../../include/opensplice/sys;/usr/share/opensplice/cmake/../../../include/opensplice/dcps/C++/SACPP  
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.5") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.5m
-- Using PYTHON_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libpython3.5m.so
-- Using Python nosetests: /usr/bin/nosetests3 (1.3.7)
-- Added test 'copyright' to check for copyright in CMake / C / C++ / Python code
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'pep257' to check Python code against some of the style conventions in PEP 257
-- Added test 'pep8' to check Python code against some of the style conventions in PEP 8
-- Added test 'pyflakes' to perform static code analysis on Python code
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/ros2_ws/build/rosidl_generator_py
==> '. /home/pi/ros2_ws/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make -j3' in '/home/pi/ros2_ws/build/rosidl_generator_py'
Scanning dependencies of target rosidl_generator_py__py
Scanning dependencies of target rosidl_generator_py__cpp
[  0%] Built target rosidl_generator_py__cpp
Scanning dependencies of target rosidl_generator_py__rosidl_typesupport_introspection_cpp
[  0%] Building CXX object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/rosidl_generator_py/msg/byte__type_support.cpp.o
[  0%] Built target rosidl_generator_py__py
Scanning dependencies of target rosidl_generator_py__rosidl_generator_c
[  0%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/bool__functions.c.o
[  0%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/byte__functions.c.o
[  1%] Building C object CMakeFiles/rosidl_generator_py__rosidl_generator_c.dir/rosidl_generator_c/rosidl_generator_py/msg/char__functions.c.o
[... 54 lines cut ...]
[ 25%] Built target rosidl_generator_py__rosidl_typesupport_introspection_cpp
[ 26%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_int32_s.c.o
[ 26%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_int64_s.c.o
[ 26%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_int8_s.c.o
[ 27%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_nested_s.c.o
[ 27%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_primitives_s.c.o
[ 27%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_strings_s.c.o
[ 28%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_uint16_s.c.o
[ 28%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_uint32_s.c.o
[ 28%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_uint64_s.c.o
[ 28%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_uint8_s.c.o
[ 29%] Building C object CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/msg/_various_s.c.o
[ 29%] Linking C shared library rosidl_generator_py/rosidl_generator_py/rosidl_generator_py_s__rosidl_typesupport_introspection_c.cpython-35m-arm-linux-gnueabihf.so
[ 29%] Built target rosidl_generator_py__rosidl_typesupport_introspection_c__pyext
CMakeFiles/rosidl_generator_py__rosidl_typesupport_opensplice_cpp.dir/build.make:84: recipe for target 'rosidl_typesupport_opensplice_cpp/rosidl_generator_py/msg/dds_opensplice/bool__type_support.hpp' failed
CMakeFiles/Makefile2:412: recipe for target 'CMakeFiles/rosidl_generator_py__rosidl_typesupport_opensplice_cpp.dir/all' failed
Makefile:138: recipe for target 'all' failed
```

## Run error

Trying to run a copy of excecuteables on V1.2 always show this error:

    pi:bin$ ./talker
    Illegal instruction

rohbotics, 2016-09-02:
What operating system are you running on the Pi?

As far as I know there is only one model of the Raspberry Pi 2, so I am confused as to which models you are referencing in your table.

rohbotics, 2016-09-02:
After taking a closer look at the error, it seems like you are trying to run on a Raspberry Pi 1.
The Pi1 (all models) have an old ARMv6 CPU. Looks like you pulled in opensplice from the debs that OSRF provides, which are compiled for a minimum ARMv7 architecture.

Debian and Ubuntu use the code name 'armhf' to refer to 32-bit ARMv7 with hardware floating point.
Confusingly Raspbian reuses 'armhf' to mean 32-bit **ARMv6** with hardware floating point.

This means that you don't get any errors when installing the opensplice deb, which is installable on 'armhf' but only works on ARMv7 and up.

You should either remove opensplice, or try to build opensplice from source (hard).

@tfoote I don't think that OSRF has any intention of supporting binaries on ARMv6?

Harin55, 2016-09-02:
Hi, thanks for your response. And sorry for my negligence. One of them was an Raspberry Pi Model B+ V1.2 (Not a Rasp 2!).
At the moment I work with Raspbian Jessie which I upgraded to Stretch. After that ROS2 runs like a charm. Together with ROS Kinetic it comes to a conflict at the Python libraries, but this solvable with exporting `PYTHONPATH` to the ROS Kinetic libs.

And thanks for your good explanation. If I found some time I will try to build Opensplice as you recommended. But first I finish my demo project. The use of ROS2 together with ARM architectures is really interesting for us here (use-cases in aviation and astronautics).

closed 2016-09-02 by Harin55