ros2/ros2 #266: ROS 1 bridge not in Linux binary distribution

wjwwood, 2016-09-07:
@geoffviola I reverted your wiki change here:

https://github.com/ros2/ros2/wiki/Linux-Install-Binary/_compare/419c58fb171c6f66baa7d5fd8c861c5afc43b8bb...ae967707b9f9651a88ae377eb412444c1ffadb93

The dynamic_bridge is supposed to be in the Linux binary package, but that broke when we changed to Xenial (kinetic) from Trusty (indigo). Thanks for catching the problem for us though, we're looking into it.

geoffviola, 2016-09-07:
I guess it was a little rash to delete the whole thing without review. My intention was to delete it since it didn't reflect the current alpha, but I can see it act as a TODO. Thanks for the update.

wjwwood, 2016-09-07:
Well, it's not a TODO so much as what should be. I consider the current state to be a bug. We'll fix the binary.

wjwwood, 2016-09-07:
Ok, I updated the binaries and posted them here:

https://github.com/ros2/ros2/releases/tag/release-alpha7

The binaries with `-1` added to the name have the dynamic bridge.

closed 2016-09-07 by wjwwood

wjwwood, 2016-09-07:
Sure, no problem. Thanks for pointing it out to us.

dirk-thomas, 2016-09-14:
The released `ros1_bridge` binaries with OpenSplice still don't work. There are linked against the wrong type support: see ros2/ros1_bridge#30

PR #267 (2016-09-07): use Kinetic instead if Indigo for the bridge
Until now the packaging jobs tried to source indigo which isn't available on Xenial and therefore skipped building the bridge.

Fixes #266.