ros2/ros2 #268: [windows] hardcoded filepath for release.bat in opensplice

Karsten1987, 2016-09-12:
I couldn't get the talker to run as the setup.bat failed beforehand. Reason for this was the hardcoded filepath `c:\dev\opensplice\HDE...` for opensplice. AFAIK, there is no check whether the OSPL_HOME variable was already set.

quick fix solution:

```
if "%OSPL_HOME%" == "" (
	set OSPL_HOME=C:\dev\opensplice\HDE\x86_64.win64
	echo OSPL_HOME variable not set. Assuming %OSPL_HOME%
)
```

Karsten1987, 2016-09-12:
I've just seen [this](https://github.com/ros2/rmw_opensplice/blob/master/opensplice_cmake_module/env_hook/opensplice.bat.in#L1) which seems to fix that. However, the latest windows release suffers still from the problem.

dirk-thomas, 2016-09-12:
The proposed patch looks good to me. Please create a pull request with it.

dirk-thomas, 2016-09-13:
The proposed patch looks good to me. Please create a pull request with it.

For Connext we do a [similar check](https://github.com/ros2/rmw_connext/blob/master/connext_cmake_module/env_hook/connext.bat.in). Maybe warn about the difference?

Karsten1987, 2016-09-13:
I think the problem may only occur on the release side. 
the first line of release.bat of the windows release: `set "OSPL_HOME=C:/dev/opensplice/HDE/x86_64.win64"` 

where are the values set in the first place?

dirk-thomas, 2016-09-13:
The file is expanded from CMake (https://github.com/ros2/rmw_opensplice/blob/919206678d6028edb56d5c99aa99434bd58edba4/opensplice_cmake_module/CMakeLists.txt#L16) using [configure_file](https://cmake.org/cmake/help/v3.5/command/configure_file.html),

dhood, 2018-05-01:
closed by https://github.com/ros2/rmw_opensplice/pull/219

closed 2018-05-01 by dhood