ros2/ros2 #269: [windows] fastrtps talker fails because of missing boost dependency

Karsten1987, 2016-09-13:
talker.exe System Error
```
The program can't start because boost_thread-vc140-mt-1_61.dll is missing from your computer. Try reinstalling the program to fix this problem.
```

I am downloading boost 1.60 from [here](https://sourceforge.net/projects/boost/files/boost-binaries/1.60.0/).

dhood, 2016-09-13:
This is from binaries right?
If you follow [this step](https://github.com/ros2/ros2/wiki/Windows-Development-Setup#installing-boost) from the source install does it fix it? That section might need to be added to the binary install page now. It looks like the page was written at a time when [only opensplice was packaged](https://github.com/ros2/ros2/wiki/Windows-Install-Binary#getting-a-dds-vendor)

Karsten1987, 2016-09-13:
@dhood solution works. 

For completeness, the binary installation only required the `PATH` variable to be set.

closed 2016-09-13 by Karsten1987

dhood, 2016-09-14:
so @Karsten1987 the binary instructions for windows should mention that if you're using fastrtps you need to install boost and check/set the PATH variable, is that the outcome?