ros2/ros2 #271: listener fails to reconnect

Karsten1987, 2016-09-15:
I ran a couple of tests with the talker/listener examples on windows/linux and different vendors such as opensplice and fastrtps.

I noticed that the `listener__rmw_fastrtps` would not reconnect to the `talker__*` once you stop and restart it while keeping the talker alive. However, same setup seems to work fine with the opensplice implementation. 

This might be related to https://github.com/ros2/ros2/issues/79. Can anybody reproduce this?

wjwwood, 2016-09-15:
I'll try to reproduce it on my Mac. I think #79 is specific to opensplice, but could be related. This sounds like an issue with discovery, perhaps unique to FastRTPS, but it's hard to be sure.

wjwwood, 2016-09-15:
I was unable to reproduce it on my mac, though after talking with @Karsten1987 it may be because I'm using master and he is using alpha7. He's going to try master out I think.

Karsten1987, 2016-09-15:
problem does not occur on latest master build. 
Seems to be an issue of the alpha-7 release.

closed 2016-09-15 by Karsten1987