ros2/ros2 #272: swap naming for local_setup.bat and setup.bat

Karsten1987, 2016-09-15:
Why is the ws source file on windows called differently from the one on linux? Would it make sense to stay consistent here and call both entry files `setup.py` and `setup.bat`, respectively?

wjwwood, 2016-09-15:
I think you mean `setup.bash` and `setup.bat`?

Both Linux and Windows have both the `setup.*` and `local_setup.*` files. For a long time, and maybe still as I couldn't find the issue that tracks this right off, Windows did not have a `setup.bat` file implemented because... batch is awful. That's likely the reason they were different. If Windows now has `setup.bat` being created, we can change them to both use `setup.*`.

Karsten1987, 2016-09-15:
I am referring to [this](https://github.com/ros2/ros2/wiki/Windows-Development-Setup#testing-and-running) part of the install wiki, where it says `call install\local_setup.bat`.

mikaelarguedas, 2016-09-15:
If this of any help I always source local_setup.* whatever the platform I'm
using.
I think this section
<https://github.com/ros2/ros2/wiki/Linux-Development-Setup#build-the-prototype-using-the-bootstrap-script-from-ament_tools>
also
refer to local_setup.bash as the file to source after compiling your
workspace

Karsten1987, 2016-09-15:
my bad, I blindly assumed ros2 would be sourced similar to ros1 with a `setup.bash`. The wiki is therefore consistent.

closed 2016-09-15 by Karsten1987

dirk-thomas, 2016-09-15:
To clarify the difference:
* The `local_setup` file under a prefix sources all package-specific `local_setup` files inside this prefix.
* The `setup` file sources all parent prefixes before and then invokes the sibling `local_setup` file.

wjwwood, 2016-09-15:
No worries. As I said, sourcing `setup.*` should also work (I think we got around to implementing it on Windows). So your assumption is not a bad one. We've just added a distinction between source this workspace plus all workspaces it used (`setup.*`) and just source this workspace (`local_setup.*`). Since most of these tutorials use just one workspace, local setup is fine.