ros2/ros2 #274: Building dlls on windows

firesurfer, 2016-09-19:
I'm trying to run my C# wrapper on windows and realized that there are currently no dlls built. Unfortunately I need them, because the native interop of the .Net Runtime does native calls by loading dlls at runtime of the program. Would it be possible to the tell be build tool to create dlls and lib files?

wjwwood, 2016-09-19:
What is not building dll and lib files? As far as I know they should be being built by our packages.

firesurfer, 2016-09-19:
Sry for not making a clear statement.
So I took the steps describes here ( https://github.com/ros2/ros2/wiki/Windows-Development-Setup ) in  order to setup ros2 on windows. The compiler(or buildsystem) output in the `<ros2_ws>/install/lib` directory only consists of *.lib files which are static libraries on windows (so for example there is a rcl.lib and a rclcpp.lib) . So my question is how to build ros2 with dynamic libraries on windows?

dirk-thomas, 2016-09-19:
Have you checked the `install/bin` directory? That is the location where the `.dll` files are being installed to.

firesurfer, 2016-09-19:
Thank you very much. I really wouldn't have expected to have dlls in the `install/bin` directory.

closed 2016-09-19 by firesurfer

wjwwood, 2016-09-19:
It's a Windows thing. It was a surprise to me too.